KLayout/klayout #1164: Deleting cells: 'basic_string: construction from null is not valid'

patrickschulz, 2022-09-29:
Whenever I'm deleting any cells, I get this message, which is problematic since I can not delete a bunch of cells simultaneously. There is a similar issue (https://github.com/KLayout/klayout/issues/1099), but in my case this does not have to do anything with any PDK or pcells. I can reproduce this with a fresh config, I use the (default) technology and inserted an empty cell in the toplevel layout. When I delete that, I get this error.

My setup:
arch linux (kernel 5.19.11)
klayout 0.27.11
ruby 2.7
qt 5.15

Kind regards,
Patrick

klayoutmatthias, 2022-09-29:
Thanks for this report

However I seem not to be able to reproduce it (0.27.12, Ubuntu-22). Here is what I did: 

```
KLAYOUT_HOME=/dev/null klayout -e
```

(`KLAYOUT_HOME=/dev/null` forces default config).

Then:
* File/New Layout
* Right click on cell list -> New Cell, call it "A"
* Select cell A, right click on cell list -> Delete Cell

There is no error message :(

patrickschulz, 2022-09-29:
Thanks for your quick reply. I tried it again exactly like you, unfortunately I **can** reproduce this even then...
Is there a way to see where this error is coming from? I've tried setting a higher debug level but the result is not really helpful to me. I looked through it but nothing caught my eye. I'm happy to dig deeper, but I need some pointer on how to do that.

klayoutmatthias, 2022-09-30:
The message is pretty generic - it says that a `std::string` has been created from a null `const char *`. 

I can see the error on an archlinux docker image with klayout from the package repo. I'll try to debug the problem.

Matthias

joamatab, 2022-10-31:
Any change we could get a new 0.27.11 release with this patch?

I see it on Arch and Debian

https://github.com/KLayout/klayout/issues/1176

klayoutmatthias, 2022-11-30:
Was released with 0.27.12

closed 2022-11-30 by klayoutmatthias