KLayout/klayout #1165: Successfully built in Ubuntu 22.04 but crashes on launching

AbhishekDas2000, 2022-09-30:
The build does not run properly
build options
`./build.sh -j4`
Output of build
`Build successfully done.
Artefacts were installed to /home/abhishek/Downloads/klayout-0.27.11/bin-release`
However on running ./klayout it gives the following error:- 
 `./klayout: error while loading shared libraries: libklayout_tl.so.0: cannot open shared object file: No such file or directory
`

Kazzz-S, 2022-09-30:
I use the following Bash script under the "bin-release" directory.

```
#!/bin/bash
#--------------------------------------------------------------------
# File: klayout.sh
#
# Description: Startup script for KLayout 0.25 or later
#--------------------------------------------------------------------
SCRIPT_DIR=$(cd $(dirname $(readlink -f $0 || echo $0));pwd -P)

KLAYOUT_LIBS=$SCRIPT_DIR
KLAYOUT_EXE=$SCRIPT_DIR/klayout
RLATED_LIBS=/usr/local/lib

export LD_LIBRARY_PATH=$KLAYOUT_LIBS:$RLATED_LIBS:$LD_LIBRARY_PATH
eval "$KLAYOUT_EXE -style=fusion $*"
```

klayoutmatthias, 2022-09-30:
@AbhishekDas2000 This is not a crash, you need to point KLayout to the shared objects using the LD_LIBRARY_PATH environment variable as @Kazzz-S pointed out. Alternatively you can specify an RPATH to point to your final installation location using the "-rpath" option in the build script.

Matthias

AbhishekDas2000, 2022-10-01:
Thanks everyone

closed 2022-10-01 by AbhishekDas2000