KLayout/klayout #1176: Basic string: construction from Null is not valid

joamatab, 2022-10-31:
How can we remove `Basic string: construction from Null is not valid` when modifying Klayout Pcells?

[image]

We saw this when wrapping gdsfactory Pcells in klayout
https://github.com/gdsfactory/gdsfactory/issues/800

However we noticed it also happens with klayout builtin PCells

klayoutmatthias, 2022-11-01:
Could you try whether this is fixed with master? Master builds can be downloaded here: https://www.klayout.org/downloads/master/

I can release some 0.27.12 version with the patch for #1164, but the message is pretty generic. There are many reasons for this error.

If this problem can still be reproduced on master, maybe you can give me a hint how to reproduce it.

Matthias

joamatab, 2022-11-01:
Hi Matthias,

The release from master works perfectly without any issues

thank your for your help

joamatab, 2022-11-01:
Thank you Matthias!

closed 2022-11-01 by joamatab