KLayout/klayout #1179: Mac DMGs for 0.27.12

Kazzz-S, 2022-11-05:
Dear @klayoutmatthias,

Thank you for the maintenance release.

As usual, different DMG packages of **KLayout 0.27.12 for Mac** are available as below ([0.27.12 Dropbox Link](https://www.dropbox.com/sh/pz0rhc901nw29sh/AACEaqX3GPpyJmdHUJyR37sKa?dl=0)).
**Please note that LW-\*.dmg**
**1. uses Ruby 3.1 instead of 2.[5|7] from version 0.27.11 and**
**2. includes _pymod_(\*.whl, \*.egg) from version 0.27.9 for experimental purposes.**

## DMG file naming convention ##
**0.27.12 for Catalina is taken as an example here.**
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp31: Ruby3.1 from MacPorts; hb31: Ruby3.1 from Homebrew; ana3: Ruby3.1 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp38: Python 3.8 from MacPorts; hb38: Python 3.8 from Homebrew; ana3: Python3.8 from Anaconda3

## For the primary platform, macOS Catalina (10.15.7) ##
Five macOS DMGs for **Catalina** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 12.2.4 on Monterey (12.6).

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.27.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.27.12-macOS-Catalina-1-qt5MP-Rmp31Pmp38.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.27.12-macOS-Catalina-1-qt5Brew-Rhb31Phb38.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.27.12-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.27.12-macOS-Catalina-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those with the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For macOS Big Sur (11.7.1) ##
Four macOS DMGs for **Big Sur** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 12.2.4 on Monterey (12.6).

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21a|~~BigSur (11.6)~~|~~ST-klayout-0.27.12-macOS-BigSur-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(21a)|
21b|Catalina (10.15)|ST-klayout-0.27.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21b)|
22|BigSur (11.6) with MacPorts dev. env.|LW-klayout-0.27.12-macOS-BigSur-1-qt5MP-Rmp31Pmp38.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.6) with Homebrew dev. env.|LW-klayout-0.27.12-macOS-BigSur-1-qt5Brew-Rhb31Phb38.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.6) with Anaconda3 dev. env.|LW-klayout-0.27.12-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.6)|HW-klayout-0.27.12-macOS-BigSur-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.1 does not link the legacy Python 2.7 library.
(21b) However, ST*.dmg for Catalina, which shares OS-bundled Ruby2.6 and Python-2.7, and Qt5 from MacPorts is embedded, can run on Big Sur.
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For macOS Monterey (12.6.1) ##
Four macOS DMGs for **Monterey** are available.

**LW- and HW- packages use Python 3.8 in MacPorts, Homebrew and Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
31a|~~Monterey (12.6)~~|~~ST-klayout-0.27.12-macOS-Monterey-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31a)|
31b|~~Catalina (10.15)~~|~~ST-klayout-0.27.12-macOS-Catalina-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31b)|
32|Monterey (12.6) with MacPorts dev. env.|LW-klayout-0.27.12-macOS-Monterey-1-qt5MP-Rmp31Pmp38.dmg|MacPorts|MacPorts|MacPorts|(22)|
33|Monterey (12.6) with Homebrew dev. env.|LW-klayout-0.27.12-macOS-Monterey-1-qt5Brew-Rhb31Phb38.dmg|Homebrew|Homebrew|Homebrew|(33)|
34|Monterey (12.6) with Anaconda3 dev. env.|LW-klayout-0.27.12-macOS-Monterey-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(34)|
35|Monterey (12.6)|HW-klayout-0.27.12-macOS-Monterey-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew|(35)|

(31a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.1 does not link the legacy Python 2.7 library.
(31b) **Monterey 12.3.1 finally eliminated Python-2.7 binary. Hence, ST-\* package no longer runs.**
(32) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(33) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(34) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(35) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.8 from Homebrew are embedded, hence, heavy.

## For the latest macOS Ventura (13) ##
**I've not yet set up the development environment for Ventura.** 
However, I could start this version (Srl. No.45) on a Ventura virtual machine. 
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
45|Monterey (12.6)|HW-klayout-0.27.12-macOS-Monterey-1-qt5Brew-RsysPhb38.dmg|Homebrew|OS-bundled|Homebrew| |

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
7|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
8|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) I expect that build is still possible.

Would you please close this ticket after publishing the packages?
With warm regards,
Kazzz-S

klayoutmatthias, 2022-11-06:
Files are on the server. Many thanks for providing continuous support for almost a decade now! :)

closed 2022-11-06 by klayoutmatthias

Kazzz-S, 2022-11-06:
Time flies by so fast. I had not noticed that :-)
So I have been a beneficiary of KLayout for more than a decade.
Thank you very much.