KLayout/klayout #1184: friendly suggestions to improve the initial experience of finding KLayout for newcomers

gabrielgrant, 2022-11-08:
Hey @klayoutmatthias, thanks for this awesome-looking tool! 

I didn't find the sources for the website github, so hoping it's ok to file this here!

Thought it might be useful feedback that, despite having a background in electrical engineering, and some interest in the field of chip design, it took me a little while to figure out what KLayout was actually for (I also work in other areas of design where "layout" and "mask" have very different meanings). At the very least, I'd strongly suggest at least using the term "integrated circuit" somewhere on the homepage (ideally in the main headline/tagline: "KLayout - Your _Integrated Circuit_ Mask Layout Friend"

Beyond this, it would also be great if there were a bit of basic info about what this tool is, and why I might want (or not want) to use it (ideally both on the homepage and in the README). This is useful both for people who aren't experienced in the field, but also so that those who are looking for exactly this can find it! (right now KLayout doesn't seem to appear in the first few pages of a google search for "integrated circuit mask layout tool"), which isn't so surprising, since as far as I can tell the term "integrated circuit" doesn't appear anywhere on the site 😶 

[image]

Again, appreciate all the work you've put into KLayout, and just hoping this can help it find a wider audience!

klayoutmatthias, 2022-11-09:
I'd not dare to call KLayout IC design related - there are design systems out there for this job. KLayout is not a design system. It's a tool primarily for handling GDS2 (and OASIS) files are related tasks.

If you care about KLayout visibility: why not creating a Wikipedia article and cross-linking this KLayout page on some of the pages related to EDA tools? As a general policy there, I am not supposed to self-promote my tool on Wikipedia, so I cannot do that myself. But you can help other people finding KLayout by exposing it. 

Matthias

klayoutmatthias, 2022-12-06:
Please consider the Wikipedia option and contributing. I cannot and don't want to self-promote.

closed 2022-12-06 by klayoutmatthias