KLayout/klayout #1194: very slow width_check in DRC flat mode on a very small layout

atorkmabrains, 2022-11-23:
Hi @klayoutmatthias ,

I'm still working on investigating the performance issues in DRC that we are facing. I ran another profiling to see where is the main issue:
```
2022-11-22 21:36:11 +0200: Memory Usage (7321292K) : DRC Total Run time 556.591414 seconds
  %   cumulative   self              self     total
 time   seconds   seconds    calls  ms/call  ms/call  name
 42.15   232.23    232.23       14 16587.62 16587.62  RBA::Edges#width_check
 13.18   304.86     72.63       48  1513.22  1513.22  RBA::Region#complex_op
  4.97   332.24     27.38       66   414.83   414.83  RBA::Region#not_inside
  4.76   358.45     26.21       70   374.49   374.49  RBA::Edges#&
  4.54   383.45     25.00      175   142.85   142.85  RBA::Region#not_interacting
  3.49   402.69     19.23       88   218.56   218.56  RBA::Edges#interacting
  2.63   417.16     14.47       46   314.55   314.55  RBA::Region#space_check
  2.52   431.03     13.87      195    71.14    71.14  RBA::Region#-
  2.41   444.28     13.25       81   163.59   163.59  RBA::Region#not_outside
  1.92   454.86     10.58      153    69.17    69.17  RBA::Region#interacting
  1.89   465.27     10.41      159    65.48    65.48  RBA::Region#|
  1.68   474.51      9.24       79   116.93   116.93  RBA::Region#enclosing_check
  1.55   483.05      8.54       55   155.32   155.32  RBA::Region#sized
  1.48   491.20      8.14      146    55.77    55.77  RBA::Region#&
  1.22   497.94      6.74      283    23.81    23.81  RBA::Region#inside
  1.16   504.35      6.41       41   156.43   156.43  RBA::Edges#with_length
  1.04   510.07      5.72       26   219.84   219.84  RBA::Edges#enclosing_check
  0.84   514.70      4.64      243    19.08    19.08  RBA::Region#edges
  0.84   519.31      4.61       11   418.94   418.94  RBA::Edges#enclosed_check
  0.82   523.85      4.54       68    66.74    66.74  RBA::Edges#-
  0.80   528.24      4.39       14   313.66   313.66  RBA::EdgePairs#edges
  0.57   531.36      3.12      149    20.94    20.94  RBA::Region#outside
  0.48   534.00      2.64      106    24.91    24.91  RBA::Region#merged
  0.44   536.42      2.42       94    25.70    25.70  RBA::Region#separation_check
  0.32   538.15      1.74       15   115.82   115.82  RBA::Edges#separation_check
  0.25   539.51      1.36       15    90.35    90.35  RBA::Region#corners
  0.24   540.82      1.31       28    46.88    46.88  RBA::Edges#centers
  0.21   541.99      1.17       35    33.45    33.45  RBA::Edges#extended
  0.20   543.11      1.12        6   185.92   185.92  RBA::Region#with_bbox_min
  0.17   544.02      0.92        3   305.57   305.57  RBA::Edges#with_angle
  0.15   544.83      0.81       19    42.44    42.44  RBA::Region#covering
  0.13   545.56      0.73       10    72.59    72.59  RBA::Edges#not_in
  0.09   546.03      0.47       41    11.44    11.44  RBA::Region#width_check
  0.08   546.48      0.45        1   450.54   450.54  RBA::Edges#in
  0.08   546.92      0.45    27555     0.02     0.06  DRC::DRCEngine#src_line
  0.07   547.33      0.41       12    34.03    34.03  RBA::Edges#not_interacting
  0.06   547.64      0.31    56698     0.01     0.01  Kernel#!~
  0.04   547.85      0.21       23     8.99     8.99  RBA::EdgePairs#first_edges
  0.03   548.02      0.18     4212     0.04   130.27  DRC::DRCEngine#run_timed
  0.03   548.20      0.17       11    15.85    15.85  RBA::EdgePairs#with_length_both
  0.02   548.33      0.13    23536     0.01    26.06  DRC::DRCEngine#_context
  0.02   548.45      0.13    36806     0.00     0.01  DRC::DRCLayer#data
  0.02   548.58      0.13       22     5.73     5.73  RBA::Edges#|
  0.02   548.70      0.13      106     1.18     1.18  RBA::Region#with_angle
  0.02   548.83      0.12       22     5.58     5.58  RBA::EdgePairs#second_edges
  0.02   548.94      0.11    56698     0.00     0.00  String#=~
  0.02   549.05      0.11     8426     0.01     0.02  DRC::DRCEngine#_process_events
  0.02   549.16      0.11     4735     0.02     0.18  Array#each
  0.02   549.26      0.09      620     0.15     0.15  Kernel#`
  0.02   549.35      0.09      345     0.26     0.26  RBA::EdgePairs#polygons
  0.02   549.44      0.09        4    21.40    21.40  RBA::Edges#space_check
  0.01   549.52      0.08     5996     0.01   181.42  DRC::DRCEngine#_tcmd
  0.01   549.60      0.08       70     1.17     1.17  RBA::Region#overlapping
  0.01   549.68      0.08      106     0.72     0.72  RBA::Region#grid_check
  0.01   549.74      0.06     4212     0.02     0.02  Kernel#caller
  0.01   549.80      0.06     4214     0.01     0.04  DRC::DRCEngine#info
  0.01   549.86      0.06     9020     0.01     0.01  Class#new
  0.01   549.91      0.04    29048     0.00     0.00  Kernel#is_a?
  0.01   549.95      0.04       19     2.25     2.25  RBA::Region#holes
  0.01   549.99      0.04        1    37.38    37.38  RBA::Layout#read
  0.01   550.02      0.04     1229     0.03     0.04  DRC::DRCEngine#_prep_value
  0.01   550.06      0.04    16852     0.00     0.00  RBA::Application.instance
  0.01   550.09      0.03      620     0.05     0.23  DRC::DRCExecutable#execute
[... 295 lines cut ...]
  0.00   550.79      0.00        1     0.00     0.00  Integer._dbu=
  0.00   550.79      0.00        1     0.00     0.00  DRC::DRCEngine#_generator
  0.00   550.79      0.00        1     0.00     0.00  DRC::DRCEngine#verbose=
  0.00   550.79      0.00        2     0.00     0.00  DRC::DRCSource#layout
  0.00   550.79      0.00        1     0.00     0.00  DRC::DRCSource#cell_obj
  0.00   550.79      0.00        1     0.00     0.00  Integer#fdiv
  0.00   550.95      0.00        1     0.00 550954.64  #toplevel
22-Nov-2022 21:36:15 | INFO    | 
Congratulations !!. No DRC Violations found in user_proj_example for gf180mcu.drc rule deck with switch gfC
22-Nov-2022 21:36:15 | INFO    | Klayout GDS DRC Clean
```

The layout was not that big. I don't recommend running tiled version as it gives false errors.

I don't recommend using Google version as it's not updated. Please use Mabrains version as it's updated.

To duplicate this run, you need the following:
* Download the repo and use the run_drc.py command on the following GDS:
https://github.com/mabrains/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/rules/klayout/drc/run_drc.py
https://github.com/efabless/caravel_user_project/blob/gf180mcu/gds/user_proj_example.gds

Here is the run command:
```
python run_drc.py --gf180mcu=C --path=./user_proj_example.gds
```

atorkmabrains, 2022-11-23:
Also, I believe you don't do pruning for edges for example for width_check here:
https://github.com/KLayout/klayout/blob/7a9e9989d3fb8259d9ee3b5d6c8fe803092e7f2b/src/db/db/dbAsIfFlatEdges.cc#L695

Would it be possible to prune the edges to the ones that are in the same polygon only:
https://github.com/KLayout/klayout/blob/7a9e9989d3fb8259d9ee3b5d6c8fe803092e7f2b/src/db/db/dbAsIfFlatEdges.cc#L705

klayoutmatthias, 2022-11-23:
Here is another remark because I just noticed that:

In my modified code above this expression is given:

```
mdn3b_l1 = poly2.
            edges.
            and(ncomp).
            or(mvsd_mdn).
            and(ldmos_xtor).
            and(dualgate).
            not(ngate.
                  not(mvsd).
                  edges.
                  interacting(poly2.
                                edges.
                                and(ncomp).
                                or(mvsd_mdn)
                             ).
                  and(wide_ngate)
            )
```

I contains the following subexpression two times:

```
poly2.
   edges.
   and(ncomp).
   or(mvsd_mdn)
```

This expression is actually computed twice, so it can be optimized:

```
ngate_wsides = poly2.edges.and(ncomp).or(mvsd_mdn)

mdn3b_l1 = ngate_wsides.
            and(ldmos_xtor).
            and(dualgate).
            not(ngate.
                  not(mvsd).
                  edges.
                  interacting(ngate_wsides).
                  and(wide_ngate)
            )
```

I think there are many such optimization opportunities. But again, difficult without a test case (or better: a test suite).

Matthias

atorkmabrains, 2022-11-24:
@klayoutmatthias Thanks for the code review. We have a test suite for this rule deck. I'll send you the link.

I'll take a look at the coding of this rule and see if we can change the style of coding avoiding the use of edge with. BTW, in the commercial tool it keeps track of the edge from which polygon it comes from. I thought you might be doing something similar.

As for tiling, we have used double the maximum measurement in the rule deck as border, and yet it generated weird results. Let me test it again.

Deep didn't improve the runtime either.

klayoutmatthias, 2022-11-24:
Tiling is not necessarily a runtime improvement, but it keeps memory low as it only works on one tile and with multiple cores, tiles are run in parallel. However there is some overhead (because of border work example) and hence there is some trade-off. But I have hardly encountered a case when tiling creates artefacts and core scaling usually is very good.

If you work on larger designs - specifically with memory blocks - you cannot use flat mode. Memory footprint will simply explode.

Matthias

atorkmabrains, 2022-11-24:
@klayoutmatthias Thanks for your input. I believe. I need to redo many parts of the script as I believe we use edges then width/separation check very often. It was a bad assumption on my part that you prune based on original polygon.

klayoutmatthias, 2022-11-24:
@atorkmabrains I don't think you need to undo all - in fact only a few. 

If the width you check is a small one, the edge width check is very efficient. Even if it does not consider the polygon it comes from it will do an efficient local clustering (scan line algorithm in flat case). So if all relevant edges are within the search range and the local clusters do not involve many foreign or irrelevant edges, there is no performance or memory penalty.

The issue with the large-range width check is that the local cluster will be very large and in case of dense layout you collect a huge neighbourhood for each input edge. This creates manifold interactions as we've seen in the green error markers before. The number of interactions roughly grows with the square of the check distance which explains the extreme runtime penalty.

My gut feeling is that for this technology node every distance below 1 or 2µm can safely be checked with edge operations (I assume you will not test against pathologic cases :) ). 

Best regards,

Matthias

atorkmabrains, 2022-11-24:
@klayoutmatthias that's really helpful information. I would consider this in our implementation.

atorkmabrains, 2023-04-10:
Thanks @klayoutmatthias I'll close this issue as it's no longer needed.

closed 2023-04-10 by atorkmabrains