KLayout/klayout #1196: Partial tool is broken in 0.27.12

iqmtestd, 2022-11-29:
Hi @klayoutmatthias,

Adjusting our waveguides with the Partial tool is not possible any more. Normally I can grab a control point and move it:

[image]

But not any more with the latest code. I've created a minimal Test PCell that works fine in 0.27.11 but not in 0.27.12:

```
import pya

class Test(pya.PCellDeclarationHelper):
  def __init__(self):
    super(Test, self).__init__()

    self.param("l", self.TypeLayer, "Layer", default = pya.LayerInfo(1, 0))
    self.param("path", self.TypeShape, "", default=pya.DPath([pya.DPoint(0, 0), pya.DPoint(100, 0)], 0))

  def produce_impl(self):
    path = list(self.path.each_point())
    pts = [
        path[0],
        path[0] + pya.DPoint(0, 10),
        path[1] + pya.DPoint(0, 10),
        path[1],
    ]
    shape = pya.DPolygon(pts)
    self.cell.shapes(self.l_layer).insert(shape)

class MyLib(pya.Library):
  def __init__(self):
    self.description = "My First Library"
    self.layout().register_pcell("Test", Test())
    self.register("MyLib")

MyLib()
```

 I think this is a bug in KLayout since the commit `5e621091`  "Issue 1145 (#1151)". I don't know how to fix it. Do you have some workaround for this issue?

Thanks,
David

klayoutmatthias, 2022-11-29:
Hi @iqmtestd  Thanks for this report.

I can reproduce the problem with 0.27.12, but not with the latest master. I suspect a merge issue.

I want to release 0.28 soon, so I ask myself if it's easier to focus on that. Would you like to try it? Downloads are available here: https://www.klayout.org/downloads/master/ (disclaimer: binaries are not signed).

Best regards,

Matthias

iqmtestd, 2022-11-30:
Hi @klayoutmatthias,

Thanks for testing it. I've tried latest master and indeed, this demo script works there fine.

Unfortunately, KQCircuits itself does not work at all with 0.28. I guess this is related to some change in KLayout's environment variable handling or that it does not detect our auto-run init scripts under .klayout/python/. How soon do you plan the next release? Maybe I should concentrate on fixing this first.

I close this issue as it is fixed in master.

Best regards,
David

klayoutmatthias, 2022-11-30:
I think I found the problem. It was a missing commit in the issue fix merge.

I'll create a 0.27.13 version which fixes some other issues too for the stable 0.27 branch. 0.28 is slightly experimental.

I'm not aware of a different handling of init scripts. But there are some API changes. Is there an error message when you start KLayout?

iqmtestd, 2022-12-01:
Great, thanks, we look forward to the  fixed version!

About 0.28. No error message, it's just empty, i.e. without KQCircuits.

The `[Local - python branch]` appears empty in the Macro development window. Normally we have a script directory with our KQCircuits init scripts there. Apparently the symbolic link from `.klayout/python/kqcircuits_scripts` to our script directory is ignored by KLayout 0.28.

klayoutmatthias, 2022-12-01:
@iqmtestd I have just release 0.27.13 which should fix your issue (plus two more)

I have created a follow-up ticket for the symlink problem: https://github.com/KLayout/klayout/issues/1198

I will close this ticket and continue on the follow-up one. Thanks for bringing this up.

closed 2022-12-01 by klayoutmatthias

iqmtestd, 2022-12-02:
Thanks, I've verified that this works fine with KQCircuits too!