KLayout/klayout #1200: OAS/GDS writer: fallback needed in case of duplicate cell names

klayoutmatthias, 2022-12-02:
Duplicate cell names should not happen, but scripts may generate them. Also program bugs may trigger such a case.

Currently, the OASIS/GDS writer refuses to write such layout because that may lead to damaged files. A method is needed to repair such layouts.

For example: offering an option to repair such cell names on File/write.

stefanottili, 2022-12-03:
I would not try of repair duplicate cell names.
pcells or scripts that create duplicate cell names in a library should error out the moment that happens.

klayoutmatthias, 2022-12-03:
@stefanottili I considered this, but I turned this feature off again. It makes coding extremely difficult for some applications.

Like this:

When you want to obfuscate a layout (e.g. to generate a test case for me), you can hide the cell names by giving the cells names like "C1", "C2" etc. 

So the code is loop like this:

```
index = 1
foreach cell:
   cell.name = "C" + str(index)
   index += 1
```

So now, if your original layout contains a cell called "C20" as cell number 30 for example, the skript will stop when it tries to assigne "C20" to cell number 20, because this will generate a duplicate.

So with rename refusal in place, the algorithm has to be something like this:

```
foreach cell:
   if cell.name is like "Cnnn":
      cell.name = some_new_and_guaranteed_to_be_unique_name   # good luck

index = 1
foreach cell:
   cell.name = "C" + str(index)
   index += 1
```

Matthias

closed 2023-08-09 by klayoutmatthias