KLayout/klayout #1201: hang in .merged, deep-only, in at least 0.26.11, 0.27.13, backtrace: db::DeepRegion::ensure_merged_polygons_valid()

sto6, 2022-12-03:
A script to recreate. Data source shown within; only layer 55/0 needed. Hangs at about VSIZE=1.5GB,  and does not increase further. My tests were all/only on centos-7.9. Thank you.

```
#!/bin/bash
# in bash:
x=%{

# self-log
lg=min-pmerge.log
mv -f $lg $lg-bak
exec </dev/null &> >(/usr/bin/tee $lg)

exec klayout -b -r "$0"
}
# in klayout(ruby):
STDOUT.sync = true
STDERR.sync = true
# get it here: https://github.com/whutddk/Rift2Go_2300_GF180_MPW0/blob/a269c5678626b943e0d1121c67e60913fc28f53f/gds/user_project_wrapper.gds.gz
source("user_project_wrapper.gds.gz", "user_project_wrapper")
report("report", "/tmp/dg2.km")
deep
puts "dg     = polygons(55,0)"
      dg     = polygons(55,0)
puts "dg_mrg = dg.merged"
      dg_mrg = dg.merged  # <-- hang
puts "done"
```
Sample output with backtrace (kill -ABRT):

```
dg     = polygons(55,0)
dg_mrg = polygons(55,0).merged
addr2line: 'klayout': No such file
ERROR: Signal number: 6
Address: 0x3de0000043a
Program Version: KLayout 0.27.13 (2022-11-30 r40a2677)

Backtrace:
/usr/lib64/klayout/libklayout_lay.so.0 +0x27b6e2 lay::enable_signal_handler_gui(bool) [??:?]
/lib64/libpthread.so.0 +0xf630 __restore_rt [sigaction.c:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x116d8f3 db::Connectivity::connect_global(db::DeepLayer const&, std::string const&) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x1163bc0 db::Connectivity::connect_global(db::DeepLayer const&, std::string const&) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x11bb925 db::hier_clusters<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::build_hier_connections(db::cell_clusters_box_converter<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >&, db::Layout const&, db::Cell const&, db::Connectivity const&, std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> > const*, db::instance_interaction_cache<db::InstanceToInstanceInteraction, std::list<std::pair<db::ClusterInstance, db::ClusterInstance>, std::allocator<std::pair<db::ClusterInstance, db::ClusterInstance> > > >&) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x11bdff1 db::hier_clusters<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::build_hier_connections_for_cells(db::cell_clusters_box_converter<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >&, db::Layout const&, std::vector<unsigned int, std::allocator<unsigned int> > const&, db::Connectivity const&, std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> > const*, tl::RelativeProgress&, db::instance_interaction_cache<db::InstanceToInstanceInteraction, std::list<std::pair<db::ClusterInstance, db::ClusterInstance>, std::allocator<std::pair<db::ClusterInstance, db::ClusterInstance> > > >&) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x11c5e6b db::hier_clusters<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::do_build(db::cell_clusters_box_converter<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >&, db::Layout const&, db::Cell const&, db::Connectivity const&, std::map<unsigned int, tl::equivalence_clusters<unsigned long>, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, tl::equivalence_clusters<unsigned long> > > > const*, std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> > const*) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x11c632e db::hier_clusters<db::polygon_ref<db::polygon<int>, db::disp_trans<int> > >::build(db::Layout const&, db::Cell const&, db::Connectivity const&, std::map<unsigned int, tl::equivalence_clusters<unsigned long>, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, tl::equivalence_clusters<unsigned long> > > > const*, std::set<unsigned int, std::less<unsigned int>, std::allocator<unsigned int> > const*) [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x114ad7e db::DeepRegion::ensure_merged_polygons_valid() const [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0x114fce8 db::DeepRegion::merged() const [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0xdd0191 db::Region::merged() const [??:?]
/usr/lib64/klayout/libklayout_db.so.0 +0xdd43ed gsi::VectorAdaptorIteratorImpl<std::vector<db::Region, std::allocator<db::Region> > >::get(gsi::SerialArgs&, tl::Heap&) const [??:?]
/usr/lib64/klayout/libklayout_rba.so.0 +0x22ebb rba::RubyStackTraceProvider::stack_trace() const [??:?]
/lib64/libruby.so.2.0 +0x1637d2 rb_iter_break_value [??:?]
/lib64/libruby.so.2.0 +0x1751be rb_funcall_with_block [??:?]
/lib64/libruby.so.2.0 +0x175796 rb_funcall_with_block [??:?]
/lib64/libruby.so.2.0 +0x17557e rb_funcall_with_block [??:?]
/lib64/libruby.so.2.0 +0x168af5 rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x16dd0f rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x16f40b rb_yield_splat [??:?]
/lib64/libruby.so.2.0 +0x16fa33 rb_yield_splat [??:?]
/lib64/libruby.so.2.0 +0x1637d2 rb_iter_break_value [??:?]
/lib64/libruby.so.2.0 +0x1751be rb_funcall_with_block [??:?]
/lib64/libruby.so.2.0 +0x168b6c rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x16dd0f rb_call_super [??:?]
/lib64/libruby.so.2.0 +0x164494 rb_catch [??:?]
/lib64/libruby.so.2.0 +0x165c5b rb_check_funcall [??:?]
/lib64/libruby.so.2.0 +0x69d0b rb_protect [??:?]
/usr/lib64/klayout/libklayout_rba.so.0 +0x40f61 void std::vector<std::pair<std::string, gsi::MethodBase const*>, std::allocator<std::pair<std::string, gsi::MethodBase const*> > >::_M_emplace_back_aux<std::pair<std::string, gsi::MethodBase const*> >(std::pair<std::string, gsi::MethodBase const*>&&) [??:?]
/usr/lib64/klayout/libklayout_rba.so.0 +0x45296 std::vector<tl::BacktraceElement, std::allocator<tl::BacktraceElement> >::reserve(unsigned long) [??:?]
/usr/lib64/klayout/libklayout_gsi.so.0 +0x8212d std::vector<std::string, std::allocator<std::string> >::~vector() [??:?]
/usr/lib64/klayout/libklayout_gsi.so.0 +0x823b4 std::vector<std::string, std::allocator<std::string> >::~vector() [??:?]
/usr/lib64/klayout/libklayout_tl.so.0 +0xb76b5 tl::Executable::do_execute() [??:?]
/usr/lib64/klayout/libklayout_lym.so.0 +0x3f2f6 lym::MacroInterpreter::execute_macro(lym::Macro const*) [??:?]
/usr/lib64/klayout/libklayout_lym.so.0 +0x44221 lym::Macro::run() const [??:?]
/usr/lib64/klayout/libklayout_lay.so.0 +0x1148d6 lay::ApplicationBase::run() [??:?]
klayout +0x3168
/usr/lib64/klayout/libklayout_rba.so.0 +0x1ec40
/lib64/libruby.so.2.0 +0x1637d2
/lib64/libruby.so.2.0 +0x1751be
/lib64/libruby.so.2.0 +0x168b6c
/lib64/libruby.so.2.0 +0x16dd0f
/lib64/libruby.so.2.0 +0x16fd9f rb_iseq_eval_main
/lib64/libruby.so.2.0 +0x67f3a
/lib64/libruby.so.2.0 +0x6964d ruby_exec_node
/lib64/libruby.so.2.0 +0x6b18e ruby_run_node
/usr/lib64/klayout/libklayout_rba.so.0 +0x1f880 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
klayout +0x2cd0
klayout +0x2682
/lib64/libc.so.6 +0x22555 __libc_start_main
klayout +0x2878
```

klayoutmatthias, 2022-12-03:
Thanks a lot, nice test case.

Could you try with master?

My tests give 24.5s runtime for the "dg.merged" (1 thread)

Basically, you can also simply take the bounding box:

```
dg_bbox = polygons
dg_bbox.insert(dg.bbox)
```

This takes less than 0.01s :)

Matthias

sto6, 2022-12-04:
trying using 0.28 master:
b3968c31e 2022-12-01T16:29:25-08:00 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #1197 from KLayout/wip

So far appears still hanging (over 20min), same place; but I'll let it run...
I see more recent commit (16e95fa45) I'll try that later.

RE: ".insert(dg.bbox)", I take that is a work-around, when using 0.27.13? Just do that prior to any/first individual layer .merged?

klayoutmatthias, 2022-12-04:
Funny, just checked the master builds from `https://www.klayout.org/downloads/master`:

* CentOS 7.9: hangs 
* Ubuntu 22.04: terminates after 24s

Deadlock maybe?

I'll try to debug.

klayoutmatthias, 2022-12-04:
Got it ... this TODO hits:

[image]

CentOS 7 uses a STL with slow `std::list::size` :(

Matthias

sto6, 2022-12-05:
What then is the prognosis?
I build using Software-Collections (scl devtoolset-7) which provides newer gcc 7.3. But result still uses original OS /lib64/libstdc++.so.6. I tried to get it to statically link & failed (possibly due not having static Qt libs?), yet appears the scl does have its own .a for libstdc++.

klayoutmatthias, 2022-12-05:
@sto6 No worries, I have included a custom implementation for a single-linked list with fast size method. It's included in `wip` branch already and CentOS 7 performance is like Ubuntu now.

closed 2022-12-06 by klayoutmatthias