KLayout/klayout #1206: Segfault at startup with Ubuntu 22.04

mattgodbolt, 2022-12-04:
Using the "official" klayout I get a segfault (that looks similar to #701) on Ubuntu 22.04:

```
$ klayout -v
KLayout 0.26.2
$ apt list klayout
Listing... Done
klayout/jammy,now 0.26.2-3build3 amd64 [installed]
```

On startup:

```
Signal number: 11
Address: 0x8
Program Version: KLayout 0.26.2 (2022-01-13 rLatestSourcePackage)

Backtrace:
/usr/lib/klayout/libklayout_lay.so.0 +0x24b786 lay::SaltDownloadManager::execute(lay::SaltManagerDialog*, lay::Salt&) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x42520
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xbf639
/usr/lib/klayout/libklayout_rba.so.0 +0x43782
/usr/lib/klayout/libklayout_rba.so.0 +0x47ce2
/usr/lib/klayout/libklayout_rba.so.0 +0x2a78d
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc35
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fc6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c32
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a30d rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x241295
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x241da6 rb_funcallv_kw
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x13e1ab rb_class_new_instance_pass_kw
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc35
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fc6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c32
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a30d rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23b072
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23f2f7
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x24293c rb_funcallv
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa69ad rb_protect
/usr/lib/klayout/libklayout_rba.so.0 +0x40d9e
/usr/lib/klayout/libklayout_lym.so.0 +0x2f2e2 _ZNK3lym5Macro3runEv
/usr/lib/klayout/libklayout_lym.so.0 +0x2f455
/usr/lib/klayout/libklayout_lym.so.0 +0x2f401
/usr/lib/klayout/libklayout_lay.so.0 +0x10e4ed _ZN3lay15ApplicationBase8init_appEv
/usr/lib/klayout/klayout +0x4595
/usr/lib/klayout/libklayout_rba.so.0 +0x24944
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x22fc35
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x231fc6
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x234c32
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0x23a30d rb_vm_exec
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa5318
/lib/x86_64-linux-gnu/libruby-3.0.so.3.0 +0xa8e1a ruby_run_node
/usr/lib/klayout/libklayout_rba.so.0 +0x273e2 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/usr/lib/klayout/klayout +0x3d1e
/lib/x86_64-linux-gnu/libc.so.6 +0x29d90
/lib/x86_64-linux-gnu/libc.so.6 +0x29e40 __libc_start_main
/usr/lib/klayout/klayout +0x3e95
```

If I "ok" through that a few times it "works" but obviously I'm a little concerned about data loss :)

Please let me know how I can help further.

klayoutmatthias, 2022-12-04:
@mattgodbolt 0.26.2 is super old and has seen a lot of bug fixes. Latest stable version is 0.27.13.

I know that Ubuntu ships with this version, but I am not happy with that. You can try a download from www.klayout.org which supports Ubuntu 22.04 (my own OS).

From the stack trace I think the problem stems from some auto-start macro. You can try to rename "~/.klayout" to some other name and start with a fresh configuration. I assume one of the files in "~/.klayout/macros" or "~/.klayout/ruby" causes the instability you see.

Matthias

mattgodbolt, 2022-12-06:
I'll be happy to update! I'll post back here with details. Sorry I missed your quick reply!

mattgodbolt, 2022-12-07:
Just for completeness:

- renaming `~/.kaylout` out of the way did not fix the problem with 0.26.2
- installing the latest from the website (0.27.13-1) required a bit of `apt` shenanigans (`apt --fix-broken install` did the job, removing some `gles` packages and installing some new `qt5` packages)
- but most importantly _it works!_

Thanks so much for this! Closing :)

closed 2022-12-07 by mattgodbolt