KLayout/klayout #1207: Update dev version to test pypi

proppy, 2022-12-05:
It seems that the current development version (0.28) is not uploaded to test pypi:
https://test.pypi.org/project/klayout/#history
despite the `upload_to_test_pypi` job succeeding:
https://github.com/KLayout/klayout/actions/runs/3615727326/jobs/6093247182

proppy, 2022-12-05:
looking at the log of the jobs there seems to be a few authorization error:
```
25hWARNING  Error during upload. Retry with the --verbose option for more details. 
ERROR    HTTPError: 403 Forbidden from https://test.pypi.org/legacy/            
         Invalid or non-existent authentication information. See                
         https://test.pypi.org/help/#invalid-auth for more information
```

proppy, 2022-12-05:
Not this could also be due to test pypi preventing a release for a given version to be overwritten.

According to https://stackoverflow.com/a/63944201 it should be possible to workaround this by leveraging `{build tag}` in the standard python artefact naming scheme:
https://peps.python.org/pep-0427/#file-name-convention

klayoutmatthias, 2022-12-05:
Problem I see is that on PyPI (even test instance) you cannot overwrite files after they have been uploaded once. I had to continuously update `KLAYOUT_PYPI_VERSION` in `version.sh` and I dont't want to spam PyPI with per-commit releases.

I think PyPI should be more generous to allow rolling releases for test purposes. As long as they are not, there will be distinct snapshots and no continuous updates on PyPI.

But maybe the artefact repositories are a way to obtain a current built?

proppy, 2022-12-05:
Apparently there is `--build-number` option you can pass to `bdist_wheel` to override the build tag part of the wheel naming scheme and rev a pypi upload.

I wonder if there is a way to pass that flag down from `cibuildwheel`

klayoutmatthias, 2022-12-05:
Yes, I think that can be done (also patching `version.sh` works). But does it make sense to do that on every commit?

proppy, 2022-12-06:
I this that would be fine for test pypi and keep the production pypi for regular release.

klayoutmatthias, 2022-12-06:
Okay, but wouldn't they mind being spammed by lots of releases?

proppy, 2022-12-06:
Maybe @di could comment if it's fine to "spam" Test PyPI with continuous integration wheels.

thomaslima, 2023-02-13:
It would be technically possible and 'easy' to have a git-based version string for the wheels. But as Matthias said it would cause significant spam -- every commit generates ~300MB of wheels. That's only 33 commits until we reach the project size limit mentioned by @di . I just recommend leaving it as it is. It's useful to test if the twine command is working before tagging a specific commit for the main pypi.

closed 2024-03-23 by klayoutmatthias