KLayout/klayout #1208: LEF with legal DATABASE and UNITs stanza cannot be read

growly, 2022-12-05:
We have a LEF that we believe to be syntactically correct, according to the LEF spec, but which has this `UNITS` stanza at the start:

```
UNITS
    DATABASE MICRONS 1000 ;
END UNITS
```

(The full file is here: [test.lef.zip](https://github.com/KLayout/klayout/files/10154162/test.lef.zip))

Trying to read this in klayout yields an error:
[image]

But we think it shouldn't?
[image]
[image]

This file was written by @dan-fritchman's [Layout21/lef21](https://github.com/dan-fritchman/Layout21/tree/main/lef21) library.

klayoutmatthias, 2022-12-05:
There is a misunderstanding. KLayout does not read LEF files. It reads DEF files, and supports LEF as companion files. There is no "LEF" format support on "File/Open".

The only way currently to import LEF alone is by using "File/Import/LEF". 

LEF-only support is questionable as LEF files often are only meaningful together with other LEF files (technology LEF) and a single LEF file often is not comprehensible without that context. A DEF file in that sense it is top level entity from which reading of a layout makes sense.

Matthias

growly, 2022-12-05:
I see! Thank you for clarifying. I will rephrase the issue.

klayoutmatthias, 2022-12-05:
Very good. I'd like to know what the master of LEF/DEF (@stefanottili) has to say to that :)

stefanottili, 2022-12-07:
The use case to read Macro lef is to be able to compare (dbdiff/xor) IP deliverables.
Unlike reading def there is no need to read technology lef to read macro lef.
So there is no reason not to read the pin and obs geometry and pin text and treat it the same way as layout data as you would gds/oasis/def or drc results.

growly, 2022-12-13:
Is this as simple as adding support for the `DATABASE` unit-spec line for LEFs?

klayoutmatthias, 2022-12-13:
No that is not the issue. LEF reading works. But when you use "File/Open", DEF is expected. LEF is not supported yet.

klayoutmatthias, 2022-12-13:
Did not want to close it.

closed 2022-12-13 by klayoutmatthias