KLayout/klayout #1211: problems with: build.sh :   BIN=`readlink -f $BIN`

sto6, 2022-12-06:
I always comment the build.sh line:   BIN=`readlink -f $BIN`
because I had troubles due to it in my first compile; didn't tested for this issue recently, but this line is still there.

When install actually happens, typically there's a 'mkdir -p $PREFIX', which will mkdir any missing intervening dirs in the path.
Say prefix=/tmp/a/b where /tmp/a does not yet exist. The install-time 'mkdir -p' explicitly handles that. But the 'readlink -f' yields an empty string, which leads to unworkable makefile install rules. The user doesn't notice the issue until after 40 minutes of compile time, at which point it's hard to correct the damage in the Makefiles (without starting over from scratch).

When I give a build -prefix, I consider it sacred: Do not mutilate or mangle. Say when distributed & run it will be under /opt/...; but on build-machine the /opt is setup as a symlink to another mounted disk. If the /opt symlink is resolved (by readlink -> /mnt/...), then the build will only be seen to fail _after_ distribution. (Fortunately IIRC the RPATH already properly does _not_ get so modified [defaults to the pre-readlink-ed BIN]: so that may not be a legitimate concern. But resolving symlinks to paths given to configure a build just raises a red flag for me.)

What if the given prefix is a relative path? Then ok, IFF relative: make it absolute (or flag & abort); but then I'd think RPATH must also be made absolute. In some unavoidable case of must adjust it, if it became empty: flag & abort; don't make garbage Makefiles, nor proceed with compilation.

klayoutmatthias, 2022-12-06:
This is MacOS specific. I cannot comment on that as I am not a Mac user.

klayoutmatthias, 2022-12-06:
Could you check if `readlink -fm $BIN` works for you.

According to the doc, this should render a path even if it does not exist.

Matthias

sto6, 2022-12-07:
Sorry, didn't specify I'm building on centos-7.9 (not macos). The build.sh uses `readlink -f` IFF _not_ on macos.

On linux, `readlink -m` is an _improvement_, in that: It still makes relative paths absolute, and: When more than just final leaf of the input path doesn't exist yet, it does not yield an empty string. I don't think option `-fm` is documented (as to what happens); and in practice the last char (-m) takes precedence so it's equivalent to `readlink -m`.

But I would continue to comment out that line. It still violates the sanctity of "The Build Prefix", by mutate or mangle (resolving symlinks), e.g.:

```
$ ls -ld /opt2
lrwxrwxrwx. 1 root root 9 Dec  7 12:36 /opt2 -> /mnt/vol2
$ BIN=/opt2/klay/0.28
$ BIN=$(readlink -m $BIN)
$ /bin/printf '--> %s <--\n' $BIN
--> /mnt/vol2/klay/0.28 <--
```
If either BIN or RPATH become symlink-resolved value of prefix, and such get embedded in the installed artifacts, where at run-time (on _other_ machines) they are relied upon for locating related content in the same installation... then the build works fine on build machine, but fails in the target deploy environments (where it actually is installed under (non-symlink) /opt2).

If I configure -prefix=/blah (i.e. path already absolute), please do not rewrite this ultimate configuration parameter of the build. thx.

klayoutmatthias, 2022-12-08:
I couldn't trace this feature back further than 2017-02-12 when I switched from SVN to git ...

Anyway, I don't recall the reason it's there. Let's make a deal: you file a PR and I merge it. So I can quote you when the original reason pops up :)

Matthias