KLayout/klayout #1215: Deep mode DRC false positive

egorxe, 2022-12-07:
Deep mode DRC produces false positives on GF180MCU DRC CO.6a rule in latest KLayout. I'm attaching an archive with a minimized GDS from my Open MPW project where error manifests itself and a DRC script with only CO.6a rule left. 

If launched in deep mode KLayout incorrectly finds 7 DRC errors (see co6a_deep.lyrdb):
```
klayout -b -r co6.a.drc -rd input=deep_drc_co6a_fail.gds -rd topcell=user_project_wrapper -rd run_mode=deep -rd report=co6a_deep.lyrdb
```
In flat mode DRC is clean:
```
klayout -b -r co6.a.drc -rd input=deep_drc_co6a_fail.gds -rd topcell=user_project_wrapper -rd run_mode=flat -rd report=co6a_deep.lyrdb
```
[deep_drc_co6a_fail.tar.gz](https://github.com/KLayout/klayout/files/10174313/deep_drc_co6a_fail.tar.gz)

klayoutmatthias, 2022-12-07:
Do you have evidence that this is a KLayout bug? Otherwise please submit an issue for the DRC:

Plus: what is the KLayout version and DRC version (commit). There is a lot of development happening currently.

egorxe, 2022-12-07:
I also would like to note that deletion of some absolutely unrelated GDS parts (even millimeters apart from DRC errors) made errors vanish.

egorxe, 2022-12-08:
Thank you very much for your analysis and superb piece of open silicon software! I'm very sorry for making you sleep deprived, I was not anticipating you to react so promptly. I've operated under assumption that deep mode is kind of hierarchical speedup for large designs and should always produce same results as flat (not like tiled). So when I've found a discrepancy I've reported it in hope that this fairly complex testcase will help with your ongoing effort in improving deep mode to speedup DRC (#1189 and others). Next time I'll hold back from reporting DRC problems until I'll find a root cause myself.

klayoutmatthias, 2022-12-08:
Perfect! But be careful to get enough sleep yourself! :)

My goal is to make deep mode the preferred version and to be (as far as possible) compatible with flat mode, so I am interested in test cases. It's just easier to focus on the real problem than digging through a complex test case. Thanks for your efforts though.

I created smaller version of the test case that shows the issue. It's attached.

The effect are twofold edge pairs causing the problem.

As an interesting fact, the problem seems to be related to #1195. And more interesting, the workaround I proposed for this issue in https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/pull/5 - and which got merged in between - avoids the false positive marker in co6.a. The key is substituting

```
cop6a_cond = metal1.drc( width <= 0.34.um).with_length(0.24.um,nil,both)
```

by

```
cop6a_cond = metal1.width(0.34.um + 1.dbu).with_length(0.24.um, nil, both)
```

So maybe you can try with the DRC from https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr?

Matthias

[small.tar.gz](https://github.com/KLayout/klayout/files/10189515/small.tar.gz)

egorxe, 2022-12-11:
> So maybe you can try with the DRC from https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr?

These rules produce same result on my design in flat and deep mode. So it seems that substituting ```drc( width <= 0.34.um).with_length``` with ```cop6a_cond = metal1.width(0.34.um + 1.dbu).with_length(0.24.um, nil, both)``` really helps.