KLayout/klayout #1219: First macro "stuck" with 0.28

iqmtestd, 2022-12-12:
Hi @klayoutmatthias,

KLayout 0.28 keeps running the first macro even if an other macro is selected. 
To reproduce create two macros run the first one and then try to run the second. Observe the first being executed again:

[image]

This also happens if the first tab is closed. Even after restarting KLayout. Moreover, the **`-`** (Close tab) button does not seem to work at all.

Thanks,
David

klayoutmatthias, 2022-12-12:
Are you using the "run with exclamation mark button"? The "normal run" does not switch the tab. This is intended so you can edit a submodule while running the original (top level) macro.

You're right about "-" - I am always using the tab's "x" button to close, so I did not notice myself.

Matthias

iqmtestd, 2022-12-12:
No, I was using "normal run". It used to work. I've changed the tab manually. You are right, "run with exclamation mark" works.

I think it is confusing that "normal run" runs the first script even if that tab is already closed.

klayoutmatthias, 2022-12-12:
I think I understand your point. So a closed view should not be run and instead the current one should be taken. Fair request.

It's Ok to leave this open. I will fix that issue and the "-" thing.

Matthias

closed 2023-08-09 by klayoutmatthias