KLayout/klayout #1224: Add Python 3.11.1 to PyPi klayout module build

ejprinz, 2022-12-13:
I found that Python 3.11.1 works with the KLayout module but it has been built from source. At some point please add Python 3.11.1 to the PyPi files. This Python version is a bit faster than Python 3.10 and older.

klayoutmatthias, 2022-12-13:
I bumped pypa/cibuildwheel to v2.11.3 which to my understanding should enable Python 3.11. Let's see if that works.

joamatab, 2022-12-21:
Hi Matthias,

i think to make it work for python3.11 you need to trigger a new release

[image]

[image]

https://github.com/gdsfactory/gdsfactory/issues/940

ejprinz, 2022-12-21:
> gdsfactory/gdsfactory#940

Hi Joaquin:
I was able to install klayout in Python 3.11.1 with pip install but I had
to first install two dependencies which I didn't have installed yet (in
Ubuntu 22.04 LTS):
IIRC the packages were: libxml2-dev, libxslt-dev
So check the ERROR messages from pip, and find out which dependency you are
missing. I did a (very complex) test case with Python 3.11.1 and the 0.28
klayout and everything worked as expected.
Best regards, Erwin

klayoutmatthias, 2022-12-25:
With 0.28.2 there are Python 3.11 wheels available for MacOS, Linux and Windows. 

So no need to build from sources anymore :)

I'll close this ticket. If you see issues, feel free to open a new one.

Matthias

closed 2022-12-25 by klayoutmatthias