KLayout/klayout #1228: Add option to have Show parameter names on by default

nikosavola, 2022-12-21:
The _Show parameter names_ is something very useful I didn't think I would need. I would prefer to have it always on, so I suggest adding an option somewhere in Setup to have it on by default. The default behaviour for this should still stay as it is now, i.e., the parameter names are not shown.

klayoutmatthias, 2022-12-25:
I can make this parameter persistent, so you don't need to set it every time you start KLayout. This is easier than some additional configuration parameter.

nikosavola, 2022-12-26:
> I can make this parameter persistent, so you don't need to set it every time you start KLayout. This is easier than some additional configuration parameter.

This seems like a good solution. This issue can be closed after merging https://github.com/KLayout/klayout/pull/1231.

closed 2023-01-04 by nikosavola