KLayout/klayout #1236: Problem with Python 2 build

EugeneZelenko, 2022-12-27:
Hi, Matthias!

I got next compile-time error when I tried to build KLayout with Python 2.7.12:

```
../../../source/src/pya/pya/pyaCallables.cc: In function ‘PyObject* pya::method_adaptor(int, PyObject*, PyObject*)’:
../../../source/src/pya/pya/pyaCallables.cc:666:7: error: ‘Py_RETURN_NOTIMPLEMENTED’ was not declared in this scope
```

klayoutmatthias, 2022-12-28:
Hi Eugene,

Yes, version 0.28 is no longer compatible with Python 2.
Frankly I did not bother any longer about it as Python 2 has reached end of life in April 2020 with version 2.7.18. Even old CentOS 6 ships with Python 3 now.

Do you really need Python 2?

Matthias

klayoutmatthias, 2022-12-28:
I have pushed an experimental commit on the wip branch to restore Python 2 compatibility: https://github.com/KLayout/klayout/commit/ca32b34dff688892a82c29c7f8692cde33ee2034

I cannot however, guarantee Python 2 compatibility in general as support for this version has been removed from the CI platforms in use.

EugeneZelenko, 2022-12-28:
Hi, Matthias!

I think it's fine to drop Python 2 compatibility, but will be good idea to reflect this fact in Release Notes. Otherwise, it's not obvious and happens in the middle of the build. Ideally, build also should fail at beginning.

klayoutmatthias, 2022-12-29:
Fixed.

closed 2022-12-29 by klayoutmatthias