KLayout/klayout #1237: Python error messages should include more information

simonwaid, 2022-12-30:
I'm trying to write a PCell using the integrated editor. 

Of course I make mistakes. From other development tools (eclipse with PyDev) I'm used to get error messages from python that point me the location of the mistake. 

In Klayout I get error messages such as:
```
ERROR: syntax error (class SyntaxError)
ERROR: argument 2 must be sequence of length 4, not 6 (class TypeError)
```

No information about which line is affected. It is very, very painful to debug code like that.

klayoutmatthias, 2022-12-31:
No, that is not true in general:

[image]

So there is a specific situation that will not give you the line number. Please indicate more precisely how to reproduce the problem and I will see what I can do.

lukasc-ubc, 2023-01-16:
@simonwaid maybe the debugging mode is not enabled in your case?

klayoutmatthias, 2023-01-16:
Even outside the debugger, the message should be detailed as shown above.

Either that request refers to an older version or the mentioned behaviour is specific to a certain use case. Specifically during PCell code execution, error messages are handled differently. But I cannot imagine syntax errors happen in that phase - you need to load code before the PCell gets executed. And syntax errors should be seen during that phase already.

Matthias

simonwaid, 2023-01-17:
Here is a minimal example:
```
import pya

class PCellLib(pya.Library)

  def __init__(self):
    pass  
    
PCellLib()
```
The Klayout version is 2.28.2 running on Ubuntu 22.04.

This is the error message I get if I run the script:
[image]

Enabling and disabling the debugging mode does not change anything. The workaround is to copy the code to another editor an run it there.
Unfortunately the code is saved in xml files (.lym) so simply running the script with python is not possible.

klayoutmatthias, 2023-01-17:
@simonwaid Thanks for these pointers. 

The problem seems to be related to Python 3.10.6 which is default on Ubuntu. With the 3.8.10 version I am using for my own builds on Ubuntu 22, everything works as expected:

[image]

This will give me enough pointers to debugging.

Apart from that you don't need to use the XML format. You can use plain text files too (chose "Plain Python file" for the macro template). KLayout will add some pseudo-comments at the beginning to mark such files for auto-execution on startup.

Some users reported to have used KLayout within PyCharm by configuring KLayout as the Python interpreter. I'm not a regular PyCharm user myself, so I cannot give more details as of now.

Matthias

closed 2023-02-02 by klayoutmatthias