KLayout/klayout #1242: KLayout 0.28.2 crashes when registering a plugin if a layout exists

caspar-iqm, 2023-01-02:
The following Python code defines a simple dummy plugin, creates a new layout, and then registers the plugin. Running it crashes the KLayout application since 0.28 (tested to crash on 0.28.2, verified that this worked on the last version of 0.27).

```py
import pya

class DummyPlugin(pya.Plugin):
    def __init__(self, manager, view):
        self.manager = manager
        self.view = view

class DummyPluginFactory(pya.PluginFactory):
    def __init__(self):
        self.register(1000, "dummy_plugin", "Dummy Plugin")
        
    def create_plugin(self, manager, _, view):
        return DummyPlugin(manager, view)

# Create a new layout
main_window = pya.MainWindow.instance()
main_window.create_layout(2)

# Register plugin
DummyPluginFactory()
```

Reversing the order, i.e. calling `DummyPluginFactory()` before `main_window.create_layout(2)`, works fine.

So there is a workaround, just make sure the plugins are registered before any layouts are created. Since this is a hard crash and new behavior in 0.28 I still report it as bug.

klayoutmatthias, 2023-01-03:
Thanks for these creative border cases :) 

I will try to debug and fix it.

Matthias

klayoutmatthias, 2023-01-04:
Problem is found and a fix will be released with 0.28.3.

Thanks again for reporting this problem.

Matthias

caspar-iqm, 2023-01-10:
Thanks a lot!

klayoutmatthias, 2023-01-16:
Should be fixed with 0.28.3

closed 2023-01-16 by klayoutmatthias