KLayout/klayout #1244: Converting one LEFDEF shape to multiple layer datatypes

nayiri-k, 2023-01-09:
I am using Klayout to convert a DEF file to GDS using [this Python script](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/def2stream.py) and [this XML tech file](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/sky130hd.lyt). I want to convert all PORTS in the DEF file to 2 layer datatypes instead of 1, e.g. for each met3 port, generate a met3.pin (70/16) and met3.drawing (70/20) shape. Is this currently possible in KLayout?

I have tried specifying a list in the XML file like below, but only the last element of the list is used.
```
   <produce-pins>true</produce-pins>
   <special-pins-suffix-string>.pin,.drawing</special-pins-suffix-string>
   <special-pins-datatype-string>16,20</special-pins-datatype-string>
```

klayoutmatthias, 2023-01-11:
@nayiri-k I'd recommend to use layer map files which are easier to use than the pattern-based transformation rules:

[image]

In this map file you can specify multiple destination layers, e.g.

```
DIEAREA ALL                              100      0
M1      PIN                              12       2
M1      PIN                              12       20
M1      NET                              12       3
M1      SPNET                            12       4
M1      VIA                              12       5
```

This will put M1.pin on layer 12/2 and 12/20.

Matthias

nayiri-k, 2023-01-11:
Hi Matthias,

Thank you so much for this help!
I hadn't realized Klayout recognized map files, this is exactly what I was trying to achieve. Design is now LVS clean :)

stefanottili, 2023-01-12:
Have a look at the buddy tool strm2gds. You should be able to convert def to gds and merge with leaf cell gds using its command line options.