KLayout/klayout #1249: Include expanded/collapsed state of layer properties into session

EugeneZelenko, 2023-01-13:
It'll be great to expanded/collapsed state of layer properties will be stored in/restored from session, especially when layers properties have several levels of nesting.

EugeneZelenko, 2023-01-20:
[Patch.txt](https://github.com/KLayout/klayout/files/10462934/Patch.txt)

This is my naive implementation of functionality. Python/Ruby API is not supported and most likely is not necessary.

klayoutmatthias, 2023-02-17:
This is kind of crazy ... 

I picked this topic today and implemented a solution before I noticed your patch. And guess what ... my solution is identical to yours, almost up to single characters. So naive? Surely not ...

However, I added the Ruby/Python interface :)

Nice thing is that expanded/collapsed state now gets persisted in .lyp files as well. A side effect which I think is quite useful. And setting the attribute in Ruby/Python will expand/collapse the tree nodes. So that provides a way to control that feature programmatically.

Matthias

EugeneZelenko, 2023-02-17:
Hi, Matthias!

Thank you for taking care about this issue! Sure, I could guess some statements, but Ruby/Python interface is complete mystery for me :-) So your solution is definitely better fit into "common picture"!

klayoutmatthias, 2023-02-17:
But your guess was a pretty good one! :)