KLayout/klayout #1250: Mac DMGs for 0.28.3

Kazzz-S, 2023-01-15:
Dear @klayoutmatthias,

Thank you for the maintenance release.
As usual, different DMG packages of **KLayout 0.28.3 for Mac** are available as described below ([0.28.3 Dropbox Link](https://www.dropbox.com/sh/c8saozbign1hn4f/AAA7WtcBc_KEM03XTTM0PLkpa?dl=0)).
**Please note the following.**<br>
1. Newly supported **macOS Ventura**
2. Restored ST*.dmg using the OS-bundled Python 2.7 for Catalina.
3. Python 3 frameworks for LW*.dmg and HW*.dmg are upgraded from 3.8.x to 3.9.y.

## DMG file naming convention ##
**0.28.3 for Catalina is taken as an example here.**
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:   sys: OS-bundled; mp31: Ruby3.1 from MacPorts; hb31: Ruby3.1 from Homebrew; ana3: Ruby3.1 from Anaconda3
(f) **P**ython type: sys: OS-bundled; mp39: Python 3.9 from MacPorts; hb39: Python 3.9 from Homebrew; ana3: Python3.9 from Anaconda3

## For the primary platform, macOS Catalina (10.15.7) ##
Five macOS DMGs for **Catalina** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.0 on Monterey (12.6.2).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|Catalina (10.15)|ST-klayout-0.28.3-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(1)|
2|Catalina (10.15) with MacPorts dev. env.|LW-klayout-0.28.3-macOS-Catalina-1-qt5MP-Rmp31Pmp39.dmg|MacPorts|MacPorts|MacPorts|(2)|
3|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.28.3-macOS-Catalina-1-qt5Brew-Rhb31Phb39.dmg|Homebrew|Homebrew|Homebrew|(3)|
4|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.28.3-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(4)|
5|Catalina (10.15)|HW-klayout-0.28.3-macOS-Catalina-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(5)|

(1) shares OS-bundled Ruby2.6 and Python-2.7; Qt5 from MacPorts is embedded, which is the standard (ST-).
(2) **experimental**; shares MacPorts development environment; for those with the matching dev. env.
(3) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(4) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(5) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For macOS Big Sur (11.7.2) ##
Four macOS DMGs for **Big Sur** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.0 on Monterey (12.6.2).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
21a|~~BigSur (11.7)~~|~~ST-klayout-0.28.3-macOS-BigSur-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(21a)|
21b|Catalina (10.15)|ST-klayout-0.28.3-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(21b)|
22|BigSur (11.7) with MacPorts dev. env.|LW-klayout-0.28.3-macOS-BigSur-1-qt5MP-Rmp31Pmp39.dmg|MacPorts|MacPorts|MacPorts|(22)|
23|BigSur (11.7) with Homebrew dev. env.|LW-klayout-0.28.3-macOS-BigSur-1-qt5Brew-Rhb31Phb39.dmg|Homebrew|Homebrew|Homebrew|(23)|
24|BigSur (11.7) with Anaconda3 dev. env.|LW-klayout-0.28.3-macOS-BigSur-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(24)|
25|BigSur (11.7)|HW-klayout-0.28.3-macOS-BigSur-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(25)|

(21a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.1 does not link the legacy Python 2.7 library.
(21b) However, ST*.dmg for Catalina, which shares OS-bundled Ruby2.6 and Python-2.7, and Qt5 from MacPorts is embedded, can run on Big Sur.
(22) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(23) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(24) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(25) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For macOS Monterey (12.6.2) ##
Four macOS DMGs for **Monterey** are available.

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
31a|~~Monterey (12.6)~~|~~ST-klayout-0.28.3-macOS-Monterey-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31a)|
31b|~~Catalina (10.15)~~|~~ST-klayout-0.28.3-macOS-Catalina-1-qt5MP-RsysPsys.dmg~~|~~MacPorts~~|~~OS-bundled~~|~~OS-bundled~~|(31b)|
32|Monterey (12.6) with MacPorts dev. env.|LW-klayout-0.28.3-macOS-Monterey-1-qt5MP-Rmp31Pmp39.dmg|MacPorts|MacPorts|MacPorts|(32)|
33|Monterey (12.6) with Homebrew dev. env.|LW-klayout-0.28.3-macOS-Monterey-1-qt5Brew-Rhb31Phb39.dmg|Homebrew|Homebrew|Homebrew|(33)|
34|Monterey (12.6) with Anaconda3 dev. env.|LW-klayout-0.28.3-macOS-Monterey-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(34)|
35|Monterey (12.6)|HW-klayout-0.28.3-macOS-Monterey-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(35)|

(31a) **NOT PROVIDED**; The ST-* package is no longer provided since Xcode >= 13.1 does not link the legacy Python 2.7 library.
(31b) **Monterey 12.3.1 finally eliminated Python-2.7 binary. Hence, ST-\* package no longer runs.**
(32) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(33) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(34) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(35) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For the latest macOS Ventura (13.1) ##
Four macOS DMGs for **Ventura** are available.
They were built on an Intel-CPU-based (not M1 chip) VMware Fusion 13.0.0 on Monterey (12.6.2).

**LW- and HW- packages use Python 3.9 from MacPorts, Homebrew, or Anaconda3.**
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
41|Ventura (13.1) with MacPorts dev. env.|LW-klayout-0.28.3-macOS-Ventura-1-qt5MP-Rmp31Pmp39.dmg|MacPorts|MacPorts|MacPorts|(41)|
42|Ventura (13.1) with Homebrew dev. env.|LW-klayout-0.28.3-macOS-Ventura-1-qt5Brew-Rhb31Phb39.dmg|Homebrew|Homebrew|Homebrew|(42)|
43|Ventura (13.1) with Anaconda3 dev. env.|LW-klayout-0.28.3-macOS-Ventura-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(43)|
44|Ventura (13.1)|HW-klayout-0.28.3-macOS-Ventura-1-qt5Brew-RsysPhb39.dmg|Homebrew|OS-bundled|Homebrew|(44)|

(41) **experimental**; shares MacPorts development environment; for those who have the matching dev. env.
(42) **experimental**; shares Homebrew development environment; for those who have the matching dev. env.
(43) **experimental**; shares Anaconda3 development environment; for those who have the matching dev. env.
(44) **experimental**; shares OS-bundled Ruby2.6; Qt5 and Python-3.9 from Homebrew are embedded, hence, heavy.

## For previous OS ##
|Srl.No.|Target OS|DMG file name|Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Mojave (10.14)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
7|HighSierra (10.13)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
8|Sierra (10.12)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
9|ElCapitan (10.11)|From 0.27.x onward, DMGs will not be provided for this OS.||||(1p)|
x|Yosemite (10.10)|has been dropped due to the compiler issue reported in #360.|

(1p) I expect that build is still possible.

Please close this ticket after publishing the packages. 
With warm regards,
Kazzz-S

closed 2023-01-16 by klayoutmatthias

lukasc-ubc, 2023-01-16:
@Kazzz-S 
I am seeing a strange effect. When I create a new layout using a specific technology, the technology is not shown next to the T, and libraries are not avaialble.  When I close the layout, the technology is then displayed.

New layout, with [EBeam](https://github.com/SiEPIC/SiEPIC_EBeam_PDK) technology:
[image]

After closing the layout, T=EBeam shows up:
[image]

I create the new layout again, the T= empty, yet the Layout Properties show EBeam.
[image]

Changing the technology changes the layer table, but not the T on the bottom left.
[image]

Interestingly, if I create the layout using a [Python script](https://github.com/SiEPIC/SiEPIC_EBeam_PDK/blob/master/klayout_dot_config/tech/EBeam/pymacros/Example_scripted_layouts/example_Ring_resonator_sweep.lym), it works:
```
mw = pya.Application().instance().main_window()
ly = mw.create_layout('EBeam', 1).layout()
```
[image]

The above problem didn't exist in 0.27.11.  I haven't checked with the other 0.28 versions, but I can do that if you can't reproduce the problem.

lukasc-ubc, 2023-01-16:
Also to add, there is a work-around.

If there is a library present that is generic and not associated with a specific technology (SiEPIC General in this case):
[image]

Then when I switch to that library, things magically correct themselves:
[image]

klayoutmatthias, 2023-01-16:
Hi Lukas,

I can reproduce the issue with the library browser. Apparently if there are less than 2 regular libraries, only the Basic one is shown. I have created a separate ticket for this: https://github.com/KLayout/klayout/issues/1253.

Regarding the "T" label: on Windows and Linux, the T label follows the current technology. That appears to be Mac specific. It does not vanish when closing the layout, which is a flaw. I have created another follow-up ticket: https://github.com/KLayout/klayout/issues/1254

Best regards, 

Matthias