KLayout/klayout #1254: MacOS: "T" label not updated when changing the technology

klayoutmatthias, 2023-01-16:
This is a follow-up ticket for the issue mentioned by @lukasc-ubc here: https://github.com/KLayout/klayout/issues/1250.

klayoutmatthias, 2023-01-24:
I have some feeling that this problem is going to be fixed together with #1253, but that needs confirmation.

Kazzz-S, 2023-01-25:
I've merged ab8cfee1 to 1d8568e4, and rebuilt two DMGs with Homebrew in Monterey. Files are [here](https://www.dropbox.com/sh/chzycnydbhwu2p3/AACddSZfRRc09HFP0OmLqCkGa?dl=0)(Dropbox link).

[image]

[image]

[image]

[image]

[image]

[image]

lukasc-ubc, 2023-01-25:
Tried it and the problem persists.  T=empty, then I select the 2nd library, and then T=EBeam and the libraries show up.

klayoutmatthias, 2023-01-26:
Hi Lukas,

thanks for that feedback. 

I could reproduce and debug the problem on my Catalina VirtualBox instance. After fixing some events that had not been sent, I think the problem should be solved now. I tried number of different ways (load layouts, create new layouts, replace layouts etc.) and I could no longer see the issue.

@kazzz maybe you can help (a last time I hope) and try out the issue-1259 branch. If that works, I'll release it with 0.28.4.

Thanks and best regards,

Matthias

closed 2023-02-05 by klayoutmatthias