KLayout/klayout #1256: libklayout_db.so.0 cannot open shared object file: No such file or directory

tristonw109, 2023-01-19:
klayout was built locally with the following command:

`./build.sh -prefix /usr/bin
`

when using klayout with [OpenROAD Project](https://openroad-flow-scripts.readthedocs.io/en/latest/index.html) I get the following error:

```[INFO][FLOW] Using platform directory ./platforms/sky130hd
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /usr/bin/klayout -zz -rd design_name=ibex_core \
        -rd in_def=results/sky130hd/ibex/base/6_final.def \
        -rd in_files="./platforms/sky130hd/gds/sky130_fd_sc_hd.gds  " \
        -rd config_file=./platforms/sky130hd/fill.json \
        -rd seal_file="" \
        -rd out_file=results/sky130hd/ibex/base/6_1_merged.gds \
        -rd tech_file=./objects/sky130hd/ibex/base/klayout.lyt \
        -rd layer_map= \
        -r ./util/def2stream.py) 2>&1 | tee ./logs/sky130hd/ibex/base/6_1_merge.log
/usr/bin/klayout: error while loading shared libraries: libklayout_db.so.0: cannot open shared object file: No such file or directory
Command exited with non-zero status 127
Elapsed time: 0:00.00[h:]min:sec. CPU time: user 0.00 sys 0.00 (100%). Peak memory: 4300KB.
make: *** [Makefile:685: results/sky130hd/ibex/base/6_1_merged.gds] Error 127
```

however, libklayout_db.so.0 seems to be in `/usr/bin/` based on the output below

```
triston@FrameworkLinux:/usr/bin$ ls | grep libklayout_db.so.0

libklayout_db.so.0
libklayout_db.so.0.27
libklayout_db.so.0.27.1
```
any ideas about what the issue maybe?

Kazzz-S, 2023-01-19:
Does #1165 help?

Kazzz-S, 2023-01-19:
Can you start KLayout in a GUI mode by typing `/usr/bin/klayout`? 
If not, there is an issue in loading shared libraries.
In that case, as explained by Matthias in #1165,  you can specify an RPATH using the "-rpath" option in the build script.
Then you need to rebuild KLayout.

The alternative, the Bash script I use, can take arguments by the last `$*`.
I suggest using the `klayout.sh` script instead of `the klayout` command.
Let's take an example from https://www.klayout.de/forum/discussion/2227/convert-many-box-objects-to-circles-efficiently.

**PYA script to run: `myscript.py`**
```
# https://www.klayout.de/forum/discussion/2227/convert-many-box-objects-to-circles-efficiently
#
import numpy as np
import pya

def make_cir(r):
  radius = r/2
  nr_points = 32
  angles = np.linspace(0,2*np.pi,nr_points+1)[0:-1]
  # "List Comprehensions" must be faster than list.append()
  points = [ pya.DPoint(radius*np.cos(angle),radius*np.sin(angle)) for ind,angle in enumerate(angles) ]
  circle = pya.DSimplePolygon(points)
  return circle

circle = make_cir(10.0)
print( "# Number of vetices = %d" % circle.num_points() )
```

**Execution**
```
(base) sekigawa@LM19:~/zzz (1)$ which klayout.sh
/home/sekigawa/bin/klayout.sh

(base) sekigawa@LM19:~/zzz (2)$ klayout.sh -z -r myscript.py 
# Number of vetices = 32
```

tristonw109, 2023-01-19:
> Can you start KLayout in a GUI mode by typing `/usr/bin/klayout`? If not, there is an issue in loading shared libraries. In that case, as explained by Matthias in #1165, you can specify an RPATH [...]

I tried to running klayout in  GUI mode the result was

```
/usr/bin/klayout: error while loading shared libraries: libklayout_db.so.0: cannot open shared object file: No such file or directory
```
so I tried to rebuild with the following command

`./build.sh -prefix /usr/bin/ -rpath /usr/bin/
`
still end up with the shared libraries error

Kazzz-S, 2023-01-20:
It's weird :-(
I've rebuilt KLayout from scratch in the same way as you.

```
#-------------------------------------------------------------------------
# [1] Build
#-------------------------------------------------------------------------
<MyHostName>{kazzz-s}(1)$ uname -a
Linux <MyHostName> 5.4.0-137-generic #154-Ubuntu SMP Thu Jan 5 17:03:22 \
      UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

<MyHostName>{kazzz-s}(2)$ ./build.sh \
                          -prefix $HOME/opt/klayout \
                          -rpath  $HOME/opt/klayout \
                          -option -j7
:
: -- patience...---
:
:
Build successfully done.
Artefacts were installed to /home/kazzz-s/opt/klayout  <=== as expected

<MyHostName>{kazzz-s}(3)$ pwd
/home/kazzz-s/opt/klayout

<MyHostName>{kazzz-s}(4)$ ll -rt | grep klayout
:
:
:
    0 lrwxrwxrwx 1 kazzz-s kazzz-s       26 Jan 20 09:17 libklayout_icons.so.0 -> libklayout_icons.so.0.28.3*
    0 lrwxrwxrwx 1 kazzz-s kazzz-s       26 Jan 20 09:17 libklayout_icons.so.0.28 -> libklayout_icons.so.0.28.3*
  452 -rwxr-xr-x 1 kazzz-s kazzz-s   459568 Jan 20 09:17 libklayout_bd.so.0.28.3*
    0 lrwxrwxrwx 1 kazzz-s kazzz-s       23 Jan 20 09:17 libklayout_bd.so -> libklayout_bd.so.0.28.3*
    0 lrwxrwxrwx 1 kazzz-s kazzz-s       23 Jan 20 09:17 libklayout_bd.so.0.28 -> libklayout_bd.so.0.28.3*
    0 lrwxrwxrwx 1 kazzz-s kazzz-s       23 Jan 20 09:17 libklayout_bd.so.0 -> libklayout_bd.so.0.28.3*
   32 -rwxr-xr-x 1 kazzz-s kazzz-s    31176 Jan 20 09:17 klayout*  <=== here it is!

#-------------------------------------------------------------------------
# [2] Confirm the value of RUNPATH embedded in "klayout"
#-------------------------------------------------------------------------
<MyHostName>{kazzz-s}(5)$ objdump -x klayout | grep -E 'R*PATH'
  RUNPATH              /home/kazzz-s/opt/klayout  <=== as expected

#-------------------------------------------------------------------------
# [3] Confirm that LD_LIBRARY_PATH is NOT set
#-------------------------------------------------------------------------
<MyHostName>{kazzz-s}(6)$ echo $LD_LIBRARY_PATH
<=== empty line

#-------------------------------------------------------------------------
# [4] Invoke KLayout from the installation directory with a relative path
#-------------------------------------------------------------------------
<MyHostName>{kazzz-s}(7)$ pwd
/home/kazzz-s/opt/klayout

<MyHostName>{kazzz-s}(8)$ ./klayout -v
KLayout 0.28.3

<MyHostName>{kazzz-s}(9)$ ./klayout
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
+--------------------------+
|                          |
|      GUI shows up        |
|                          |
+--------------------------+

#-------------------------------------------------------------------------
# [5] Invoke KLayout from the root directory with a full path
#-------------------------------------------------------------------------
<MyHostName>{kazzz-s}(10)$ pwd
/

<MyHostName>{kazzz-s}(11)$ $HOME/opt/klayout/klayout -v
KLayout 0.28.3

<MyHostName>{kazzz-s}(12)$ $HOME/opt/klayout/klayout
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
+--------------------------+
|                          |
|      GUI shows up        |
|                          |
+--------------------------+

```

One fundamental doubt.
The `/usr/bin/` directory is usually owned by the root and not write-permissible to ordinary users.
Could you successfully install the binaries there?
Or did you run the build command with `sudo`?

By default, the build directory, which is write-permissible, is `build-release/`, as shown below.

```
<MyHostName>{kazzz-s}(1)$ pwd
/home/kazzz-s/GitWork/klayout

<MyHostName>{kazzz-s}(2)$ ll -rt | grep -E 'build*'
   0 lrwxrwxrwx  1 kazzz-s kazzz-s      23 Nov  8 12:39 makeDMG4mac.py -> macbuild/makeDMG4mac.py*
   0 lrwxrwxrwx  1 kazzz-s kazzz-s      21 Nov  8 12:39 build4mac.py -> macbuild/build4mac.py*
   4 -rwxrwxr-x  1 kazzz-s kazzz-s     630 Nov  8 12:39 travis-build.sh*
  12 -rw-rw-r--  1 kazzz-s kazzz-s    9285 Jan 18 08:37 build.bat
  24 -rwxrwxr-x  1 kazzz-s kazzz-s   20500 Jan 18 08:37 build.sh*       <=== the build script
   4 drwxrwxr-x  3 kazzz-s kazzz-s    4096 Jan 18 08:37 macbuild/
  12 drwxrwxr-x 30 kazzz-s kazzz-s   12288 Jan 20 09:16 build-release/  <=== build (not install) directory
```

tristonw109, 2023-01-20:
I did prepend `sudo` when building. I am thinking about deleting everything and starting over. I will report what happens

tristonw109, 2023-01-21:
The issue was on the flow side and not on Klayout's side.

closed 2023-01-21 by tristonw109