KLayout/klayout #1265: Mac OSX, Python: import klayout.dbcore, missing libpng16.16.dylib

lukasc-ubc, 2023-01-25:
@Kazzz-S 

The KLayout Python package is referencing a library "libpng16.16.dylib" that is not installed on my system. I understand that it can be installed by brew.  Perhaps this library could be bundled as part of the "pip install" klayout package?

Thank you

```
(base) iMac-Hickup:anaconda3 lukasc$ pip install klayout
Collecting klayout
  Downloading klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl (16.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.7/16.7 MB 10.4 MB/s eta 0:00:00
Installing collected packages: klayout
Successfully installed klayout-0.28.3
(base) iMac-Hickup:anaconda3 lukasc$ python 
Python 3.9.13 (main, Aug 25 2022, 18:29:29) 
[Clang 12.0.0 ] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
im>>> import klayout.dbcore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/lukasc/opt/anaconda3/lib/python3.9/site-packages/klayout/dbcore.cpython-39-darwin.so, 2): Library not loaded: /usr/local/opt/libpng/lib/libpng16.16.dylib
  Referenced from: /Users/lukasc/opt/anaconda3/lib/python3.9/site-packages/klayout/lib_tl.cpython-39-darwin.so
  Reason: image not found

```

Kazzz-S, 2023-01-26:
Hi @lukasc-ubc,

```
(base) iMac-Hickup:anaconda3 lukasc$ pip install klayout
Collecting klayout
  Downloading klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl (16.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.7/16.7 MB 10.4 MB/s eta 0:00:00
Installing collected packages: klayout
Successfully installed klayout-0.28.3
(base) iMac-Hickup:anaconda3 lukasc$ python
Python 3.9.13 (main, Aug 25 2022, 18:29:29)
[Clang 12.0.0 ] :: Anaconda, Inc. on darwin
                   ^^^^^^^^^^^^^^
Type "help", "copyright", "credits" or "license" for more information.
im>>> import klayout.dbcore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dlopen(/Users/lukasc/opt/anaconda3/lib/python3.9/site-packages/klayout/dbcore.cpython-39-darwin.so,
 2): Library not loaded: /usr/local/opt/libpng/lib/libpng16.16.dylib  <===== (1)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  Referenced from: /Users/lukasc/opt/anaconda3/lib/python3.9/site-packages/klayout/lib_tl.cpython-39-darwin.so
  Reason: image not found
```

(1) It looks like Homebrew's libpng16.16.dylib is being loaded. But your current environment is Anaconda3.

----

In my Anaconda3 environment on Monterey...

```
(base) MacBookPro2{sekigawa} ~ (1)% which pip
/Applications/anaconda3/bin/pip

(base) MacBookPro2{sekigawa} ~ (2)% pip list | grep -E '(klay|png)'
klayout                       0.28.3 <===== (2)

(base) MacBookPro2{sekigawa} ~ (3)% which python3
/Applications/anaconda3/bin/python3

(base) MacBookPro2{sekigawa} ~ (4)% python3
Python 3.9.15 (main, Nov  4 2022, 11:11:31)
[Clang 12.0.0 ] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.dbcore
    ^^^^^^^^^^^^^^^^^^^^^  No errors
>>> help(klayout.dbcore)

Help on module klayout.dbcore in klayout:

NAME
    klayout.dbcore - KLayout core module 'db'

CLASSES
    builtins.object
        Box
        Cell
        CellInstArray
        CellMapping
        CompoundRegionOperationNode
        Connectivity
        CplxTrans
        DBox
        DCellInstArray
        DCplxTrans
        DEdge
        DEdgePair
        DPath
        DPoint
        DPolygon
[... 98 lines cut ...]
(3) This "libpng16.16.dylib" is Anaconda3's, which was installed by default.

```
(base) MacBookPro2{sekigawa} lib (1)% pwd
/Applications/anaconda3/lib

(base) MacBookPro2{sekigawa} lib (2)% ll | grep png
lrwxr-xr-x    1 sekigawa wheel        10 Dec 30 23:26 libpng.a -> libpng16.a
lrwxr-xr-x    1 sekigawa wheel        17 Dec 30 23:26 libpng.dylib -> libpng16.16.dylib
-rwxrwxr-x    1 sekigawa wheel    221836 Dec 30 23:26 libpng16.16.dylib <===== (3)
-rw-rw-r--    2 sekigawa wheel    305384 Apr 24  2019 libpng16.a
lrwxr-xr-x    1 sekigawa wheel        17 Dec 30 23:26 libpng16.dylib -> libpng16.16.dylib
-rwxrwxr-x    1 sekigawa wheel    247616 Dec 30 23:26 libzopflipng.1.0.3.dylib
lrwxr-xr-x    1 sekigawa wheel        24 Dec 30 23:26 libzopflipng.1.dylib -> libzopflipng.1.0.3.dylib
lrwxr-xr-x    1 sekigawa wheel        24 Dec 30 23:26 libzopflipng.dylib -> libzopflipng.1.0.3.dylib

(base) MacBookPro2{sekigawa} lib (3)% otool -L libpng16.16.dylib
libpng16.16.dylib:
  @rpath/libpng16.16.dylib (compatibility version 54.0.0, current version 54.0.0)
  @rpath/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
  /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
```

Regards,
Kazzz-S

Kazzz-S, 2023-01-26:
## In the Homebrew environment... ##

```
MacBookPro2{sekigawa} ~ (1)% which pip3
/usr/local/opt/python@3.9/bin/pip3

MacBookPro2{sekigawa} ~ (2)% pip3 list | grep -E '(klay|png)'
klayout         0.28.3

MacBookPro2{sekigawa} ~ (3)% which python3
/usr/local/opt/python@3.9/bin/python3

MacBookPro2{sekigawa} ~ (4)% python3
Python 3.9.16 (main, Dec  7 2022, 10:15:43)
[Clang 14.0.0 (clang-1400.0.29.202)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.dbcore
    ^^^^^^^^^^^^^^^^^^^^^ No errors
>>> help(klayout.dbcore)

Help on module klayout.dbcore in klayout:

NAME
    klayout.dbcore - KLayout core module 'db'

CLASSES
    builtins.object
        Box
        Cell
        CellInstArray
        CellMapping
        CompoundRegionOperationNode
        Connectivity
        CplxTrans
        DBox
        DCellInstArray
        DCplxTrans
        DEdge
        DEdgePair
        DPath
        DPoint
        DPolygon
        DSimplePolygon
        DText
        DTrans
        DVector
        DeepShapeStore
        DeviceAbstract
        DeviceAbstractRef
        DeviceClass
            DeviceClassBJT3Transistor
                DeviceClassBJT4Transistor
            DeviceClassCapacitor
                DeviceClassCapacitorWithBulk
            DeviceClassDiode
            DeviceClassInductor
            DeviceClassMOS3Transistor
                DeviceClassMOS4Transistor
            DeviceClassResistor
                DeviceClassResistorWithBulk
        DeviceClassFactory
        DeviceExtractorBase
            DeviceExtractorBJT3Transistor
                DeviceExtractorBJT4Transistor
:
:
:
```

----

```
MacBookPro2{sekigawa} klayout (1)% pwd
/usr/local/opt/python@3.9/Frameworks/Python.framework/Versions/Current/lib/python3.9/site-packages/klayout

MacBookPro2{sekigawa} klayout (2)% ll
total 60692
drwxr-xr-x 33 sekigawa admin     1056 Jan 15 08:04 .
drwxr-xr-x 46 sekigawa admin     1472 Jan 15 08:04 ..
-rw-r--r--  1 sekigawa admin       34 Jan 15 08:04 __init__.py
drwxr-xr-x  3 sekigawa admin       96 Jan 15 08:04 __pycache__
drwxr-xr-x  5 sekigawa admin      160 Jan 15 08:04 db
drwxr-xr-x 10 sekigawa admin      320 Jan 15 08:04 db_plugins
-rwxr-xr-x  1 sekigawa admin    36648 Jan 15 08:04 dbcore.cpython-39-darwin.so
-rw-r--r--  1 sekigawa admin  2376826 Jan 15 08:04 dbcore.pyi
drwxr-xr-x  4 sekigawa admin      128 Jan 15 08:04 lay
-rwxr-xr-x  1 sekigawa admin    36632 Jan 15 08:04 laycore.cpython-39-darwin.so
-rw-r--r--  1 sekigawa admin   585462 Jan 15 08:04 laycore.pyi
drwxr-xr-x  4 sekigawa admin      128 Jan 15 08:04 lib
-rwxr-xr-x  1 sekigawa admin   822368 Jan 15 08:04 lib_ant.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin 36299944 Jan 15 08:04 lib_db.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  1227800 Jan 15 08:04 lib_edt.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  1255296 Jan 15 08:04 lib_gsi.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  1086184 Jan 15 08:04 lib_img.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin 12106800 Jan 15 08:04 lib_laybasic.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin   255128 Jan 15 08:04 lib_layview.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin   372080 Jan 15 08:04 lib_lib.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin   635912 Jan 15 08:04 lib_lym.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  2068888 Jan 15 08:04 lib_pya.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin    43272 Jan 15 08:04 lib_rba.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  1523696 Jan 15 08:04 lib_rdb.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin  1073824 Jan 15 08:04 lib_tl.cpython-39-darwin.so
-rwxr-xr-x  1 sekigawa admin    36440 Jan 15 08:04 libcore.cpython-39-darwin.so
-rw-r--r--  1 sekigawa admin      151 Jan 15 08:04 libcore.pyi
drwxr-xr-x  4 sekigawa admin      128 Jan 15 08:04 rdb
-rwxr-xr-x  1 sekigawa admin    36304 Jan 15 08:04 rdbcore.cpython-39-darwin.so
-rw-r--r--  1 sekigawa admin    71673 Jan 15 08:04 rdbcore.pyi
drwxr-xr-x  4 sekigawa admin      128 Jan 15 08:04 tl
-rwxr-xr-x  1 sekigawa admin    36080 Jan 15 08:04 tlcore.cpython-39-darwin.so
-rw-r--r--  1 sekigawa admin   118783 Jan 15 08:04 tlcore.pyi

MacBookPro2{sekigawa} klayout (3)% otool -L lib_tl.cpython-39-darwin.so | grep png
    /usr/local/opt/libpng/lib/libpng16.16.dylib (compatibility version 56.0.0, current version 56.0.0)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

MacBookPro2{sekigawa} lib (4)% pwd
/usr/local/opt/libpng/lib

MacBookPro2{sekigawa} lib (5)% ll
total 436
drwxr-xr-x  8 sekigawa admin    256 Nov 21 06:53 .
drwxr-xr-x 13 sekigawa admin    416 Nov 25 06:45 ..
lrwxr-xr-x  1 sekigawa admin     10 Nov 21 06:53 libpng.a -> libpng16.a
lrwxr-xr-x  1 sekigawa admin     14 Nov 21 06:53 libpng.dylib -> libpng16.dylib
-rw-r--r--  1 sekigawa admin 189368 Nov 25 06:45 libpng16.16.dylib
-rw-r--r--  1 sekigawa admin 253560 Nov 21 06:53 libpng16.a
lrwxr-xr-x  1 sekigawa admin     17 Nov 21 06:53 libpng16.dylib -> libpng16.16.dylib
drwxr-xr-x  4 sekigawa admin    128 Nov 25 06:45 pkgconfig

```

Kazzz-S, 2023-01-26:
## In the MacPorts environment... ##

```
MacBookPro2{sekigawa} ~ (1)% which pip3
/opt/local/bin/pip3

MacBookPro2{sekigawa} ~ (2)% pip3 list | grep -E '(klay|png)'
klayout           0.28.3

MacBookPro2{sekigawa} ~ (3)% which python3
/opt/local/bin/python3

MacBookPro2{sekigawa} ~ (4)% python3
Python 3.9.16 (main, Dec  7 2022, 02:41:07)
[Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.dbcore
>>> help(klayout.dbcore)
    ^^^^^^^^^^^^^^^^^^^^^ No errors
>>> help(klayout.dbcore)

Help on module klayout.dbcore in klayout:

NAME
    klayout.dbcore - KLayout core module 'db'

CLASSES
    builtins.object
        Box
        Cell
        CellInstArray
        CellMapping
        CompoundRegionOperationNode
        Connectivity
        CplxTrans
        DBox
        DCellInstArray
        DCplxTrans
        DEdge
        DEdgePair
        DPath
        DPoint
        DPolygon
        DSimplePolygon
        DText
        DTrans
        DVector
        DeepShapeStore
        DeviceAbstract
        DeviceAbstractRef
        DeviceClass
            DeviceClassBJT3Transistor
                DeviceClassBJT4Transistor
            DeviceClassCapacitor
                DeviceClassCapacitorWithBulk
            DeviceClassDiode
            DeviceClassInductor
            DeviceClassMOS3Transistor
                DeviceClassMOS4Transistor
            DeviceClassResistor
                DeviceClassResistorWithBulk
        DeviceClassFactory
        DeviceExtractorBase
            DeviceExtractorBJT3Transistor
                DeviceExtractorBJT4Transistor
:
:
:
```

----

```
MacBookPro2{sekigawa} klayout (1)% pwd
/opt/local/Library/Frameworks/Python.framework/Versions/Current/lib/python3.9/site-packages/klayout

MacBookPro2{sekigawa} klayout (2)% ll
total 51496
drwxr-xr-x 33 root wheel     1056 Jan 15 08:04 .
drwxr-xr-x 53 root wheel     1696 Jan 15 08:04 ..
-rw-r--r--  1 root wheel       34 Jan 15 08:04 __init__.py
drwxr-xr-x  3 root wheel       96 Jan 15 08:04 __pycache__
drwxr-xr-x  5 root wheel      160 Jan 15 08:04 db
drwxr-xr-x 10 root wheel      320 Jan 15 08:04 db_plugins
-rwxr-xr-x  1 root wheel    36808 Jan 15 08:04 dbcore.cpython-39-darwin.so
-rw-r--r--  1 root wheel  2376826 Jan 15 08:04 dbcore.pyi
drwxr-xr-x  4 root wheel      128 Jan 15 08:04 lay
-rwxr-xr-x  1 root wheel    36632 Jan 15 08:04 laycore.cpython-39-darwin.so
-rw-r--r--  1 root wheel   585462 Jan 15 08:04 laycore.pyi
drwxr-xr-x  4 root wheel      128 Jan 15 08:04 lib
-rwxr-xr-x  1 root wheel   724384 Jan 15 08:04 lib_ant.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel 28863104 Jan 15 08:04 lib_db.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel  1098504 Jan 15 08:04 lib_edt.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel  1107688 Jan 15 08:04 lib_gsi.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel   922744 Jan 15 08:04 lib_img.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel 11181016 Jan 15 08:04 lib_laybasic.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel   228768 Jan 15 08:04 lib_layview.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel   319816 Jan 15 08:04 lib_lib.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel   560840 Jan 15 08:04 lib_lym.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel  2088888 Jan 15 08:04 lib_pya.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel    43384 Jan 15 08:04 lib_rba.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel  1331664 Jan 15 08:04 lib_rdb.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel   885072 Jan 15 08:04 lib_tl.cpython-39-darwin.so
-rwxr-xr-x  1 root wheel    36440 Jan 15 08:04 libcore.cpython-39-darwin.so
-rw-r--r--  1 root wheel      151 Jan 15 08:04 libcore.pyi
drwxr-xr-x  4 root wheel      128 Jan 15 08:04 rdb
-rwxr-xr-x  1 root wheel    36304 Jan 15 08:04 rdbcore.cpython-39-darwin.so
-rw-r--r--  1 root wheel    71673 Jan 15 08:04 rdbcore.pyi
drwxr-xr-x  4 root wheel      128 Jan 15 08:04 tl
-rwxr-xr-x  1 root wheel    36088 Jan 15 08:04 tlcore.cpython-39-darwin.so
-rw-r--r--  1 root wheel   118783 Jan 15 08:04 tlcore.pyi

MacBookPro2{sekigawa} klayout (3)% otool -L lib_tl.cpython-39-darwin.so | grep png
    /opt/local/lib/libpng16.16.dylib (compatibility version 56.0.0, current version 56.0.0)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

MacBookPro2{sekigawa} lib (4)% pwd
/opt/local/lib

MacBookPro2{sekigawa} lib (5)% ll | grep png
lrwxr-xr-x   1 root     wheel        10 Dec  7 01:26 libpng.a -> libpng16.a
lrwxr-xr-x   1 root     wheel        14 Dec  7 01:26 libpng.dylib -> libpng16.dylib
-rwxr-xr-x   1 root     wheel    189464 Dec  7 01:26 libpng16.16.dylib
-rw-r--r--   1 root     wheel    251760 Dec  7 01:26 libpng16.a
lrwxr-xr-x   1 root     wheel        17 Dec  7 01:26 libpng16.dylib -> libpng16.16.dylib

```

Kazzz-S, 2023-01-26:
Oh, I understand!

## Again in the Anaconda3 environment ##

```
(base) MacBookPro2{sekigawa} ~ (1)% which pip3
/Applications/anaconda3/bin/pip3

(base) MacBookPro2{sekigawa} ~ (2)% pip3 uninstall klayout <===== Uninstall the existing "pymod"
Found existing installation: klayout 0.28.3
Uninstalling klayout-0.28.3:
  Would remove:
    /Applications/anaconda3/lib/python3.9/site-packages/klayout-0.28.3.dist-info/*
    /Applications/anaconda3/lib/python3.9/site-packages/klayout/*
    /Applications/anaconda3/lib/python3.9/site-packages/pya/*
Proceed (Y/n)? Y
  Successfully uninstalled klayout-0.28.3

(base) MacBookPro2{sekigawa} ~ (3)% pip3 list | grep kl
cloudpickle                   2.0.0
mkl-fft                       1.3.1
mkl-random                    1.2.2
mkl-service                   2.4.0
pickleshare                   0.7.5

(base) MacBookPro2{sekigawa} ~ (4)% pip3 install klayout  <===== Reinstall "pymod" from PyPI
Collecting klayout
  Downloading klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl (16.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.7/16.7 MB 1.7 MB/s eta 0:00:00
Installing collected packages: klayout
Successfully installed klayout-0.28.3

(base) MacBookPro2{sekigawa} ~ (5)% pip3 list | grep kl
cloudpickle                   2.0.0
klayout                       0.28.3 <===== OK
mkl-fft                       1.3.1
mkl-random                    1.2.2
mkl-service                   2.4.0
pickleshare                   0.7.5

(base) MacBookPro2{sekigawa} ~ (6)% which python3
/Applications/anaconda3/bin/python3

(base) MacBookPro2{sekigawa} ~ (7)% python3
Python 3.9.15 (main, Nov  4 2022, 11:11:31)
[Clang 12.0.0 ] :: Anaconda, Inc. on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.dbcore
    ^^^^^^^^^^^^^^^^^^^^^ No errors
>>> help(klayout.dbcore)

Help on module klayout.dbcore in klayout:

NAME
    klayout.dbcore - KLayout core module 'db'

CLASSES
    builtins.object
        Box
        Cell
        CellInstArray
        CellMapping
        CompoundRegionOperationNode
        Connectivity
        CplxTrans
        DBox
        DCellInstArray
        DCplxTrans
        DEdge
        DEdgePair
        DPath
        DPoint
        DPolygon
        DSimplePolygon
        DText
        DTrans
        DVector
        DeepShapeStore
        DeviceAbstract
        DeviceAbstractRef
        DeviceClass
            DeviceClassBJT3Transistor
                DeviceClassBJT4Transistor
            DeviceClassCapacitor
                DeviceClassCapacitorWithBulk
            DeviceClassDiode
            DeviceClassInductor
            DeviceClassMOS3Transistor
                DeviceClassMOS4Transistor
            DeviceClassResistor
                DeviceClassResistorWithBulk
        DeviceClassFactory
        DeviceExtractorBase
            DeviceExtractorBJT3Transistor
                DeviceExtractorBJT4Transistor
:
:
:

```

----

```
(base) MacBookPro2{sekigawa} klayout (1)% pwd
/Applications/anaconda3/lib/python3.9/site-packages/klayout

(base) MacBookPro2{sekigawa} klayout (2)% ll
total 64528
drwxr-xr-x  33 sekigawa wheel     1056 Jan 26 21:59 .
drwxr-xr-x 698 sekigawa wheel    22336 Jan 26 21:59 ..
-rw-r--r--   1 sekigawa wheel       34 Jan 26 21:59 __init__.py
drwxr-xr-x   3 sekigawa wheel       96 Jan 26 21:59 __pycache__
drwxr-xr-x   5 sekigawa wheel      160 Jan 26 21:59 db
drwxr-xr-x  10 sekigawa wheel      320 Jan 26 21:59 db_plugins
-rwxr-xr-x   1 sekigawa wheel    37416 Jan 26 21:59 dbcore.cpython-39-darwin.so
-rw-r--r--   1 sekigawa wheel  2376826 Jan 26 21:59 dbcore.pyi
drwxr-xr-x   4 sekigawa wheel      128 Jan 26 21:59 lay
-rwxr-xr-x   1 sekigawa wheel    37208 Jan 26 21:59 laycore.cpython-39-darwin.so
-rw-r--r--   1 sekigawa wheel   585462 Jan 26 21:59 laycore.pyi
drwxr-xr-x   4 sekigawa wheel      128 Jan 26 21:59 lib
-rwxr-xr-x   1 sekigawa wheel   816520 Jan 26 21:59 lib_ant.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel 40088912 Jan 26 21:59 lib_db.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  1249784 Jan 26 21:59 lib_edt.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  1252608 Jan 26 21:59 lib_gsi.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  1093376 Jan 26 21:59 lib_img.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel 12203128 Jan 26 21:59 lib_laybasic.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel   245984 Jan 26 21:59 lib_layview.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel   383624 Jan 26 21:59 lib_lib.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel   634560 Jan 26 21:59 lib_lym.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  2075936 Jan 26 21:59 lib_pya.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel    43968 Jan 26 21:59 lib_rba.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  1522320 Jan 26 21:59 lib_rdb.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel  1078096 Jan 26 21:59 lib_tl.cpython-39-darwin.so
-rwxr-xr-x   1 sekigawa wheel    37000 Jan 26 21:59 libcore.cpython-39-darwin.so
-rw-r--r--   1 sekigawa wheel      151 Jan 26 21:59 libcore.pyi
drwxr-xr-x   4 sekigawa wheel      128 Jan 26 21:59 rdb
-rwxr-xr-x   1 sekigawa wheel    36864 Jan 26 21:59 rdbcore.cpython-39-darwin.so
-rw-r--r--   1 sekigawa wheel    71673 Jan 26 21:59 rdbcore.pyi
drwxr-xr-x   4 sekigawa wheel      128 Jan 26 21:59 tl
-rwxr-xr-x   1 sekigawa wheel    36640 Jan 26 21:59 tlcore.cpython-39-darwin.so
-rw-r--r--   1 sekigawa wheel   118783 Jan 26 21:59 tlcore.pyi

(base) MacBookPro2{sekigawa} klayout (3)% otool -L lib_tl.cpython-39-darwin.so | grep png
  /usr/local/opt/libpng/lib/libpng16.16.dylib (compatibility version 56.0.0, current version 56.0.0)
  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
             !Homebrew's library!

```

It seems PyPI's "pymod" was built with Homebrew.
Because I have the Homebrew environment, `import klayout.dbcore` was successful.
However, the twisted environment is dirty.

Kazzz-S, 2023-01-26:
If you use "LW-klayout-0.28.3-macOS-Monterey-1-qt5Ana3-Rana3Pana3.dmg", reinstall the **homemade** "pymod" as follows.

```
(base) MacBookPro2{sekigawa} pymod-dist (1)% pwd
/Applications/klayout.app/Contents/pymod-dist

(base) MacBookPro2{sekigawa} pymod-dist (2)% ll
total 27272
drwxr-xr-x 4 sekigawa admin      128 Jan 15 06:27 .
drwxr-xr-x 9 sekigawa admin      288 Jan 15 06:27 ..
-rwxr-xr-x 1 sekigawa admin 13961592 Jan 15 06:27 klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl
-rwxr-xr-x 1 sekigawa admin 13962423 Jan 15 06:27 klayout-0.28.3-py3.9-macosx-10.9-x86_64.egg

(base) MacBookPro2{sekigawa} pymod-dist (3)% pip3 install klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl \
--force-reinstall

Processing ./klayout-0.28.3-cp39-cp39-macosx_10_9_x86_64.whl
Installing collected packages: klayout
  Attempting uninstall: klayout
    Found existing installation: klayout 0.28.3
    Uninstalling klayout-0.28.3:
      Successfully uninstalled klayout-0.28.3
Successfully installed klayout-0.28.3

```

I recalled how I find the <png.h> and <libpng*> to build "pymod" in the three environments.

[image]

klayoutmatthias, 2023-01-26:
I have no real idea how to package additional files.

The PyPI packages are built using github actions. The builds depend on pypa/cibuildwheel (https://github.com/pypa/cibuildwheel) which itself employs auditwheel or delocate which (to my understanding) should add all missing files to the package.

So either the problem is that something is broken there or the github hosts already have libpng and delocate does not have a reason to add these files.

Either way, I'm lost. Maybe someone else can help.

Matthias

lukasc-ubc, 2023-01-26:
Hi folks,

Thanks for your thoughts.  I realized that the HW and LW packages are for the GUI build, and pip import is a different beast all together that uses the local Python installation (rather than shipping with Python like the HW version).  

I was able to solve it by installing libpng using brew.  In my case, it took more effort, including upgrading brew and installaing libpng:

```
git -C /usr/local/Homebrew/Library/Taps/homebrew/homebrew-core fetch --unshallow
brew upgrade
brew install libpng
```

thank you

Jan-David-Black, 2023-02-02:
Hi folks, same issue here. However my homebrew installs libpng here
```sh
/opt/homebrew/lib/libpng16.16.dylib
```
meaning, that I cannot rectify the issue the way that @lukasc-ubc proposed

Jan-David-Black, 2023-02-02:
Installing the **homemade** pymod works. It is however a major inconvenience to have to do that on every install e.g. in virtual environments.

klayoutmatthias, 2023-02-02:
Bad news, but I can only emphasize, that I have no clue how to fix it. This needs someone with a deeper knowledge about packaging wheels. The details are hidden far too deep in the guts of these CI "helpers". 

I'm personally tired of digging into that stuff. I can do so, but that would put an end to KLayout feature development. No time left.

sebastian-goeldi, 2023-02-03:
This seems to be something related to where homebrew install libpng on m1. We are currently also running into this problem (I am running linux but colleagues are using M1 or newer). I requested an M1 from my IT and should get it soon:tm:. On suggestion that I was thinking about and that probably would make sense is to instead of relying on the dynamic library, compile libpng into the executable (probably makes it bigger, but might be less annoying).

https://github.com/Homebrew/discussions/discussions/2625

( @hsuan-tung could you try this as long as we don't have a testing machine yet? )

Hsuan-Tung, 2023-02-03:
When using conda environment to install, it works for me. I am using M1 mac. Here is what I did, 
```
1. conda create -n your_venv_name python=3.10
2. conda activate your_venv_name
3. conda install libpng
4. conda install libpng-devel-cos7-aarch64 (may not be necessary)
5. export LDFLAGS="-L $(libpng-config --libdir)"
6. cd klayout (go to source klayout repo folder)
7. python -m pip install .
```
But directly doing `brew install libpng` and `pip install klayout` won't work for me. Always getting this error, `src/tl/tl/tlPixelBuffer.cc:28:12: fatal error: 'png.h' file not found
      #  include <png.h>
                 ^~~~~~~
`

lukasc-ubc, 2023-02-04:
Hi folks,

thank you for the help.  I have completely removed Anaconda from my system, and installed Python using brew and pyenv, and the problem is fixed.

```
iMac-Hickup:matplotlib_example_node_pack-main lukasc$ pip install klayout
Requirement already satisfied: klayout in /usr/local/lib/python3.10/site-packages (0.28.3)
iMac-Hickup:matplotlib_example_node_pack-main lukasc$ python
Python 3.10.9 (main, Dec 15 2022, 18:20:40) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.dbcore
>>>
```

klayoutmatthias, 2023-02-04:
That is interesting. Sorry for not being more constructive.

Do you feel able to summarize the analysis for me? I am not really familiar with the ways HomeBrew, Anaconda and the system libraries interact. I'm simply a Linux guy.

Thanks,

Matthias

sebastian-goeldi, 2023-02-04:
From Hsuan-Tung's machine, I am pretty sure on m1 homebrew puts the libraries in a separate path (for the arm64 part, works fine if you use x86_64 brew and rely on rosetta doing work for you) which is not in PATH (into `/opt/homebrew/Cellar/`).

I tried yesterday with Hsuan-Tung to compile the library into klayout, but it doesn't seem straight forward. In conda it seems to work if you link against the library during compilation manually.

joamatab, 2023-02-09:
cc

@thomaslima

thomaslima, 2023-02-09:
Thanks for tagging me. I have a macOS M1 Ventura 13.0.1 and building it doesn't work for me because there is no libpng in my machine, so I can't include `png.h`. The build assumes that libpng is there. I'll see what I can do.

sebastian-goeldi, 2023-02-09:
That's my understanding. libpng on M1 when installed is not installed to somewhere , where PATH (or ots equivalents) can find it. By manually linking as in Hsuan-Tung's description works (at least for conda). You can add it manually for other environments, but then it crashes while importing and not on compile

thomaslima, 2023-02-09:
So Lukas's first installation was on x64, whereas yours was on M1. the macOS environment in GitHub actions seems to have curl, expat and png libraries. Maybe no longer the case in the newest macOS. I just confirmed I could build klayout without png on my Mac. I think that python klayout without png would fail at loading png files. "tlPixelBuffer.cc". @klayoutmatthias was png always a dependency? I don't recall.

I propose editing setup.py to search for homebrew libpng, and if it can't find it, install a version without libpng. Do you agree?

thomaslima, 2023-02-09:
Did some investigation. Matplotlib used to have libpng as a dependency as well. Here's how they checked and obtained flags for compilation: https://github.com/matplotlib/matplotlib/blob/73f6ac0bec7cb3845253adce82286530173f3a86/setupext.py#L653

However, in more recent versions they dropped the dependency and used PIL instead. Last version they used it was 3.2.x.

So I downloaded their wheel from pypi and checked the linked libraries in their `_png` module:

```bash
$ otool -L _png.cpython-38-darwin.so
_png.cpython-38-darwin.so:
	@loader_path/.dylibs/libpng16.16.dylib (compatibility version 38.0.0, current version 38.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)

$ ls .dylibs
libpng16.16.dylib*	libz.1.2.10.dylib*
```

As you can see, they simply shipped libpng and its dependency libz together with their macOS wheel. So it can be done but it may be a PITA to maintain.

@klayoutmatthias we have two options: 1. disable libpng for the python package; or 2. ship libpng with the wheel AND edit the source install so it searches a homebrew version, fail if user does not have libpng installed (like in matplotlib's case). The libpng linking flags are `libpng-config --ldflags` and the include flags are `libpng-config --cflags`. I propose option 1 for the python package, since I am not aware of a use case for libpng from klayout.tl.

sebastian-goeldi, 2023-02-10:
Libpng is used for the new lay submodule afaik (as of 0.28). Though db now also fails without it. But I guess it is used for the screenshot and pixelbuffers. I'd like to keep it because it enables us to do awesome stuff directly in python, such as the klayout integration of gdsfactory into a jupyter notebooks

thomaslima, 2023-02-10:
@sebastian-goeldi , can you provide a script example of a use case of the screenshot/pixelbuffer? I would like to add it as a test in the suite of tests.

thomaslima, 2023-02-11:
@Hsuan-Tung @sebastian-goeldi 
In #1288 I have patched setup.py so it would link with whatever libpng is available in the PATH (via libpng-config). I have tested building from source with libpng installed via homebrew or conda (anticipating a future conda package, which now would need libpng dependency). Please help test it, if you can. It would be great to add a test case that specifically relies on libpng, though. It also works on my M1 Mac.

I also fixed @lukasc-ubc 's original issue where the wheel was shipped with a hardcode link to home-brew's libpng. Now future macOS wheels will have libpng.dylib embedded.

thomaslima, 2023-02-13:
@sebastian-goeldi is this what you are referring to? https://github.com/gdsfactory/gdsfactory/blob/b1d857446e1918b6f3db5b1cce3ba4de165652a5/gdsfactory/widgets/layout_viewer.py#L14

Very cool. Would be nice to provide a minimal example so I can add to the test suite.

sebastian-goeldi, 2023-02-13:
@thomaslima sorry for the late reply. Had quite the busy week last week. Kinda. This is the move fleshed out way we are trying to use it https://github.com/gdsfactory/kweb . But yes, the widget is probably better. kweb is more based on Matthias' canvas2canvas project

Hsuan-Tung, 2023-02-13:
Thanks @thomaslima ! I just tested it and it worked on my M1 Mac as well.

sebastian-goeldi, 2023-02-13:
@thomaslima minimal example:

```python
import klayout.db as kdb
import klayout.lay as lay

#### create waveguide.gds
# this wouldn't need to be done by the test, this is just for show casing
# it is probably better to actually make this a separate file, so that it can be recreated,
# but the test doesn't create a false positive if e.g. the waveguide isn't drawn in the gds

layout = kdb.Layout()
cell = layout.create_cell("waveguide")
cell.shapes(layout.layer(1, 0)).insert(kdb.Box(0, -500, 10000, 500))
cell.write("wg.gds")

# load the gds through a layoutview object

lv = lay.LayoutView()
lv.load_layout("wg.gds")
lv.resize(400, 400)
lv.max_hier()

pb = lv.get_screenshot_pixels()
pb.write_png("wg.png")
```

thomaslima, 2023-02-13:
Added test case in #1292 .

@Kazzz-S As a separate issue, I am wondering if the libpng makes its way into the packaged .dmg file. If you confirm, we can close the ticket.

klayoutmatthias, 2023-02-14:
Many thanks for these contributions here! I really appreciate this.

I'm sorry - I had some hardware+tapeout trouble, so I was not able to follow here. I merged the two PRs and the solution will go into the next 0.28.6 PyPI module/DMGs. I think that will something like in two weeks. Will that be good enough?

Thanks and best regards,

Matthias

thomaslima, 2023-05-22:
I believe this issue has been fully addressed by now. libpng is shipped with python wheels. the app uses Qt's PNG engine, so it doesn't depend on libpng.

Installing klayout from source in linux and macOS will require libpng libraries to be installed (it uses the `libpng-config` command to locate it). Can be installed with apt-get or homebrew.

closed 2023-05-22 by thomaslima