KLayout/klayout #1266: Ruby macro script became much slower after updating to klayout 0.28.3

dsus4wang, 2023-01-28:
Hi Matthias,
I am a Mac M1 Klayout user .Now I use klayout 0.28.3 (build by myself , use qt5, ruby 2.6.10(system version) and python3.10.9(homebrew)) .After I updated klayout to the 0.28.3 version from 0.28.2(compiled by myself , the same qtBrew+rubySys+pyBrew) , the script run speed became very slow .Here is the example:
(load the lib)
```
include RBA
t1 = Time.new
p t1.inspect + " " + t1.usec.to_s
module TestLib
  class Poly1 < PCellDeclarationHelper
    def initialize
      super
      param :layer_a, TypeLayer, "layer_a", :default => LayerInfo::new(900101, 0, "a/layer_a")

      def display_text_impl
      "123"
      end

      def produce_impl
        hull = [RBA::Point::new(0, 0), RBA::Point::new(1000, 0), RBA::Point::new(1000, 1000), RBA::Point::new(0, 1000)]
        poly = RBA::Polygon::new(hull)
        p2 = poly.sized(1000)
        cell.shapes(layer_a_layer).insert(p2)
      end
    end
  end

  class TestLib < Library
    def initialize
      layout.register_pcell("poly1", Poly1::new)
      register("TestLib")
    end
  end

  TestLib::new
end
t2 = Time.new
p t2.inspect + " " + t2.usec.to_s

```
If I use 0.28.2 , the running time is from "2023-01-28 15:19:28 +0800 805626" to "2023-01-28 15:19:28 +0800 821340" , about 15714 us
If I use 0.28.3 , the running time is from "2023-01-28 15:19:56 +0800 783456" to "2023-01-28 15:19:56 +0800 785124" ,about 1668 us

I have a big library like this above, the loading time cost of 0.28.3 is also 10 times slower than that of 0.28.2.

Besides , when I try to draw something using script , the draw speed is also much slower than 0.28.2 version.

Could you please help me solve the problem? thanks.

klayoutmatthias, 2023-01-28:
I can't help. I'm not a Mac user. My rough guess is maybe you created a debug build by mistake.

There is no substantial performance difference between 0.28.2 and 0.28.3 on Linux.

Matthias

klayoutmatthias, 2023-01-28:
Here are my numbers from Ubuntu 22, .deb packages from www.klayout.org:

KLayout 0.28.2:

```
"2023-01-28 20:47:13.996926124 +0000 996926"
"2023-01-28 20:47:13.997235989 +0000 997235"
```

KLayout 0.28.3

```
"2023-01-28 20:50:15.821744426 +0000 821744"
"2023-01-28 20:50:15.822062082 +0000 822062"
```

So that's 318 vs. 309 µs  ... 

Matthias

dsus4wang, 2023-01-30:
Thank you Matthias , I just compiled and tested it on my Windows PC and it worked well. I think the question maybe caused by mac build script.
@Kazzz-S Hi Kazzz-S ,I have tested all version of 0.28.3 for mac (both the official release and dmg built my myself),the running speed is much slower than the last version 0.28.2 . Could you please help me solve the question? thanks a lot.

Kazzz-S, 2023-01-30:
Hi @dsus4wang,

OK, I'll check it.
I was also thinking of the possibility of a **debug build** as Matthias commented. Because In our earlier discussion on the build issue [here](https://github.com/Kazzz-S/klayout/issues/46#issuecomment-1369529968) I can see:

```
:
rm -f ../../../../qt5Brew.build.macos-Monterey-debug-RsysPhbauto/libklayout_QtMultimedia.0.28.dylib
                                               ^^^^^
                            This part should have been "release" if non-debug (release) mode.
:
```

Can you
 1) provide the complete command line string you used, like: `./build4mac.py -q qt5brew -r sys  -p hbauto`
 2) check the build directory name like:  `qt5Brew.build.macos-Monterey-[debug|release]-RsysPhbauto`

 Kazzz-S

Kazzz-S, 2023-01-30:
Hi @dsus4wang,

Thanks for your reply. It is a **release build**. 
Unfortunately, I'm not able to install the DMG because my PC is an Intel-Mac. :-(

Kazzz-S

dsus4wang, 2023-01-30:
@Kazzz-S 
Hi Kazzz-S,
My M1 Mac can run the Intel version, the Intel version 0.28.3 is also much slower than Intel version 0.28.2 .
The running speed: 0.28.2(arm64 ver) > 0.28.2(x86 ver) >> 0.28.3(arm64 ver) ~= 0.28.3(x86 ver)

Kazzz-S, 2023-01-30:
Hi @dsus4wang,
> @Kazzz-S Hi Kazzz-S, My M1 Mac can run the Intel version, the Intel version 0.28.3 is also much slower than Intel version 0.28.2 . The running speed: 0.28.2(arm64 ver) > 0.28.2(x86 ver) >> 0.28.3(arm64 ver) ~= 0.28.3(x86 ver)

Are these four programs Ventura binaries and are they running on Ventura 13.2? Or 13.1?
It may not be relevant, but I encountered a very bad performance problem with the Dropbox application in macOS 13.1.
It resolved after upgrading to 13.2.

Another idea that just comes to my mind is using Homebrew Ruby 3.1.
The present build script assumes Ruby 3.1.x (not the latest 3.2). So please install it by `brew install ruby@3.1` if you try.

Kazzz-S

Kazzz-S, 2023-01-31:
Hi @dsus4wang,

I've tested the RBA in my environment. Yes, the performance looks deteriorated.

[image]

```
(base) MacBookPro2{sekigawa} ~ (1)% uname -a
Darwin MacBookPro2.local 21.6.0 Darwin Kernel Version 21.6.0: Mon Dec 19 20:44:01 PST 2022; \
root:xnu-8020.240.18~2/RELEASE_X86_64 x86_64
```

[issue-1266.xlsx](https://github.com/KLayout/klayout/files/10545073/issue-1266.xlsx)

Kazzz-S

dsus4wang, 2023-01-31:
Hi @Kazzz-S @klayoutmatthias ,
Thanks for your important information. Now I find a problem about the RBA function :
This is the test code
```
t1 = Time.new()
layer1 = LayerInfo::new(2, 0, "test")
for i in 10000..1000000
  Point.new(i,i+1)
end
t2 = Time.new()
p t2-t1
```
The time cost of 0.28.2 is 1.5s ,while the cost of 0.28.3 is 27s.
I find the running speed is not relevant to the plaform or ruby version , just relevant to the RBA class.
Maybe this discovery can help?

Kazzz-S, 2023-01-31:
Hi @dsus4wang,

Thanks for the additional test case.
[image]
[issue-1266a.xlsx](https://github.com/KLayout/klayout/files/10546054/issue-1266a.xlsx)

Kazzz-S

Kazzz-S, 2023-02-01:
Hi @dsus4wang,

Here are the experiment results on Linux.

[image]
[issue-1266b.xlsx](https://github.com/KLayout/klayout/files/10553935/issue-1266b.xlsx)

----

## At this point, I realized my severe mistakes! ##

I collected the above timing data in the **Macro Development Tool**.
Maybe you, too.
This means that we are NOT observing the actual performance of RBA classes but the performance of the IDE tool!

Matthias keeps improving the IDE tool for easy debugging.
So, I guess that IDE's internal work has also increased between the two versions.
I believe this is the reason.

We should have run the RBA script outside the IDE tool as below.
As you can see, basically, there is no difference in the performance.

```
===== << 0.28.2 >> ==========================================================

${HOST}{sekigawa}(1)$ pwd
/home/sekigawa/GitWork/klayout/0.28.2/qt5.bin.linux-64-gcc-release

${HOST}{sekigawa}(2)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.918288287   <=== (Run1)

${HOST}{sekigawa}(3)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.906429665   <=== (Run2)

${HOST}{sekigawa}(4)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.915584038   <=== (Run3)

===== << 0.28.3 >> ==========================================================

${HOST}{sekigawa}(1)$ pwd
/home/sekigawa/GitWork/klayout/0.28.3/qt5.bin.linux-64-gcc-release

${HOST}{sekigawa}(2)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
1.071307478   <=== (Run1)

${HOST}{sekigawa}(3)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
1.067087545   <=== (Run2)

${HOST}{sekigawa}(4)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
1.074700268   <=== (Run3)

===== << 0.28.4 >> ==========================================================

${HOST}{sekigawa}(1)$ pwd
/home/sekigawa/GitWork/klayout/0.28.4/qt5.bin.linux-64-gcc-release

${HOST}{sekigawa}(2)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.985248188   <=== (Run1)

${HOST}{sekigawa}(3)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.977775996   <=== (Run2)

${HOST}{sekigawa}(4)$ ./klayout.sh -z -r ~/.klayout/macros/issue-1266a.rb
Debug: using qt5ct plugin
Debug: D-Bus global menu: no
0.984305041   <=== (Run3)

```

Kazzz-S

klayoutmatthias, 2023-02-02:
Hello folks,

many thanks for this debugging. I am sorry I did not mention this - actually, running a macro with debugging enabled will significantly slow it down!

The reason is that with debugging, on each line executed, the interpreter will return to the IDE and check for pending breakpoints and perform UI event processing. This is a significant time eater.

And you're right, the bugfix was about enabling breakpoints again. So this explains the timing problems. Initially, the IDE did not receive callbacks.

The timing effect is not very pronounced in my case (Ubuntu 22) and the timing depends much on the Qt and Ruby version and UI overhead. With the debugger enabled, a simple loop not involving RBA objects executes like 3x slower. With RBA objects, a lot more interpreter events are generated and execution is about 20x slower with the debugger enabled. These are the numbers you see.

If you disable the debugger like this

[image]

execution times are back to normal.

The same is true, if you do not run a macro from the IDE, but bind it to a menu and close the IDE window.

In a production environment - i.e. when you execute code for PCells or add-on features - I assume that the IDE is closed, hence the timing effect should not be seen.

Matthias

Kazzz-S, 2023-02-02:
Hi @klayoutmatthias,

Thank you for your explanations.

> If you disable the debugger like this, execution times are back to normal.

I've confirmed this below.
```
17.350244 <=== debugger ON
0.68651   <=== debugger OFF
```

Kazzz-S

dsus4wang, 2023-02-04:
Thanks for your help!

closed 2023-02-04 by dsus4wang