KLayout/klayout #1274: 2.5d View on Windows has Wrong Viewport

ejprinz, 2023-02-04:
I have used the new 2.5d view on both versions 0.28.3 and 0.28.4. On Ubuntu Linux LTS 22.04 it works as expected, i.e., the entire drawing area is used. But on Windows only the lower left quadrant of the viewing area is used, not the entire window. 
[image]
This seems to be a Windows-specific issue. Please let me know if you need more info.

klayoutmatthias, 2023-02-05:
hmmm ... I could not see this problem on different Windows computers. Neither on a laptop screen (HD resolution), nor on a different computer connected to a 4k display.

But I need to say I don't have fancy graphics hardware - only entry level.

KLayout uses the OpenGL integration from Qt. Maybe there is either an issue with OpenGL or with Qt.

Matthias

ejprinz, 2023-02-05:
Good point. The laptop in question has an Intel Core 5 chip with built-in
graphics. Let me google what to do about this.

ejprinz, 2023-02-06:
I identified the root cause. To adjust the size of the fonts, I started KLayout with the environment variable $QT_SCALE_FACTOR of 0.9 (in Windows). If I delete this environment variable the viewport is OK.

klayoutmatthias, 2023-02-06:
Hi Ewin,

thanks for letting me know. Either that screws up the Qt pixel arithmetics, or there is something wrong on my side.

I'll check that.

Best regards,

Matthias

ejprinz, 2023-02-06:
Hi Matthias, I had set the QT_SCALE_FACTOR to 0.9 so for some reason the
viewport was cut down by half so it doesn't seem to be simply related to
the value of the scale factor...

closed 2024-03-23 by klayoutmatthias