KLayout/klayout #1275: Region.sized after .smooth returns empty region on 0.28.4

caspar-iqm, 2023-02-04:
Here is a pretty obscure bug report!

It looks like on 0.28.4, `Region.smooth` does something that prevents further `size` or `sized` operations on the region to work as expected.

Try the following macro:

```py
from pya import Point, Polygon, Region

polygon = Polygon([Point(-21983, -43808), Point(-37841, 16636), Point(-20484, 35228), Point(30428, 41627), Point(38312, 28960), Point(-7811, -37922)])

# NOT WORKING: Region.sized or Region.size after Region.smooth results in empty region
region = Region(polygon)
region.smooth(100)
print("Original smooth:", str(region))  # 

region_sized = region.sized(100)
print("Sized after smooth:", str(region_sized))

# WORKING: Region.sized or Region.size after Region.smoothed works as expected
region = Region(polygon)
region = region.smoothed(100)
print("Original smoothed:", str(region))  # 

region_sized = region.sized(100)
print("Sized after smoothed:", str(region_sized))
```

On 0.28.4, the output is
```
Original smooth: (-21983,-43808;-37841,16636;-20484,35228;30428,41627;38312,28960;-7811,-37922)
Sized after smooth: 
Original smoothed: (-21983,-43808;-37841,16636;-20484,35228;30428,41627;38312,28960;-7811,-37922)
Sized after smoothed: (-22037,-43939;-22054,-43930;-37952,16664;-20532,35323;30479,41734;38432,28957;-7745,-38003)
```

On 0.28.3 it works correctly ("Sized after smooth" is the same as "Sized after smoothed").

It seems related to in-place editing in `smooth` having some side-effect. Re-creating the Region object as follows also fixes the issue:

```py
region = Region(polygon)
region.smooth(100)
region = Region(list(region.each()))
print("Original smooth:", str(region))  # 

region_sized = region.sized(100)
print("Sized after smooth:", str(region_sized))
```

sebastian-goeldi, 2023-02-04:
I believe this should never have worked. If you look at the docs smooth returns void (and always did according to the docs iirc). https://klayout.de/doc-qt5/code/class_Region.html#method198 . Same with `round_corners`.

I have wondered in the past about this. @klayoutmatthias would it be possible to standardize the return types (my preference would be that they return `this`, but I can also live with void), or is there a reason for different behavior?

klayoutmatthias, 2023-02-04:
I am terribly sorry, this is a bug I have introduced in 0.28.4 during the enabling of user properties.

I'm in release mood anyway, so I'll quickly generate a 0.28.5 release with that bug fixed.

caspar-iqm, 2023-02-05:
Thank you for the quick fix, amazing work!

klayoutmatthias, 2023-02-05:
Is fixed with 0.28.5

closed 2023-02-05 by klayoutmatthias