KLayout/klayout #1279: Command line option -l seems to be ignored when -nn is given

hpretl, 2023-02-05:
When I call klayout in this way

`klayout -e -nn $PDKPATH/libs.tech/klayout/tech/$PDK.lyt -l $PDKPATH/libs.tech/klayout/tech/$PDK.lyp example.gds` then the following errors is printed:

`Warning: Initialization of layers failed: Unable to open file: /headless/.klayout/tech/sky130/sky130.lyp (errno=2)`

This is not expected, since I supply the `.lyp` via the call.

Here is the start of the respective `.lyt` file:

```
<?xml version="1.0" encoding="utf-8"?>
<technology>
 <name>sky130</name>
 <description>SkyWater 130nm technology</description>
 <group/>
 <dbu>0.001</dbu>
 <base-path>$(appdata_path)/tech/sky130</base-path>
 <original-base-path>$(appdata_path)/tech/sky130</original-base-path>
 <layer-properties_file>sky130.lyp</layer-properties_file>
 <add-other-layers>true</add-other-layers>
 <reader-options>
  <gds2>
   <box-mode>1</box-mode>
   <allow-big-records>true</allow-big-records>
   <allow-multi-xy-records>true</allow-multi-xy-records>
  </gds2>
  <common>
   <create-other-layers>true</create-other-layers>
   <layer-map>layer_map()</layer-map>
   <enable-properties>true</enable-properties>
   <enable-text-objects>true</enable-text-objects>
  </common>
```

klayoutmatthias, 2023-02-06:
Maybe the .lyt file specifies a layer properties file too? 

Like here (sorry for the ugly underscore):

```
 <layer-properties_file>tech.lyp</layer-properties_file>
```

The -l option does not override the layer properties from the .lyt file - if the technology specific .lyp file cannot be read (hint: relative paths are resolved relative to the .lyt file), then this error is raised.

Matthias

klayoutmatthias, 2023-02-06:
(deleted previous comment)

Basically, the -l option does not override the "layer-properties_file" (sorry for the ugly underscore) settings. So if the file given there cannot be read, then this error happens.

It's kind of inconsistent, that's true. But basically, if the .lyp file from the .lyt does not exist, this .lyt technology is not a valid one.

Do you have a reason to keep the .lyp file in the technology even if there is no such file?

Matthias

hpretl, 2023-02-07:
This is the current situation in the SKY130 klayout tech files. I try to find a proper fix, in the moment I patch the techfiles. It would be still helpful if klayout would check both locations, the one given via cmdline, and the other via lyt.

I can remove the -l parameter from calling klayout, and just keep patching the lyt, that is another option if you do not want to change klayout's behavior.

closed 2023-02-10 by hpretl