KLayout/klayout #1291: matplotlib, Qt

lukasc-ubc, 2023-02-13:
[Eight years ago](https://www.klayout.de/forum/discussion/750/python-matplotlib-pyplot-in-klayout), I was seeing if I could get matplotlib to work in KLayout.  I tried today and it seems to work quite well on Mac OSX, and I was able to [generate simple plots](https://github.com/SiEPIC/SiEPIC-Tools/commit/e54b5cbd86974d40be2fe68a637b3c86466e3614#commitcomment-100343212).  

I was able to even install Flex Compute's Tidy3D simulator within the HW KLayout build and have the plots [generate within KLayout](https://github.com/SiEPIC/SiEPIC-Tools/commit/646898b6c8e15ecf1dc6c185d0ea05d313a092d7#commitcomment-100344079).

My approach for installing packages is to do it within the KLayout IDE Python using the pip module:

```
import pip
pip.main(['install', 'matplotlib'])
pip.main(['install', 'tidy3d'])
```

Problems encountered:
1) On Mac OSX, it causes a KLayout crash if I close the windows.  
2) On both Windows and Linux (Ubunto 22) KLayout froze (I waited for 30 minutes), so that didn't work.

I suspect the OSX version is using the MacOSX backend.  Perhaps it would be better to use the Qt backend?

@thomaslima, perhaps you have already experience with this? 

It would be nice to be able to create Qt dialogs in KLayout with plots.

For example, it could be used to help designers choose parameters from a PCell or display other technology information.  And even display some simulation results.

@momchil-flex, @tomflexcompute, I would like to make it easy for students to use tidy3d in the edX Phot1x course.  I'm curious what you think.

Thank you
Lukas

[image]

thomaslima, 2023-02-13:
Hi Lukas. I wish there was an easy solution to this. First, to debug this, I feel like we will need to have a minimum working example. I wonder if just installing matplotlib and running a simple plot will cause this behavior. Also, have you tried the LW release? That might work better.

I tried working on a deeper python integration a while ago (#660), but never finished it. The first challenge is coming up with a solution that is compatible not only with Macs but also Windows and Linux.

To solve this problem once and for all, I think the ultimate solution is to have klayout (GUI) communicate with a native python instance and exchange data as necessary. This would be an optional featured enabled by a setting in the GUI, where you would choose the desired python location, much like VSCode's can choose which python environment to use when running scripts. This is similar (in spirit) to Tait's https://github.com/atait/klayout-ipc project, where commands are communicated via a socket server and pya's QTcpServer. In this scenario, macros would be run by the chosen python executable, instead of Klayout's internal interpreter. Maybe we can borrow some code from the ipython/jupyter project which enable this functionality by using python kernels independent from the "code editor". @klayoutmatthias what do you think? This would require enabling a different process control pya.Application etc. Something like this project (https://github.com/tomerfiliba-org/rpyc)

joamatab, 2023-02-13:
Hi Lukas,

in gdsfactory we use the klayout python API to render plots and avoid using the python inside klayout to avoid these types of issues. This is similar to what Thomas proposes with using a native pyhton instance.

https://gdsfactory.github.io/gdsfactory/notebooks/00_geometry.html

We are also starting to use [kweb](https://github.com/gdsfactory/kweb) where we can expose the klayout python API through a web interface

@proppy
@thomasdorch
@atait
@flaport

thomaslima, 2023-02-13:
This wouldn't work for Lukas because his use case is to launch third-party python modules from within klayout's macro environment. So a "reverse API". This sounds difficult to me at the moment -- like having klayout host a jupyter_client, with a remote python kernel, but with bidirectional communication because the scripts would need access to the host's GSI. Welcome thoughts on possible implementation.

thomaslima, 2023-02-13:
Here's an idea. Rather than burdening klayout's maintainers, what if we enable the macro editor to use an optional external python binary to execute the macro? Something along these lines:

1. We help the user by auto-finding available binaries in known places (conda, brew, pyenv etc) and creating a dropdown menu.
2. We modify the macro execution so that if it uses an external binary/environment, klayout calls that binary as a subprocess (like in jupyter). We can implement it as a jupyter client and use the remote environment as a 'kernel'.
3. We can remember which macro used which 'kernel'. Macros by default uses klayout's interpreter.
4. Only macros running in klayout's interpreter can use pya objects directly.

Salt package developers and python developers can then be free to use whatever framework they choose for inter-process communication. 
- @lukasc-ubc can ask students to install an anaconda environment and point klayout to that environment. 
- Basic scripts can then take care to install dependencies on that environment with `$REMOTEPYTHON -m pip install whatever`.
- That python installation can be a different version entirely, disconnected from klayout's.
- pya is not going to be available. Developers need to create a salt package if they want to manipulate klayout's windows and layout views. They also need to figure out how to transfer database objects back and forth such as polygons or layouts. (Maybe in the future we can provide a reference implementation).

I think this will enhance UX significantly, much in the same way that jupyter/vscode's ability to select & remember python kernels improved my life a lot. I also think that this will simplify salt packages too by making it be simple scripts that connect to more featured python packages maintained externally by some anaconda environment or package manager.

flaport, 2023-02-14:
I would love having access to an external python in KLayout!

For the record, there is also an effort to build the KLayout GUI for conda, such that it has access to all the packages installed in a conda environment. See here for example https://github.com/conda-forge/staged-recipes/pull/20396.

However, being able to access an external python might be a better, more flexible solution as long as it does not restrict which external python version can be used.

proppy, 2023-02-14:
@flaport we do have a klayout package in conda-eda too https://anaconda.org/LiteX-Hub/klayout/files (that does include the GUI).

klayoutmatthias, 2023-02-14:
Coming back to the original question, I'll check why mathplotlib isn't working with KLayout. I think it has to do with the coexistence of Qt and another event loop. But if that was working once, it should be possible to revive that option.

Becoming more general, I think you overestimate KLayouts "workbench" abilities somewhat. After all, that is a simple single-threaded application with all the restrictions that come with it. I created Python support for simple automation tasks. 

As a general solution I was actually thinking about turning KLayout into a remote-controlled engine (on a deeper level than rpyc) and running the IDE as a separate process. That avoids manifold problems connected with single-threaded applications. However this is not a small project, but it goes in the direction of having built-in RPC support with your own choice of client.

As of now I - because of the simplicity of that approach - I personally favor the "build from scratch" idea that is behind the LayoutView object of the PyPI klayout module: this eventually is the atomic form of the KLayout GUI and you should be able to integrate it into whatever application you're building - even if this application is a web server (see kweb). This gives you maximum freedom of integration, but of course you need to spend some effort building the application the way you need it.

I'm aware that this needs more activity. My personal focus right now is on PDK support which is a somewhat different topic.

Best regards,

Matthias

thomaslima, 2023-02-14:
> Coming back to the original question, I'll check why mathplotlib isn't working with KLayout. I think it has to do with the coexistence of Qt and another event loop. But if that was working once, it should be possible to revive that option.

I have since tested on my machine. No issues with plotting simple structures and closing windows. I suspect it has something to do with tidy3d or a particular installation.

> Becoming more general, I think you overestimate KLayouts "workbench" abilities somewhat. After all, that is a simple single-threaded application with all the restrictions that come with it. I created Python support for simple automation tasks.

I see your point. But I think many people already use it as their workbench. There's lots of potential in the way you implemented python/ruby-based extensions to the GUI. Like you said (simple automation). This functionality would simply extend their automation capabilities. The only question is whether this would create more technical debt than it would remove. My money is that it would be a net positive for simplicity. My proposed change would launch a new thread to run the macro, communicating only via pre-defined socket-based I/O. I can explore this idea a bit more later.

> As a general solution I was actually thinking about turning KLayout into a remote-controlled engine (on a deeper level than rpyc) and running the IDE as a separate process. That avoids manifold [...]

That would be awesome. And I would still use klayout as a client. :)

> As of now I - because of the simplicity of that approach - I personally favor the "build from scratch" idea that is behind the LayoutView object of the PyPI klayout module: this eventually is the [...]

I always wanted to have an iPad app that would render layout files. :) Are you suggesting that folks willing to integrate klayout with other tools (like simulation tools) would need to code a new GUI? Or that the new klayout engine would need to be integrated into the other "workbench" software?

> I'm aware that this needs more activity. My personal focus right now is on PDK support which is a somewhat different topic.

Thanks for considering my thoughts! I also like @flaport and @proppy's proposal of using anaconda for advanced python users.

joamatab, 2023-02-15:
I like the idea of linking a separate python to facilitate for Windows and MacOs users 
@Jan-David-Black

For example, we have klayout packages for GF180nm and Skywater130nm and they only work well on Linux
https://sami.klayout.org/ 
because they depend on gdsfactory

the conda build of klayout+GUI does not work on MacOs yet, @thomasdorch did the windows one and Floris made the linux one

https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/28

klayoutmatthias, 2023-02-23:
@lukasc-ubc Coming back to the original question, I did some debugging.

matplotlib comes with a number of backends. In my case (Ubuntu without PyQt installated) the default was gtk. The gtk backend does not work inside KLayout because it ends in a deadlock (nested event loops I suppose). 

After I installed PyQt, things worked nicely except in the macro IDE. When running macros from inside the IDE, the application crashed. The reason was that the IDE was not aware of the foreign Python events that PyQt generates out of the UI system. I can fix that problem. 

So basically, with PyQt installed, matplotlib seems to be functional.

In generate, installing the PyQt should be easy with package management (conda, Linux, MacOS with homebrew or Anaconda). I'll check if I can include matplotlib and PyQt5 in the next Windows release (it's already heavy enough, so that should not be an issue).

Matthias

proppy, 2023-03-01:
> the conda build of klayout+GUI does not work on MacOs yet, @thomasdorch did the windows one and Floris made the linux one

@xobs recently did some work around this for conda-eda, see: https://github.com/hdl/conda-eda/pull/298

klayoutmatthias, 2023-03-01:
@lukasc-ubc Yes, I will try that for 0.28.6. If I am lucky, the MSYS system comes with the proper packages already. I hope there aren't too many integration issues.

xobs, 2023-03-04:
[image]
It works on my machine with the Conda packages I've built. The chip pictured was routed in GF180MCU entirely on an M1 machine using native binaries for openroad, magic, netgen, and klayout.

The only funny bit is that klayout.app isn't installed into Applications, because I'm not sure if/how that's supposed to happen. But it can be natively launched, as pictured.

lukasc-ubc, 2023-03-05:
Mustafa and I succeeded in getting plotly to work perfectly in KLayout for OSX, Windows and Linux this week. SiEPIC-Tools now includes a Photonic integrated circuit simulator "OPICS" which is able to create simulation plots. Simulations are done by extracting a netlist from KLayout, and a compact model library is included in the Technology/PDK as a Si folder "CML". Works beautifully, and we are teaching tomorrow at the OFC conference, in short course SC432. 
Regards
Lukas
Ps. We faced many challenges with Windows, importing some Python packages that are required by tidy3d. Many give errors about eggs (not the kind that birds lay).

klayoutmatthias, 2023-03-17:
I will release 0.28.6 today and in the Windows installer I have included matplotlib and PyQt5. It is working with the examples I tried so far. Even PyQt5 works, although I doubt that it is very easy to integrate with the application.

lukasc-ubc, 2023-03-18:
Thank you @klayoutmatthias 

**Crash on closing matplotlib windows on M1:**

In the Mac OSX version, I imported matplotlib, and after creating a window and closing it via the code below, KLayout crashes with the error reports below. I get a crash with both the version I compiled (LW), and the official version from KLayout.de, klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.  @thomaslima can you see if you can get it to crash as well?

```
import matplotlib.pyplot as plt  
import numpy as n

wavelength = n.array([1, 2, 3, 4, 5])
power = wavelength**2

plt.figure()
print(wavelength*1e9)
print(power)
plt.plot(wavelength*1e9, power)
plt.xlim((1*10**9,7*10**9)) 
plt.xlabel('Wavelength (nm)') 
plt.ylabel('Transmission (dB)')
print("TEST1")
plt.show(block=False)
print("TEST2")
for i in range(0,10):
    print("This text should show with the plot up")

```

**LW version crash report:**
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               klayout [54072]
Path:                  /Users/USER/Documents/*/klayout.app/Contents/MacOS/klayout
Identifier:            de.klayout
Version:               0.28.5 (0.28.5)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-03-17 20:28:34.5479 -0700
OS Version:            macOS 12.6.2 (21G320)
Report Version:        12
Anonymous UUID:        FEFE5F3F-C520-F094-2850-1B4D9866DCE8

Sleep/Wake UUID:       840AD793-0037-460C-B212-A0E579E92CC4

Time Awake Since Boot: 940000 seconds
Time Since Wake:       1157 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001b07f8cdc
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [54072]

Application Specific Information:
Over-release of NSResponder or subclass.

Kernel Triage:
VM - Compressor failed a blocking pager_get
VM - Compressor failed a blocking pager_get
VM - Compressor failed a blocking pager_get
[... 504 lines cut ...]
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2e5e6 RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2e34a ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x2e0e5 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3efad _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3d66a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x2fd19 -[NSApplication run]
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x3ab3a qt_plugin_instance
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1e3857 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1e7802 _ZN16QCoreApplication4execEv
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_lay.0.dylib +0x84ae2 _ZN3lay14GuiApplication4execEv
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_lay.0.dylib +0x830ff _ZN3lay15ApplicationBase3runEv
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x1318e
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b77f _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x171aa4 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16093f rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16e453 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5d8c9 ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x5d7cc ruby_run_node
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b71a _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x12c54
/Applications/klayout-0.28.5-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x127b4
/usr/lib/dyld +0x552e start
/usr/lib/dyld +0x0 start

```

lukasc-ubc, 2023-03-22:
On the Mac OSX, klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39, I ran the matplotlib [example](https://github.com/KLayout/klayout/issues/1291#issuecomment-1474662835), and closed the window, and it crashed:

```
Signal number: 4
Address: 0x7fff22e0f5a9
Program Version: KLayout 0.28.6 (2023-03-20 rd38899bf1)

Backtrace:
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_lay.0.dylib +0x20e05b _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x3d7d _sigtramp
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x368e0 -[CFPrefsSearchListSource alreadylocked_generationCountFromListOfSources:count:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x164018 -[NSTextView release]
/usr/lib/system/libsystem_blocks.dylib +0x1605 _Block_release
/usr/lib/system/libsystem_blocks.dylib +0x1605 _Block_release
/usr/lib/system/libsystem_blocks.dylib +0x1605 _Block_release
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x8076d __CFRunLoopDoBlocks
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7fa9c __CFRunLoopRun
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation +0x7e9bc CFRunLoopRunSpecific
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30a83 RunCurrentEventLoopInMode
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x307e5 ReceiveNextEventCommon
/System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox +0x30583 _BlockUntilNextEventMatchingListInModeWithFilter
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3eb12 _DPSNextEvent
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x3d2e5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25e0dc -[NSWindow(NSEventRouting) trackEventsMatchingMask:timeout:mode:handler:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25d4b5 -[NSDragEventTracker trackEvent:usingHandler:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25c7ab NSControlTrackMouse
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25c1d3 -[NSCell trackMouse:inRect:ofView:untilMouseUp:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25c09a -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x25b3c2 -[NSControl mouseDown:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x2597ae -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x1c8fc8 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x1c8386 -[NSWindow(NSEventRouting) sendEvent:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x1c6794 -[NSApplication(NSEvent) sendEvent:]
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x3e2da qt_plugin_instance
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x49f531 -[NSApplication _handleEvent:]
/System/Library/Frameworks/AppKit.framework/Versions/C/AppKit +0x2f62e -[NSApplication run]
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/PlugIns/platforms/libqcocoa.dylib +0x3ab3a qt_plugin_instance
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1e3857 _ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/QtCore.framework/Versions/5/QtCore +0x1e7802 _ZN16QCoreApplication4execEv
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_lay.0.dylib +0x84b22 _ZN3lay14GuiApplication4execEv
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_lay.0.dylib +0x8313f _ZN3lay15ApplicationBase3runEv
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x1318e
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b77f _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x174df3 rb_vm_fstring_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x16397c rb_vm_get_insns_address_table
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x171757 rb_vm_exec
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x603b0 ruby_exec_node
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/libruby.2.6.dylib +0x602b3 ruby_run_node
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/Frameworks/libklayout_rba.0.dylib +0x1b71a _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x12c54
/Applications/klayout-0.28.6-macOS-BigSur-1-qt5Brew-RsysPhb39.app/Contents/MacOS/klayout +0x127b4
/usr/lib/system/libdyld.dylib +0x15f3d start

```

klayoutmatthias, 2023-03-22:
That's bad ... the stack trace does not indicate anything specific.

Maybe it's worth trying other combinations. The PyQt5 integration that is needed by matplotlib complicates matters. It is very important that this module is built against the same Qt5 than KLayout is - if not, these kind of crashes are likely. In the Windows case I am using Python, PyQt5, matplotlib and Qt5 from the MSYS2 package manager. Same for Linux where everything comes from the Ubuntu repo in my case. That might explain why I do not see issues there.

Matthias

Kazzz-S, 2023-03-23:
I tried KLayout 0.28.6 (source code release-build) on Linux Mint 20.3 with another combination.
Valgrind output is attached (I hope it's useful).  [valgrind-klayout-0.28.6-linux.zip](https://github.com/KLayout/klayout/files/11045918/valgrind-klayout-0.28.6-linux.zip)

Kazzz-S

```
CAEDS04{sekigawa}(1)$ uname -a
Linux CAEDS04 5.4.0-144-generic #161-Ubuntu SMP Fri Feb 3 14:49:04 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

CAEDS04{sekigawa}(2)$ g++ --version
g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

CAEDS04{sekigawa}(3)$ pip3 list | grep matplotlib
matplotlib              3.6.3   (installed by pip3)

CAEDS04{sekigawa}(4)$ python3
Python 3.8.10 (default, Mar 13 2023, 10:26:41)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>

CAEDS04{sekigawa}(5)$ ./klayout -v
KLayout 0.28.6
================================================================================================================
CAEDS04{sekigawa}(6)$ valgrind --log-file=$HOME/temp/valgrind-klayout-0.28.6-linux.log \
                      --leak-check=full \
                      --show-leak-kinds=all \
                      --track-origins=yes \
                      ./klayout

Debug: using qt5ct plugin
Debug: D-Bus global menu: no
ERROR: ../../../src/lay/lay/layTechnologyController.cc,99,dispatcher == mp_dispatcher
terminate called after throwing an instance of 'tl::InternalException'
addr2line: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)

================================================================================================================
<<< KLayout GUI Application >>>
    (1) Start KLayout
    (2) Run the above Python script in the IDE
    (3) Save the plot  ===> "Figure_1.png"
    (4) Close matplotlib's plot window
    (5) Close the IDE
    (6) Close KLayout's main window
Then, crash
================================================================================================================
Signal number: 6
Address: 0x3e80002f823
Program Version: KLayout 0.28.6 (2023-03-21 r20c4e9c9c)

Backtrace:
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x2e9427 lay::enable_signal_handler_gui(bool) [??:?]
/lib/x86_64-linux-gnu/libc.so.6 +0x43090
/lib/x86_64-linux-gnu/libc.so.6 +0x4300b gsignal
/lib/x86_64-linux-gnu/libc.so.6 +0x22859 abort
/lib/x86_64-linux-gnu/libstdc++.so.6 +0x9e911
/lib/x86_64-linux-gnu/libstdc++.so.6 +0xaa38c
/lib/x86_64-linux-gnu/libstdc++.so.6 +0xa9369
/lib/x86_64-linux-gnu/libstdc++.so.6 +0xa9d21 __gxx_personality_v0
/lib/x86_64-linux-gnu/libgcc_s.so.1 +0x10bef
/lib/x86_64-linux-gnu/libgcc_s.so.1 +0x11281 _Unwind_RaiseException
/lib/x86_64-linux-gnu/libstdc++.so.6 +0xaa69c __cxa_throw
/home/sekigawa/GitWork/klayout/bin/libklayout_tl.so.0 +0x4b11a
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x296f3b
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x156cfc _ZN3lay14GuiApplicationD1Ev
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x156d7d _ZN3lay14GuiApplicationD0Ev
/usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-38-x86_64-linux-gnu.so +0x2110c1
/lib/x86_64-linux-gnu/libpython3.8.so.1.0 +0x25f34a
/lib/x86_64-linux-gnu/libpython3.8.so.1.0 +0x2a8840 PyVectorcall_Call
/lib/x86_64-linux-gnu/libpython3.8.so.1.0 +0x10d2e7
/lib/x86_64-linux-gnu/libpython3.8.so.1.0 +0x1935dd Py_FinalizeEx
/home/sekigawa/GitWork/klayout/bin/libklayout_pya.so.0 +0x44ac5 _ZN3pya17PythonInterpreterD1Ev
/home/sekigawa/GitWork/klayout/bin/libklayout_pya.so.0 +0x44afd _ZN3pya17PythonInterpreterD0Ev
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x14fb47 _ZN3lay15ApplicationBase8shutdownEv
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x156b90 _ZN3lay14GuiApplication8shutdownEv
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x156d0d _ZN3lay14GuiApplicationD1Ev
/home/sekigawa/GitWork/klayout/bin/libklayout_lay.so.0 +0x156d7d _ZN3lay14GuiApplicationD0Ev
./klayout +0x4ca2
/home/sekigawa/GitWork/klayout/bin/libklayout_rba.so.0 +0x23fc4
/usr/local/lib/libruby.so.3.1 +0x2b4e88
/usr/local/lib/libruby.so.3.1 +0x2c9f39
/usr/local/lib/libruby.so.3.1 +0x2ca804
/usr/local/lib/libruby.so.3.1 +0x2c3132
/usr/local/lib/libruby.so.3.1 +0x2c8d13 rb_vm_exec
/usr/local/lib/libruby.so.3.1 +0xdc5db
/usr/local/lib/libruby.so.3.1 +0xe298a ruby_run_node
/home/sekigawa/GitWork/klayout/bin/libklayout_rba.so.0 +0x24424 _ZN3rba15RubyInterpreter10initializeERiPPcPFiS1_S3_E
./klayout +0x487b
./klayout +0x3f9d
/lib/x86_64-linux-gnu/libc.so.6 +0x24083 __libc_start_main
./klayout +0x405e
================================================================================================================
```

[image]
[image]

Kazzz-S, 2023-03-23:
Here are some more tests results.

Kazzz-S

## [1] Using the distributed packages ##

```
CAEDS04{sekigawa}(1)$ python3
Python 3.8.10 (default, Mar 13 2023, 10:26:41)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import matplotlib
>>> matplotlib.__version__
'3.1.2'
>>> numpy.__version__
'1.17.4'
```

**[1-A] plt.show(block=False)**
1. The plot did not show up.
2. Outputs were printed immediately.
3. No crash occurred when closing the KLayout's main window.

```
import matplotlib.pyplot as plt
import numpy as n

wavelength = n.array([1, 2, 3, 4, 5])
power = wavelength**2

plt.figure()
print(wavelength*1e9)
print(power)
plt.plot(wavelength*1e9, power)
plt.xlim((1*10**9,7*10**9))
plt.xlabel('Wavelength (nm)')
plt.ylabel('Transmission (dB)')
print("TEST1")
plt.show(block=False)
#plt.show(block=True) # <=== default
print("TEST2")
for i in range(0,10):
    print("This text should show with the plot up")
```

```
[1.e+09 2.e+09 3.e+09 4.e+09 5.e+09]
[ 1  4  9 16 25]
TEST1
TEST2
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
```
----

**[1-B] plt.show(block=True)**
1. The plot showed up.
2. Outputs were printed immediately after closing the plot.
3. No crash occurred when closing the KLayout's main window.

```
import matplotlib.pyplot as plt
import numpy as n

wavelength = n.array([1, 2, 3, 4, 5])
power = wavelength**2

plt.figure()
print(wavelength*1e9)
print(power)
plt.plot(wavelength*1e9, power)
plt.xlim((1*10**9,7*10**9))
plt.xlabel('Wavelength (nm)')
plt.ylabel('Transmission (dB)')
print("TEST1")
#plt.show(block=False)
plt.show(block=True) # <=== default
print("TEST2")
for i in range(0,10):
    print("This text should show with the plot up")
```

```
[1.e+09 2.e+09 3.e+09 4.e+09 5.e+09]
[ 1  4  9 16 25]
TEST1
TEST2
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
```

## [2] Using the packages installed by pip3 (the same as [above)](https://github.com/KLayout/klayout/issues/1291#issuecomment-1480441203) ##

```
CAEDS04{sekigawa}(1)$ python3
Python 3.8.10 (default, Mar 13 2023, 10:26:41)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import matplotlib
>>> matplotlib.__version__
'3.6.3'
>>> import numpy
>>> numpy.__version__
'1.24.2'
>>>
```

**[2-A] plt.show(block=False)**
1. The plot showed up.
2. Outputs were printed immediately.
3. Crash occurred when closing the KLayout's main window.

```
import matplotlib.pyplot as plt
import numpy as n

wavelength = n.array([1, 2, 3, 4, 5])
power = wavelength**2

plt.figure()
print(wavelength*1e9)
print(power)
plt.plot(wavelength*1e9, power)
plt.xlim((1*10**9,7*10**9))
plt.xlabel('Wavelength (nm)')
plt.ylabel('Transmission (dB)')
print("TEST1")
plt.show(block=False)
#plt.show(block=True) # <=== default
print("TEST2")
for i in range(0,10):
    print("This text should show with the plot up")
```

```
[1.e+09 2.e+09 3.e+09 4.e+09 5.e+09]
[ 1  4  9 16 25]
TEST1
TEST2
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
```
----

**[2-B] plt.show(block=True)**
1. The plot showed up.
2. Outputs were printed immediately. That is, `blocking` was not effective.
3. Crash occurred when closing the KLayout's main window.

```
import matplotlib.pyplot as plt
import numpy as n

wavelength = n.array([1, 2, 3, 4, 5])
power = wavelength**2

plt.figure()
print(wavelength*1e9)
print(power)
plt.plot(wavelength*1e9, power)
plt.xlim((1*10**9,7*10**9))
plt.xlabel('Wavelength (nm)')
plt.ylabel('Transmission (dB)')
print("TEST1")
#plt.show(block=False)
plt.show(block=True) # <=== default
print("TEST2")
for i in range(0,10):
    print("This text should show with the plot up")
```

```
[1.e+09 2.e+09 3.e+09 4.e+09 5.e+09]
[ 1  4  9 16 25]
TEST1
TEST2
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
This text should show with the plot up
```

klayoutmatthias, 2023-03-23:
@kazzz Thanks for the notes about behavior on Mint.

I could reproduce the problem, but only after fixing the backend to Qt5Agg:

```
# at the beginning of the script
import matplotlib
matplotlib.use('Qt5Agg')
```

without this, "TkAgg" is selected. In that case, no crash happens, but also no plot shows.

I debugged the problem and it is due to an unfriendly behavior of the PyQt5 version on Mint which takes ownership of the QApplication object and destroys it during the Python interpreter finalization. I do not see this behavior on the PyQt5 which comes with Ubuntu 22 in my case. That is why it does not crash. 

Ubuntu 22 is using matplotlib 3.5.1 while Mint 20.3 is using 3.1.2.

I suspect the problem is due to this code in https://github.com/matplotlib/matplotlib/blob/v3.1.x/lib/matplotlib/backends/backend_qt5.py, line 94++:

```
# make place holder
qApp = None

def _create_qApp():
    """
    Only one qApp can exist at a time, so check before creating one.
    """
    global qApp     # <-- on shutdown, this instance gets deleted probably
```

They changed this specific code somewhat in other versions, still the effect remains. If I use matplotlib from pip3 which is 3.7.1 still I see the crash on Mint while I don't see it on Ubuntu. 

I also don't see the same effect on the MacOS stack traces, but I cannot really read them. Maybe it's the same thing or not.

Bottom line is that PyQt5 + matplotlib in KLayout isn't a stable combination apparently. It is very sensitive to details and after all, PyQt5 isn't made for embedding. 

I tried "WebAgg" as an alternative backend, but the tornado server also needs to embed itself into the event loop of the application - which apparently it doesn't, so the application is dead.

SVG export comes to mind - instead of showing the plot interactively, one could display the SVG.

Another, more elaborate solution is to supply a custom backend that integrates with KLayout's Qt binding, hence operates in the framework of the application natively. The latter is pretty similar to PyQt5, so I think it should be possible to adapt the Qt5Agg backend with a few changes. There is a template for backend development available with a description how to use it: https://matplotlib.org/stable/api/backend_template_api.html. The core Qt backend code is here: https://github.com/matplotlib/matplotlib/blob/main/lib/matplotlib/backends/backend_qt.py (it's imported via backend_qt5.py and backend_qt5agg.py). So basically that should be a feasible way to solve the problem once and for all. It's some effort though.

Best regards,

Matthias

klayoutmatthias, 2023-03-26:
I was able to do a quick port of the matplotlib backend to KLayout/Qt, but not without some modifications required in KLayout (bringing it closer to PyQt). The code is here: https://github.com/klayoutmatthias/mpl_klayoutqt

The changes are substantial (e.g. taking KLayout's Qt binding from Qt 5.5 to Qt 5.12). But I'll consider releasing it within the 0.28 branch. That will eventually allow the matplotlib backend to run natively inside KLayout and without PyQt and the potential side effects.