KLayout/klayout #1300: Python: Cell.dup doesn't properly copy if done by a child class

sebastian-goeldi, 2023-02-22:
I have a class that inherits from Cell (not a piece of beauty tbh, due to having to cheat in the constructor) inheriting from `klayout.dbcore.Cell`. If i want to define `dup` in the class, because my class adds additional stuff on top of the `Cell`, the instances are empty after creation.

Could you point me in the direction of what I could do to mitigate this?

The code in question is here https://github.com/gdsfactory/kfactory/blob/main/src/kfactory/kcell.py#L1395

But I have also another piece that tries it like this:

```python
def dup(self):
        kdb_copy = deepcopy(self)

        logger.info(
            "instances original: {}, instances copy: {}",
            len(list(self.each_inst())),
            len(list(kdb_copy.each_inst())),
        )
        c = KCell(klib=self.klib, kdb_cell=kdb_copy)
        c.ports = self.ports
        for inst in kdb_copy.each_inst():
            logger.info("Copying instance {}", inst.cell.name)
            c.insts.append(Instance(cell=self.klib[inst.cell.name], reference=inst))  # type: ignore[misc]
        c._locked = False
        return c
```

I think it won't change even if I remove the rest. But if necessary, I can try.

Could it be that it is linked to the `Cell` not having a constructor that allows it to be created in a linked state to the `Layout` object?

klayoutmatthias, 2023-02-22:
No, sorry. That is not supported.

The db classes are not meant to be extensible. A cell is an object that lives in C++ space. C++ knows nothing about your Python code. Think of Cell as a way to access a C++ data structure. Not as an object.

A cell is an object created, managed and used by the C++ Layout. C++ does not recognize the Python objects. That is why you cannot create a useful Cell object outside a Layout object. In the same fashion you cannot design your own shape types and just put them into the database. Supporting such a scenario would require an entirely different (and heavy) database design.

Python-extensible classes possible when the respective classes are enabled to do so. This is the case for the Qt objects, but it comes with a lot of overhead and very slow execution of virtual methods. This is not an option for efficient objects such as the database ones.

You can design KCell as a proxy (or facade) for a KLayout cell so you keep a Cell reference inside and delegate cell specific calls to the Cell object. But that means you need to do lifetime management yourself.

Matthias

klayoutmatthias, 2023-02-23:
Let me note that it's equally difficult for a C++ coder to get used to Python ways :)

I mean people are still using Skill or Tcl for layout automation, so I think we're talking about a high-level problem here.

Anyway, I tried to understand the issue, and although the `__class__` hack gives me the creeps, I think that your approach is basically working.

Here is the essence from what I understood from your code:

```
class MyCell(pya.Cell):
  def __new__(cls, name, ly, cell = None):
    if cell is None:
      cell = ly.create_cell(name)
    cell.__class__ = cls
    return cell
  def __init__(self, name, ly, cell = None):
    pass
  def dup(self):
    cc = pya.Cell.dup(self)
    return MyCell(self.name, self.layout, cc)

ly = pya.Layout()

ca = MyCell("A", ly)

cb = ly.create_cell("B")
ca.insert(pya.CellInstArray(cb, pya.Trans(10, 20)))

print(f"MyCell #1: of type {type(ca)} (name {ca.name}) with instances:")
for i in ca.each_inst():
  print(str(i))

ca_copy = ca.dup()

print(f"MyCell #2: of type {type(ca_copy)} (name {ca_copy.name}) with instances:")
for i in ca_copy.each_inst():
  print(str(i))
```

This gives me:

```
MyCell #1: of type <class '__main__.MyCell'> (name A) with instances:
cell_index=1 r0 10,20
MyCell #2: of type <class '__main__.MyCell'> (name A$1) with instances:
cell_index=1 r0 10,20
```

So the instances are copied! I wonder there the difference is to your code.

Still, I'm not fully in favor of the copy here. "dup" is basically a lie - the cell will be duplicated, but it will still be attached to the original layout. Plus a cell is never an independent object. When you create or modify a cell this will change many things - like the topological cell order or the bounding boxes that propagate through the hierarchy.

Matthias

sebastian-goeldi, 2023-02-23:
I think I have a hunch where it is coming from. It might be because I am using an abstract base class and then python messed up the inheritance (or rather my poor understanding of it).

And the layout base copy is actually exactly what I want. I don't want to mess with the hierarchy or shapes at all, these are fully managed by klayout and not touched at all (and that should also stay that way). You could think of the construct as a (nice and fancy house) and my addition as the ugly garden shack in the backyard that was built during a weekend with too much beer. If the shack isn't there, the house still works.

I found a solution though. I can let me (totally not modified at all ;) ) layout object do the copying. Then it works. I guess it is some python shenanigans I don't fully understand causing the troubles here.

On another note, I remembered something I meant to ask a long time ago but forgot: Is it possible to save [special layers](insert_special_layer) of a layout in the gds? (klayout specific metadata is enough, nothing else requires to see them). If yes, I could totally hide my tools in the basement of the house instead of the shack that gets forgotten when it's copied or reconstructed ^^.

P.S.:

I totally understand the sentiment of python. I would jump to C++ if I could convince anybody that photonic layout is better that way ;). But I think I will be pretty much alone in that stance.

klayoutmatthias, 2023-05-13:
@sebastian-goeldi

Hi Sebastian, I'm nor sure what to do with that ticket. Will the enhanced meta info thing solve this issue? Can I close this ticket? 

Thanks,

Matthias

sebastian-goeldi, 2023-05-13:
Hi Matthias,

Yes, sorry. It was actually some weirdness in how Python resolved the `dup` function. But I have changed the inheritance anyway and am not relying on directly inheriting from cell anymore (just for the layout object, which I think is an acceptable risk).

closed 2023-05-13 by sebastian-goeldi