KLayout/klayout #1302: Ask a question about "Instances."

lizimeng996, 2023-02-25:
When "Instances" is selected, I do not want pcells to be selected by PartialService.So how can I get whether instances are is_checked in PartialService?Can you give me an example? Thank you very much.

klayoutmatthias, 2023-02-25:
Are you patching C++ code?

lizimeng996, 2023-02-25:
Yes, I'm modifying the C++ code for some special needs. I want PartialSelect to be like MoveSelect. Pcell cannot be selected when instances are selected. But I don't know how to get the status of instances.

klayoutmatthias, 2023-02-25:
I see your point. Partial mode does not consider the "Select" options. That is by design and there is no trivial solution to this problem.

You can keep this issue open and ask for a fix.

closed 2023-03-20 by klayoutmatthias