KLayout/klayout #1307: LEF/DEF topics

klayoutmatthias, 2023-02-27:
This ticket is collecting some feedback regarding LEF/DEF implementation:

1. IOTIMINGS section needs to be parsed / skipped:

```
DESIGN test ;
IOTIMINGS 206 ;
- ( PIN a ) + RISE SLEWRATE 500 500 + FALL SLEWRATE 500 500 ;
- ( PIN b ) + RISE SLEWRATE 500 500 + FALL SLEWRATE 500 500 ;
- ( PIN c ) + RISE SLEWRATE 500 500 + FALL SLEWRATE 500 500
  + CAPACITANCE 250000 ;
END IOTIMINGS
END DESIGN
```

2. COMPONENTS + UNPLACED appears to potentially have a point and rotation in some flavors -> in that case, place it.

3. PIN labels are taken from NET, not PIN name

4. reading hierarchical def files (blending?) - If there is no lef MACRO for one of the def files, use the def DIEAREA bounding box instead of the lef SIZE/ORIGIN

NOTE: right now, KLayout does not support hierarchical DEF as is. 'blending' is an option to blindly merge files. The included files do now know about details of the cell calls and vice versa.

stefanottili, 2023-09-09:
I just ran into the " + UNPLACED ( x y ) orient " components are dropped issue again.
It would be nice if strm2oas would issue a warning and that these as if they're + PLACED, + FIXED or + COVER

klayoutmatthias, 2023-09-13:
I'm not sure if there is anything left. Close that issue?

stefanottili, 2023-09-13:
Did you implement 1) ? 
ignore 4). There is a warning about missing lef, good enough.

klayoutmatthias, 2023-09-13:
Right, IOTIMINGS is not implemented. I couldn't find a reference to that in the 5.8 doc.

Matthias

stefanottili, 2024-01-31:
IOTIMINGS seem to be so ancient that this issue can be closed.

closed 2024-03-23 by klayoutmatthias