KLayout/klayout #1312: Klayout for ubuntu 22.04 arm64

risheb, 2023-03-09:
Hi, I am trying to get latest klayout for ubuntu 22.04.2 arm64 version since I am running it on M1 MacBook Pro (2020). I am using virtual machine to to run ubuntu. Klayout 0.26 is too old for other software to use this. is there any deb release for arm64 version? I tried building it from source. I got the bin-release, but I am not able to install and run it. Please let me know if there any way to get the latest 0.28.5 klayout for ubuntu 22.04.2 arm64.

klayoutmatthias, 2023-03-10:
No, there is not. At least not to my knowledge.

After all, providing the packages is a community effort. I don't own hardware to do that myself and petting CI bots is not my favorite spare time activity. 

But you can always build KLayout from source yourself. Provided Ubuntu on ARM is well maintained (as it is on x86), this process is pretty straightforward and should work with the recipes given here: https://www.klayout.de/build.html (scroll down to "Building KLayout on Linux").

Matthias

risheb, 2023-03-11:
well, I am unable to install using that method since my ubuntu is arm64 version. Even if I run build.sh. without-qt I get following error
[image]

Rishab Baradwaj

klayoutmatthias, 2023-03-11:
Looks like in your case you need to find the shared object version of libruby. Specify that version as the -rblib parameter of build.sh.

In my case that is:

```
build.sh -rblib /usr/lib/x86_64-linux-gnu/libruby-3.0.so 
```

risheb, 2023-03-14:
Thanks, I am able to complete the code. can you help me on how to run the klayout? sorry I am new to this.

risheb, 2023-03-14:
I figured out a way to run it from command line but I am still seeing below errors in terminal. 
[image]
[image]
[image]

the app loads after these errors though.

klayoutmatthias, 2023-03-17:
In the sources you find a file `etc/klayout.desktop`. Copy that to `~/Desktop` and on the desktop right-click it a select "Allow Launching". It may not work out of the box unless you have include KLayout in PATH. In that case, you can specify the full path in the `Exec` field of `klayout.dektop`.

There are plenty of articles that explain this in more detail.

risheb, 2023-03-21:
Thanks! it worked like a charm!

closed 2023-03-21 by risheb