KLayout/klayout #1314: Issues installing klayout with pip on macOS related to libpng

qpavsmi, 2023-03-15:
Hardware: Macbook Air, Apple M1 chip
System Version:	macOS 13.2.1 (22D68)
Kernel Version:	Darwin 22.3.0

Tried installing libpng using homebrew and from their sourceforge, same error.
klayout 0.27.9 installs fine, but we need to have 0.28.* working to make use of QTless features.
Full `pip install klayout` printout below, relevant parts are with `#  include <png.h>`

```
pip install klayout
Collecting klayout
  Using cached klayout-0.28.5.tar.gz (3.5 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: klayout
  Building wheel for klayout (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for klayout (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [229 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-13-arm64-cpython-310
      creating build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout
      creating build/lib.macosx-13-arm64-cpython-310/pya
      copying src/pymod/distutils_src/pya/__init__.py -> build/lib.macosx-13-arm64-cpython-310/pya
      creating build/lib.macosx-13-arm64-cpython-310/klayout/lay
      copying src/pymod/distutils_src/klayout/lay/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout/lay
      creating build/lib.macosx-13-arm64-cpython-310/klayout/rdb
      copying src/pymod/distutils_src/klayout/rdb/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout/rdb
      creating build/lib.macosx-13-arm64-cpython-310/klayout/tl
      copying src/pymod/distutils_src/klayout/tl/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout/tl
      creating build/lib.macosx-13-arm64-cpython-310/klayout/lib
      copying src/pymod/distutils_src/klayout/lib/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout/lib
      creating build/lib.macosx-13-arm64-cpython-310/klayout/db
      copying src/pymod/distutils_src/klayout/db/__init__.py -> build/lib.macosx-13-arm64-cpython-310/klayout/db
      copying src/pymod/distutils_src/klayout/db/pcell_declaration_helper.py -> build/lib.macosx-13-arm64-cpython-310/klayout/db
      running egg_info
      writing src/pymod/distutils_src/klayout.egg-info/PKG-INFO
      writing dependency_links to src/pymod/distutils_src/klayout.egg-info/dependency_links.txt
      writing top-level names to src/pymod/distutils_src/klayout.egg-info/top_level.txt
      reading manifest file 'src/pymod/distutils_src/klayout.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      warning: no files found matching 'src/plugins/*/db_plugin/*.cc'
      warning: no files found matching 'src/plugins/*/db_plugin/*.h'
      warning: no files found matching 'src/py.typed'
      adding license file 'LICENSE'
      writing manifest file 'src/pymod/distutils_src/klayout.egg-info/SOURCES.txt'
      copying src/pymod/distutils_src/klayout/dbcore.pyi -> build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/laycore.pyi -> build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/libcore.pyi -> build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/py.typed -> build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/rdbcore.pyi -> build/lib.macosx-13-arm64-cpython-310/klayout
      copying src/pymod/distutils_src/klayout/tlcore.pyi -> build/lib.macosx-13-arm64-cpython-310/klayout
      running build_ext
      building 'klayout._tl' extension
      creating build/temp.macosx-13-arm64-cpython-310
      creating build/temp.macosx-13-arm64-cpython-310/src
      creating build/temp.macosx-13-arm64-cpython-310/src/tl
      creating build/temp.macosx-13-arm64-cpython-310/src/tl/tl
      Compiling with 8 threads.
      Building build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlArch.o
      Building build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlBase64.o
      Building build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlColor.o
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -dynamiclib -DHAVE_PNG=1 -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DGSI_ALIAS_INSPECT=1 -DMAKE_TL_LIBRARY=1 -I/Users/pavel.smirnov/Sources/venv/include -I/opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/include/python3.10 -c src/tl/tl/tlArch.cc -o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlArch.o -Wno-strict-aliasing -std=c++11
[... 163 lines cut ...]
                                                            ^
      src/tl/tl/tlString.cc:623:9: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
            ::sprintf (b, "\\%03o", int ((unsigned char) *c));
              ^
      /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
      __deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
      ^
      /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
              #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                            ^
      clang: warning: argument unused during compilation: '-dynamiclib' [-Wunused-command-line-argument]
      Building build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlXMLWriter.o
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk -dynamiclib -DHAVE_PNG=1 -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DKLAYOUT_MAJOR_VERSION=0 -DKLAYOUT_MINOR_VERSION=28 -DGSI_ALIAS_INSPECT=1 -DMAKE_TL_LIBRARY=1 -I/Users/pavel.smirnov/Sources/venv/include -I/opt/homebrew/opt/python@3.10/Frameworks/Python.framework/Versions/3.10/include/python3.10 -c src/tl/tl/tlXMLWriter.cc -o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlXMLWriter.o -Wno-strict-aliasing -std=c++11
      clang: warning: argument unused during compilation: '-dynamiclib' [-Wunused-command-line-argument]
      2 warnings generated.
      clang++ -Wl,-x -dynamiclib -undefined dynamic_lookup build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlArch.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlAssert.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlBase64.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlClassRegistry.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlColor.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlCommandLineParser.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlCopyOnWrite.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlDataMapping.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlDeferredExecution.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlDeflate.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlEnv.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlEquivalenceClusters.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlEvents.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlException.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlExceptions.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlExpression.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlFileUtils.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlGlobPattern.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlHeap.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlHttpStream.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlHttpStreamCurl.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlInclude.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlInt128Support.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlInternational.o build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlList.o build/temp.macosx-13-arm6 [... cut ...] 10-darwin.so -Wl,-rpath,@loader_path/
      clang: error: no such file or directory: 'build/temp.macosx-13-arm64-cpython-310/src/tl/tl/tlPixelBuffer.o'
      error: command '/usr/bin/clang++' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for klayout
Failed to build klayout
ERROR: Could not build wheels for klayout, which is required to install pyproject.toml-based projects
```

qpavsmi, 2023-03-15:
Also tried to install from the wheel `klayout-0.28.5-cp311-cp311-macosx_10_9_x86_64.whl` and it says `ERROR: klayout-0.28.5-cp311-cp311-macosx_10_9_x86_64.whl is not a supported wheel on this platform.`, I presume due to the wheel supporting macOS versions 10 and 11 while my version is macOS 13.2.1

thomaslima, 2023-03-15:
Hello @qpavsmi , very sorry about this. This issue was already reported in #1265 and fixed in a couple of PRs. It will get sorted out when 0.28.6 is uploaded to pypi. Unfortunately version 0.28.5 has this defect. If you need png, one workaround is to install from source. If you don't, I recommend installing a previous version from pypi.

klayoutmatthias, 2023-03-15:
I'll publish 0.28.6 soon. Next week latest.

qpavsmi, 2023-03-16:
Thank you for the quick response and for pointing to further discussion, this cleared up the nature of the problem for me since I'm not super familiar with the Apple platforms but am interested in providing mac support for our product.

Excited to see the fix so soon.

klayoutmatthias, 2023-03-20:
0.28.6 is out. Please reopen if the problem still persists.

closed 2023-03-20 by klayoutmatthias