KLayout/klayout #1321: SPICE reader: quoted include paths don't get expanded

fsitok, 2023-03-29:
Using KLayout as a spice parser for lctime we noticed some difficulties with include paths starting with "~".

This example with two minimal SPICE files demonstrates the problem:

`top.sp`:
```

.include "~/include.sp"
```

`$HOME/include.sp`: (empty file)

Minimal code for triggering the problem:
```python
import klayout.db as db

n = db.Netlist()
n.read("top.sp", db.NetlistSpiceReader())
```

Yields:
```
Traceback (most recent call last):
  File "/home/user/lctime_issue4/lctime_issue4.py", line 4, in <module>
    n.read("top.sp", db.NetlistSpiceReader())
RuntimeError: Unable to open file: ./~/include.sp (errno=2) in top.sp, line 2 in Netlist.read

```

However, ngspice seems to be able to handle such include paths.
`ngspice top.sp` runs without error.

It looks like include paths in quotes are not expanded (the `~` should expand to the home directory) but instead treated as relative paths.

klayoutmatthias, 2023-03-31:
Thanks a lot for this nicely prepared report.

There is no tilde expansion in the SPICE reader. That is done by the Linux shell (e.g. try `echo ~/include.sp`) and the shell is not involved here. Every path that is not absolute (i.e. starting with "/" on Linux) is treated as a relative one, which is the reason for this behavior.

Tilde expansion is not done by application code. For example, if you try to open such a file in Python, it will fail:

```
# this fails:
$ python3 -c "f = open('~/.klayout/klayoutrc')"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
FileNotFoundError: [Errno 2] No such file or directory: '~/.klayout/klayoutrc'

# while this works:
$ python3 -c "f = open('/home/matthias/.klayout/klayoutrc')"
```

So I wonder why you're using tilde paths.

Matthias

fsitok, 2023-03-31:
People started using tilde paths because ngspice does expand `~` to the home directory. Apparently ngspice does that not only in the interactive shell but also for `.include` statements in SPICE files. This leads to an incompatibility when using the same SPICE circuits with the SPICE reader of KLayout.

I did not find out if that is general SPICE behavior or specific to ngspice.

Indeed, not using `~` in include paths might be a way to go.

Is there a way to intercept `.include` statements with `NetlistSpiceReaderDelegate`?

klayoutmatthias, 2023-04-07:
The implementation is very generic now and applies to add file paths in KLayout.

closed 2023-04-13 by klayoutmatthias