KLayout/klayout #1323: New Region cause layer index invalid which makes connectivity information invalid

Jebearssica, 2023-03-30:
When we use `connect` command to add some connectivity information, and then build a new deep region. The Layout will call `insert_layer` to make each DeepLayer's layer_index invalid. So is there a solution to solve such problem? Or such process should be forbidden?

klayoutmatthias, 2023-03-31:
Please provide more information, specifically a test case (code, layout).

Jebearssica, 2023-04-03:
Here is a test segment for antenna check. I add an useless code just invoke a copy constructor function.
```
TEST(10_Antenna)
{
  db::Layout ly;
  db::LayerMap lmap;

  unsigned int poly       = define_layer (ly, lmap, 6);
  unsigned int cont       = define_layer (ly, lmap, 8);
  unsigned int metal1     = define_layer (ly, lmap, 9);
  unsigned int via1       = define_layer (ly, lmap, 11);
  unsigned int metal2     = define_layer (ly, lmap, 12);
  unsigned int diode      = define_layer (ly, lmap, 1);

  {
    db::LoadLayoutOptions options;
    options.get_options<db::CommonReaderOptions> ().layer_map = lmap;
    options.get_options<db::CommonReaderOptions> ().create_other_layers = false;

    std::string fn (tl::testdata ());
    fn = tl::combine_path (fn, "algo");
    fn = tl::combine_path (fn, "antenna_l1.gds");

    tl::InputStream stream (fn);
    db::Reader reader (stream);
    reader.read (ly, options);
  }

  db::Cell &tc = ly.cell (*ly.begin_top_down ());

  db::DeepShapeStore dss;

  std::unique_ptr<db::Region> rdiode (new db::Region (db::RecursiveShapeIterator (ly, tc, diode), dss));
  std::unique_ptr<db::Region> rpoly (new db::Region (db::RecursiveShapeIterator (ly, tc, poly), dss));
  std::unique_ptr<db::Region> rcont (new db::Region (db::RecursiveShapeIterator (ly, tc, cont), dss));
  std::unique_ptr<db::Region> rmetal1 (new db::Region (db::RecursiveShapeIterator (ly, tc, metal1), dss));
  std::unique_ptr<db::Region> rvia1 (new db::Region (db::RecursiveShapeIterator (ly, tc, via1), dss));
  std::unique_ptr<db::Region> rmetal2 (new db::Region (db::RecursiveShapeIterator (ly, tc, metal2), dss));

  db::Layout ly2;
  ly2.dbu (ly.dbu ());
  db::Cell &top2 = ly2.cell (ly2.add_cell ("TOPTOP"));

  rdiode->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (1, 0)));
  rpoly->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (6, 0)));
  rcont->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (8, 0)));
  rmetal1->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (9, 0)));
  rvia1->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (11, 0)));
  rmetal2->insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (12, 0)));

  {
    db::LayoutToNetlist l2n (&dss);

    l2n.register_layer (*rpoly, "poly");
    l2n.register_layer (*rcont, "cont");
    l2n.register_layer (*rmetal1, "metal1");
    l2n.register_layer (*rvia1, "via1");
    l2n.register_layer (*rmetal2, "metal2");

    //  Intra-layer
    l2n.connect (*rpoly);
    l2n.connect (*rcont);
    l2n.connect (*rmetal1);
    /*  not yet:
    l2n.connect (*rvia1);
    l2n.connect (*rmetal2);
    */
    //  Inter-layer
    l2n.connect (*rpoly,      *rcont);

    // an useless Region copy
    db::Region temp = *rpoly;

    l2n.connect (*rcont,      *rmetal1);
    /*  not yet:
    l2n.connect (*rmetal1,    *rvia1);
    l2n.connect (*rvia1,      *rmetal2);
    */

    l2n.extract_netlist ();

    db::Region a1_3 = l2n.antenna_check (*rpoly, *rmetal1, 3);
    db::Region a1_10 = l2n.antenna_check (*rpoly, *rmetal1, 10);
    db::Region a1_30 = l2n.antenna_check (*rpoly, *rmetal1, 30);

    a1_3.insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (100, 0)));
    a1_10.insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (101, 0)));
    a1_30.insert_into (&ly2, top2.cell_index (), ly2.insert_layer (db::LayerProperties (102, 0)));
  }
...
```

Jebearssica, 2023-04-03:
It may be unnecessary for users to build a process like this, but as a developer, i'm wondering if there exists a potential risk？

klayoutmatthias, 2023-04-04:
Sorry, I don't get your point.

The above change does not make the test fail or anything like that. So what's the problem?

Also, insert_layer does not invalid layout indexes. If you observe invalid indexes the reason is something else.

Matthias

Jebearssica, 2023-04-07:
I mean, a `DeepLayer` copied from another one will call `DeepLayer::derived`, which will create a new layer in it's `Layout`. So if I connect them, although they share same information, but the `Connectivity` will store two different layer_index.

klayoutmatthias, 2023-04-11:
In C++ a copy is a deep copy (usually). Deep copies are identical, but not the same thing. So it is intended to treat a copy as a different layer than the original. Use pointers if you want to refer to the same thing.

Jebearssica, 2023-04-12:
Thanks for your kindly reply, it would help me a lot

closed 2023-04-12 by Jebearssica