KLayout/klayout #1325: Add Layout.each_shape

EugeneZelenko, 2023-04-03:
Hi, Matthias!

It'll be nice to have `Layout.each_shape` like currently available `LayoutView.each_layer` to simplify loops over all shapes.

klayoutmatthias, 2023-04-04:
Hi Eugene,

that's kind of difficult as there are multiple cells, so iterator has to traverse cells, hence needs to tell which cell it is in.

I wonder what the use case is.

Kind regards,

Matthias

EugeneZelenko, 2023-04-04:
Hi, Matthias!

My idea was about API consistency, but I may be wrong with too simplistic understanding and could miss many complexities.

klayoutmatthias, 2023-05-13:
I think that needs further elaboration. I'll close it for now. Please reopen when you have an opinion about specific details, e.g. how to indicate the cell a shape is in.

closed 2023-05-13 by klayoutmatthias

ejprinz, 2023-05-13:
A shape is always in a cell, so I think the correct iteration is to first determine the cells (which can be done with the Layout class) and then iterate over these cells to fnd all shapes on all layers. I don't think it is needed to have the shapes know the cells they are in. GDSII does not need that.