KLayout/klayout #1326: install klayout without sudo

joamatab, 2023-04-04:
What's the best way to install klayout without sudo?

I've been looking at https://github.com/flaport/condalayout but only works on python3.8

@sebastian-goeldi 
@helgegehring

klayoutmatthias, 2023-04-04:
I personally use Docker for quick test installs. The advantage is to have a clean system. It's somewhat tricky to configure Docker to forward X11 - I have not figured out yet how to do this without disabling X11 authentication (`xhost +`). Apart from that, this works nicely. You can also map you own working directories into docker to access files sitting on the outside.

Another thing that comes to mind is snap. I have no experience however how to package for snap.

Matthias

sebastian-goeldi, 2023-04-04:
Instead of snap I would highly recommend flatpak instead. Snap has a bad history in regards to being open, whereas for flatpak both client side and server side is open and you are less bound to ubuntu flavors ;). (see https://itsfoss.com/flatpak-vs-snap/ , disclaimer: this is just the first article that explains it better than my memory, therefore not necessarily complete or any strong opinion, just personal taste)

But both options would need setup and unless the article is wrong, snap still needs root privileges to install whereas flat doesn't

proppy, 2023-04-04:
What about the conda packages at https://anaconda.org/litex-hub/klayout?

klayoutmatthias, 2023-04-05:
Is that functional? https://beta.flathub.org/apps/de.klayout.KLayout

Anaconda was tried a number of times, but so far (AFAIK) no general solution for packaging has evolved. I assume mainly because for Linux there are better alternatives.

Matthias

sebastian-goeldi, 2023-04-05:
The flatpak works for me, yes.

atorkmabrains, 2023-04-10:
@joamatab 

Download this makefile anywhere:
https://gist.github.com/atorkmabrains/56d924c7dee111febfa741d289aba9cb

And just run:
```bash
make install_klayout-v0.28.5
```

And it will install `klayout` version `v0.28.5`. To install another version for example:
```bash
make install_klayout-v0.28.6
```

So on so forth.

atorkmabrains, 2023-04-10:
@joamatab Please make sure to install required packages for building klayout:
https://www.klayout.de/build.html

For Ubuntu 22.04 for example:
```
C++ toolchain: gcc, g++ and make
Qt including development tools: qtbase5-dev, qttools5-dev, libqt5xmlpatterns5-dev, qtmultimedia5-dev, libqt5multimediawidgets5 and libqt5svg5-dev
Ruby: ruby and ruby-dev
Python: python3 and python3-dev
zlib: libz-dev
```

klayoutmatthias, 2023-05-13:
Can I close this ticket?

ejprinz, 2023-05-13:
Specifically for a RHEL7 system, this seems to work (done only one time, so YMMV):

Download the Centos7 binary rpm archive from:
https://www.klayout.de/build.html

Transfer to the RHEL7 system.
Run this command in a new (local) directory:

rpm2cpio  klayout-0.28.7-0.x86_64.rpm |cpio -diun

Move the files from this directory into e.g. the ~/.local/(bin, lib, share) tree.
Add script to export LD_LIBRARY_PATH variable to contain ~/.local/lib.
 
Then start klayout with this script.

#! /bin/bash

export root=~/.local

cd $root

 

export LD_LIBRARY_PATH=${root}/lib/klayout

${root}/bin/klayout

Note that all dependent libraries have to be installed already with rpm's in the RHEL7 system. 

Also for RHEL7 the rpm from klayout.de is based on Qt4 so it does not include the 2.5d view feature.

closed 2023-08-09 by klayoutmatthias