KLayout/klayout #1335: Enable snapping to the center of a shape while editing and snap to a center of an edge while editing.

atorkmabrains, 2023-04-10:
Currently, snapping doesn't support snapping to center. It only supports corners and general locations on edges. But It would be very useful to allow the snapping the center of edges and objects while editing.

Partmedia, 2023-05-25:
I agree. The path tool already has something like you describe to continue existing paths, but this would also be useful for any shape.
[image]
Aside: I would also find a "continue path width" setting that sets the width of the new path to be the width of the old one.

atorkmabrains, 2023-05-26:
@Partmedia Yes, we used path as possible alternative. But still haven't such capability on polygons and edges in general is extremely helpful for manual layout.