KLayout/klayout #1336: Supporting relative object alignment for pcell development like in gdsfactory

atorkmabrains, 2023-04-10:
Currently, there is no way in `pya` package for klayout to support relative object alignment similar to the behavior documented here: https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.Component.html including the following functionalities:

- xmin assignment and usage
- xmax assignment and usage
- relative movex and usage
- relative move and usage
- relative movey and usage
- ymin assignment and usage
- ymax assignment and usage
- center assignment and usage
- Relative to a point flipping/mirroring in X and Y
- Relative objects alignment like "Connect" command in gds_factory. Please use a different name like "Align"
- Definition of named locations like ports that move with the object if the object moved.

The above is a short a list of functionality that currently exists in GDS Factory that would be very helpful for Pcell development.

Originating discussion: https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/issues/28

Example Code that use relative object alignment for Pcells:
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/cells/klayout/pymacros/cells/draw_fet.py

cc @proppy @klayoutmatthias @joamatab @mithro

klayoutmatthias, 2023-04-11:
Thanks for this ticket. Could you point me to some examples and specific functions you need to have available?

As you target device PCells there are some alternative concepts such as the graphical PCell approach which includes stretching for example. I'd like to come up with a generic concept that includes a variety of options.

Thanks,

Matthias

atorkmabrains, 2023-04-12:
@klayoutmatthias Here:
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr/blob/main/cells/klayout/pymacros/cells/draw_fet.py

lukasc-ubc, 2023-11-17:
Hi folks,

We implemented similar functionality in SiEPIC-Tools, where you can snap components together.  It more elaborate since it invovles having pins and components defined.  https://github.com/SiEPIC/SiEPIC-Tools/wiki/Scripted-Layout#connecting-two-components-together-snapping-pins-together

joamatab, 2024-06-12:
we have the same functionality in gdsfactory which uses since v8.0.0 the klayout backend

see [code](https://github.com/gdsfactory/gdsfactory) and [docs](https://github.com/gdsfactory/gdsfactory)

lukasc-ubc, 2024-06-12:
> we have the same functionality in gdsfactory which uses since v8.0.0 the klayout backend
> 
> see [code](https://github.com/gdsfactory/gdsfactory) and [docs](https://github.com/gdsfactory/gdsfactory)

Hi @joamatab 

Does this mean that we can use our PCells in KLayout format within GDSfactory?  That would be really nice.

thanks

sebastian-goeldi, 2024-06-16:
Hi @lukasc-ubc ,

That is (not yet) possible. gdsfactory and by extension kfactory (which is the intermediate layer between gdsfactory and klayout API) are not using PCell classes. Furthermore their classes do not inherit klayout classes directly (mainly as after Matthias' advice I stayed away from extending those classes as it gets interesting when you get modified classes back etc).

What we probably could do at some point is provide a decorator (or similar) for pcell registration.

My main gripe with it is that I don't see this scaling very well beyond very small photonic tapeouts (and unless I misunderstood Matthias in the past, he agrees on PCells being only somewhat scalable). Therefore, I always saw it as a rather limited work vs profit.

If you have a concrete use case, let me know and I can come up with something.

lukasc-ubc, 2024-06-16:
Hi @sebastian-goeldi 

All commercial silicon photonics products to date use a small number of components and can be drawn by hand, or with some small scripting assistance. 

There are of course large scale systems people want to build, and there is makes sense to program the whole thing. 

I believe we need a method to support both graphically drawn circuits, and scripted. 

In my work we go back and forth between the two approaches. 

So I don't see a future where 100% of designs will only be scripted. 

Hence I think we should work towards a unified framework that allows for both. 

The pcell decorator concept would be fine. 

Also, can gdsfactory now be imported within KLayout's Python environment?  This way gdsfactory functions could be used within the klayout Application. 

Thank you