KLayout/klayout #1337: .cir file from klayout LVS does not match ngspice netlist for GF180

spnadig, 2023-04-10:
When we compare the netlist generated from schematic (ngspice) to the .cir file from LVS of the layout, we see that the port orders for devices like fets are flipped (esp for Source, drain). Additionally , some of the parameters of subcircuits are also incorrect.

Is there a way to force the .cir to follow ngspice format and PDK definitions?  - such that if I run the layout extracted .cir file on ngspice it should work as expected

spnadig, 2023-04-10:
@proppy
@njcoburn

klayoutmatthias, 2023-04-11:
Could you give an example?

For simple planar FETs source and drain are symmetric, so the assignment of source and drain is not predictable. The LVS netlist compare allows source and drain to be swapped, so there is a match even if the assignment isn't the same than for the schematic netlist.

You can configure netlist output using the a `NetlistSpiceWriterDelegate` object (see https://www.klayout.de/doc-qt5/manual/lvs_io.html).

But the main purpose of the extracted netlist is matching against the schematic (input) netlist. As upon matching LVS both are identical, there is no need for tailoring the output. Usually the schematic netlist is made in a way that it is suitable for simulation (with a testbench). So emphasis is on the ability to read netlist of various, ngspice-compatible flavors (now also include parametric circuits).

Matthias

closed 2023-08-09 by klayoutmatthias