KLayout/klayout #1338: Time improvement about Boolean operation

Jebearssica, 2023-04-11:
Hi Matthias! When I run boolean operation between a region with many polygons and a big region, the computation is really slow. Here's the demo file:
[Boolean_demo.zip](https://github.com/KLayout/klayout/files/11196174/Boolean_demo.zip)

I'm wondering if there's any suggestion about efficiency improvement?

Jebearssica, 2023-04-11:
EdgeProcessor integrates many operators. When doing boolean operation, the edge is discarded later than the individual boolean progress. Is there a plan to split out a processor, such as BooleanProcessor, to improve the performance?

klayoutmatthias, 2023-04-11:
No, there is no such plan and I don't think that is a good idea. It will hardly save time, but maintenance will become more difficult. 

I event don't know what your actual problem is (the .zip file is just a single OASIS file). One layer is >6M rectangles and the other one single one. In order to speed booleans up here you have to do more than patching EdgeProcessor.

After all, this is open source - you can improve it yourself, but that does not mean I share the effort. I think the EdgeProcessor is good enough for now and yet they it is simple enough so I have a chance to maintain the code in the framework of a single-person spare time project. If you have a better boolean, please use that. If you want top notch performance, please consider using commercial-grade software.

Jebearssica, 2023-04-12:
Thanks for your kindly reply, it would help me a lot

closed 2023-04-12 by Jebearssica