KLayout/klayout #1341: Storing (arbitrary) data in metadata

sebastian-goeldi, 2023-04-13:
Hi Matthias,

I am creating this ticket to track what we talked about ;) (I hope I am mot misremembering or straight up telling wrong things)

It would be nice if we can store metadata of cells (and maybe instance) in metadata. This would enable us to use Klayout fpr conecept such as ports in photonics. My poor understanding is that this is how PCells currently handle this ajd it could be generalized to enable access to the metadata.

Feel free to edit/delete/recreate however you see fit.

@joamata @gdspaul

klayoutmatthias, 2023-04-16:
Thanks :) I have planned to include it in 0.28.8. 

Matthias

EugeneZelenko, 2023-04-25:
@sebastian-goeldi: Did you try to use user properties? They allow to store arbitrary values.

sebastian-goeldi, 2023-04-25:
> @sebastian-goeldi: Did you try to use user properties? They allow to store arbitrary values.

Hi Eugene,

Yes I am totally aware. The problem there is that user properties are not as flexible as the MetaInfo. Due to gds limitations, I would need to do a manual mapping of index <-> object. This way allows more flexibility.

klayoutmatthias, 2023-04-25:
@EugeneZelenko meta info can store arbitrary data also in GDS which is otherwise limited to strings and numeric keys. They also support complex objects and they do not pollute the user property space. Similar to user properties but better.

EugeneZelenko, 2023-04-25:
@klayoutmatthias: Thank you for clarification!

sebastian-goeldi, 2023-05-13:
@klayoutmatthias 

Hi Matthias,

I saw that you merged test cases and the enhancement into `master`. Because I am an impatient waiter, I have tried it out with running the python module compiled from `master`. What I notices is that even if the LayoutMetaInfo added to a Cell with `add_meta_info` (even with `persisted = True`, the Cell reports it properly, but when saving the layout, it doesn't save (or load not sure how to find out). Is that not implemented yet, or is there something missing in the code to make that work?

Lmk if I just need to be a bit more patient, or if there is something that I can help with. ^^

Best,
Sebastian

closed 2023-05-23 by klayoutmatthias