KLayout/klayout #1343: Global setting to avoid reload confirmation a file is watched

adriengcql, 2023-04-18:
The file watch mode comes very handy when editing a gds file outside of klayout. In certain cases it would be nice not to have to go back to the klayout interface to confirm the reload, i.e. bypass this dialog:
[image]

One solution might be to add an option to choose whether or not confirmation is required.
[image]

klayoutmatthias, 2023-04-18:
This option is already there ... what is the point?

adriengcql, 2023-04-19:
> This option is already there ... what is the point?

Hello Matthias, thanks for the reply, the need is to reload the file immediately without going through the confirmation dialog. I couldn't find that option but I maybe missing something ? The screenshot might be unclear, the idea is to add an option next to the `Watch Layout Files` one.

klayoutmatthias, 2023-04-19:
I see, but no, that feature does not work.

Reason is that you cannot tell when a file is fully written. When you detect a file update, another process may still be writing the file. While this isn't a problem with purely serial files like text files, partial layout files are usually broken. All you get are error messages in that case. Specifically if the remote process writes to a network drive.

Automatic reload does not work. I tried it. You need a separate channel to trigger the reload such as IPC through sockets or similar mechanisms.

Matthias

adriengcql, 2023-07-26:
It makes sense indeed. IPC triggers sound like a good lead, I'll try that

sebastian-goeldi, 2023-07-26:
Shameless plug here for [klive](https://github.com/gdsfactory/klive), since it does basically exactly what you want. It uses a TcpServer from Qt, that listens on 8082, and if you send it a json, it will reload/open the file specified there.

It's still a bit barebone (not much GUI interaction yet and  not super configurable), but does the trick for most cases I've come across. Lmk if something doesn't work or it doesn't suit your needs exactly but you think it could do so. Worst case it could serve as a source for inspiration.

closed 2023-08-09 by klayoutmatthias