KLayout/klayout #1344: File > Import > Other Files Into Current crashes KLayout

martin-gustafsson, 2023-04-20:
Selecting Import > Other Files Into Current from the File menu immediately shows an "Application crashed" window. Clicking "Ok" does not remove the error window. This is on MacOS 12.6.3 with KLayout 0.28.6. 

Test case: Start KLayout, make a new Layout with default technology and options, then select the command in the menu.

Trace below:
-----------------
Signal number: 6
Address: 0x7ff81288600e
Program Version: KLayout 0.28.6 (2023-03-19 rd38899bf1)

Backtrace:
/Applications/klayout.app/Contents/Frameworks/libklayout_lay.0.dylib +0x20a43b _ZN3lay25enable_signal_handler_guiEb
/usr/lib/system/libsystem_platform.dylib +0x3dfd _sigtramp
/usr/lib/system/libsystem_platform.dylib +0xffff8007ed732400 _sigtramp
/usr/lib/system/libsystem_c.dylib +0x81d24 abort
/usr/lib/libc++abi.dylib +0x10082 abort_message
/usr/lib/libc++abi.dylib +0xf4cd _ZSt11__terminatePFvvE
/usr/lib/libc++abi.dylib +0xf458 _ZSt9terminatev
/usr/lib/libc++abi.dylib +0xf458 _ZSt9terminatev
/usr/lib/libc++abi.dylib +0xf458 _ZSt9terminatev
/usr/lib/libc++abi.dylib +0xf458 _ZSt9terminatev
/usr/lib/libc++abi.dylib +0x11d05 __cxa_get_exception_ptr
/usr/lib/libc++abi.dylib +0x11ccc _ZN10__cxxabiv1L22exception_cleanup_funcE19_Unwind_Reason_CodeP17_Unwind_Exception
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x6cfe0 _ZN2tl19XMLStructureHandler10charactersERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE
/Applications/klayout.app/Contents/Frameworks/libklayout_tl.0.dylib +0x6d059 _ZN2tl19XMLStructureHandler10charactersERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEE
/Users/mgustafs/anaconda3/lib/libQt5Xml.5.15.2.dylib +0xe346 _ZN16QXmlSimpleReader5parseEPK15QXmlInputSourceb

Kazzz-S, 2023-04-20:
I'm sorry to see your report of program crashes. However, I could not reproduce the problem in my environment.

**My environment**
1. OS has been updated to **12.6.5**
2. Anaconda has been updated by **conda update --all**

I've attached some relevant screenshots.
[image]
[image]
[image]
[image]
[image]
[image]

klayoutmatthias, 2023-04-20:
I'm sorry to hear that too, but as well, I cannot reproduce the problem on Windows 10 and Ubuntu 22. 

The stack trace indicates an uncaught exception somewhere from the XML parser. It should be longer than the piece you pasted. Could you share the full stack trace? That might give a hint why that happens. I'm not aware that anything XML needs to be loaded when you import another layout.

Matthias

klayoutmatthias, 2023-05-12:
@martin-gustafsson Any updates?

martin-gustafsson, 2023-05-12:
Matthias and Kazunari, thanks for looking into this problem. I don't know how to access the full stack trace, but if you tell me I will be glad to post it.

klayoutmatthias, 2023-05-12:
You pasted some trace into the initial post. I believe it should continue a few lines below. You should be able to see these lines when you stack trace window pops up.

martin-gustafsson, 2023-05-12:
No, the line that starts with "Users/mgustafs/anaconda3/lib/libQt5Xml.5.15.2.dylib" is the last one, and it ends as above with "EPK15QXmlInputSourceb"

klayoutmatthias, 2023-05-12:
Hmm ... that's bad. That would give us a hint what's going wrong. 

Problem still is, I have now idea why some XML file is read and why it goes wrong.

Could you try the following: in your user's home you should find some folder called ".klayout". Please rename it to something else and check if that changes anything.

Matthias

klayoutmatthias, 2023-05-12:
Another thing to try: run klayout from the terminal and with "-d 31" (increased debug level) and see if the output tells anything interesting.

Matthias

martin-gustafsson, 2023-05-12:
I tried renaming the .klayout folder and the problem still persisted. I started KLayout just now with the "-d 31" flag and got a lot of output, including execution of a bunch of macro code and a couple of errors with that. There could be things in that output that I shouldn't post publicly, but I'd be happy to email it directly to you.

klayoutmatthias, 2023-05-12:
Please mail to "contact 'at' klayout.de"

martin-gustafsson, 2023-05-12:
Ok, done!

klayoutmatthias, 2023-05-12:
Hmm ... thanks. Did this log include the crash? There is nothing showing at the end unfortunately. The errors are not good, but should not harm apart from disabling the respective macros.

martin-gustafsson, 2023-05-12:
Sorry, that didn't include the crash. The crash adds the following lines below the ones I sent by email.:

Created layout L1
Sorting: started
Sorting: 0 (user) 0 (sys) 0 (wall)
Redrawing: started
Redrawing: 0 (user) 0 (sys) 0 (wall)
libc++abi: terminating with uncaught exception of type tl::XMLLocatedException
sh: addr2line: command not found

klayoutmatthias, 2023-05-13:
Ok, that's what I suspect.

Well, I think I can give some workaround here: look into `~/.klayout/klayoutrc` and search for the `<stream-import-spec2>` element.

Replace it by 

```
 <stream-import-spec2>&lt;stream-import-data/&gt;</stream-import-spec2>
```

or if it does not exist, add it.

This probably will make the function work.

But it points to a deeper problem rooted in the way the Qt libraries are built in comparison to KLayout (ABI incompatibility). Basically this can happen when the Qt libraries are built with a C++ compiler not compatible with the one that KLayout was built with.

Matthias

klayoutmatthias, 2023-08-09:
I will close this for now as that is not something I can fix in the sources.