KLayout/klayout #1346: no wheels for M1

joamatab, 2023-04-23:
How can we make `pip install klayout` work on M1?

@thomaslima

thomaslima, 2023-04-23:
Does it not work now? In mine, it works, but it builds from source. Have you managed to get ARM action runners in GitHub?

joamatab, 2023-04-23:
CICD wheels supports M1 compatible wheels

joamatab, 2023-04-24:
Hi, Matthias,

it's working, can we make a new release?

https://github.com/KLayout/klayout/actions/runs/4781311443

@thomaslima 
@klayoutmatthias

thomaslima, 2023-04-24:
@joamatab the link you mentioned only shows x86 wheels. Where are the arm64 wheels?

joamatab, 2023-04-24:
sorry, this is the latest link
https://github.com/KLayout/klayout/actions/runs/4789102928/jobs/8516569537

you can see arm wheels there

thomaslima, 2023-04-24:
Hmmmm sorry I only see the x86 ones in the artifacts of the macOS-11 job.
Am I missing something?

joamatab, 2023-04-24:
sorry, again, this should definitely have them

https://github.com/KLayout/klayout/actions/runs/4791439354

klayoutmatthias, 2023-04-25:
I have uploaded the arm64 wheels manually to PyPI, but they are not tested as testing is not supported for the M1 cross builds. Please check if they work.

klayoutmatthias, 2023-04-28:
Do they work?

joamatab, 2023-05-04:
@thomaslima 

I don't have an M1 :)

thomaslima, 2023-05-05:
I just tested on a fresh install using python 3.10. pip installs from the wheel. But upon importing klayout I get the following error:

```
Python 3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:12:31) [Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/homebrew/Caskroom/miniforge/base/envs/klayout_issue_1346/lib/python3.10/site-packages/klayout/__init__.py", line 2, in <module>
    from .tl import __version__
  File "/opt/homebrew/Caskroom/miniforge/base/envs/klayout_issue_1346/lib/python3.10/site-packages/klayout/tl/__init__.py", line 1, in <module>
    import klayout.tlcore
ImportError: dlopen(/opt/homebrew/Caskroom/miniforge/base/envs/klayout_issue_1346/lib/python3.10/site-packages/klayout/tlcore.cpython-310-darwin.so, 0x0002): symbol not found in flat namespace '_png_create_info_struct'
```

I'm not sure why that happened and how it would have passed Github's tests. Where did the wheel come from? Do we have logs for that compilation?

klayoutmatthias, 2023-05-05:
github provides cross builds for arm64 only and does not support testing. So no testing happens in fact. That is why I asked for confirmation of functionality. Apparently the solution isn't working. 

I can't tell much from the error message - if that was a missing PNG library (we had this case) I think the message should be different. Some linking problem? But why on M1 only?

thomaslima, 2023-05-22:
I inspected the wheel published on pypi and it does not ship with libpng, nor does it link to system's libpng. I'm not sure why: 
```
otool -L klayout-0.28.7-cp310-cp310-macosx_11_0_arm64/klayout/lib_tl.cpython-310-darwin.so
klayout-0.28.7-cp310-cp310-macosx_11_0_arm64/klayout/lib_tl.cpython-310-darwin.so:
	@rpath/lib_tl.cpython-310-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/usr/lib/libexpat.1.dylib (compatibility version 7.0.0, current version 8.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
```

Compare with a wheel built on my native M1 (prior to running delocate-wheel):
```
otool -L dist/klayout-0.28.7-cp310-cp310-macosx_11_0_arm64/klayout/lib_tl.cpython-310-darwin.so
dist/klayout-0.28.7-cp310-cp310-macosx_11_0_arm64/klayout/lib_tl.cpython-310-darwin.so:
	@rpath/lib_tl.cpython-310-darwin.so (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/usr/lib/libexpat.1.dylib (compatibility version 7.0.0, current version 8.0.0)
	/opt/homebrew/opt/libpng/lib/libpng16.16.dylib (compatibility version 56.0.0, current version 56.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 905.6.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.100.5)
```

Conducting my own tests in #1378

thomaslima, 2023-05-22:
Similar issue was reported [here](https://groups.google.com/g/comp.lang.python/c/0qD6yD8LTuI):

> ld: warning: ignoring file /usr/local/lib/libfreetype.dylib, building for macOS-arm64 but attempting to link with file 
built for macOS-x86_64 

I missed the original logs from @joamatab , but I suspect that during linking the host's libpng's library was ignored. It looks like we need to (cross)compile libpng from source. This is a pain. We got a few options:
- do nothing and only ship source for M1 pip installs
- buy a M1 Mac and run an external GitHub action
- cross compile libpng from source
- disable PNG for M1 wheels

This situation will go away in Q3 according to GitHub because they will enable Public M1 Runners.

thomaslima, 2023-05-22:
I was able to confirm in https://github.com/KLayout/klayout/actions/runs/5047875508 that this was exactly the problem with the arm64 wheels.

My intuition is that option 3 is the only viable option for @klayoutmatthias . I will work on that unless someone points out a better option.

klayoutmatthias, 2023-05-22:
Thanks. 

we had some trouble with libpng already: https://github.com/KLayout/klayout/issues/1314 and https://github.com/KLayout/klayout/issues/1265. I don't know what's so special about libpng - the library isn't very complex. There is not trouble with curl or expat.

Maybe the issue is "libpng-config" which produces the wrong flags for cross-compilation? After all it's called to provide compilation flags for direct builds.

Matthias

thomaslima, 2023-05-25:
M1 wheels are up in new version 0.28.8. Thanks Matthias!
@joamatab 

```
 ~ » pip install -U klayout
Requirement already satisfied: klayout in /opt/homebrew/Caskroom/miniforge/base/envs/lumos/lib/python3.10/site-packages (0.28.6)
Collecting klayout
  Downloading klayout-0.28.8-cp310-cp310-macosx_11_0_arm64.whl (16.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 16.2/16.2 MB 29.5 MB/s eta 0:00:00
Installing collected packages: klayout
  Attempting uninstall: klayout
    Found existing installation: klayout 0.28.6
    Uninstalling klayout-0.28.6:
      Successfully uninstalled klayout-0.28.6
Successfully installed klayout-0.28.8
 ~ » python
Python 3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:12:31) [Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
i>>> import klayout
>>>
```

closed 2023-05-25 by thomaslima