KLayout/klayout #1348: "Reload Files" popup prevents closing  KLayout

nmz787-intel, 2023-04-24:
I am using Windows. I have files open that are programmatically generated, and often I have several windows open to debug my cell synthesis at various stages. The "Reload Files" popups are helpful when I'm actively working. However when I finish a task I will often then go and commit my code, often re-running my synthesis if I am also cleaning up debug statements or comments. Finally when I finish, I often want to close all KLayout instances at once, which I attempt to perform by right-clicking on the taskbar icon and then clicking "Close all windows". Unfortunately KLayout's "Reload Files" popup prevents this 'close all' operation, and I will then have to click through all the windows I have open (which all are prevented from being closed, presumably because the "Reload Files" popup is waiting to be displayed, and somehow there's only one of these popups allowed per group of windows) to find the window that has a popup that I can say "yes" or "no" to, before being able to close the program.

nmz787-intel, 2023-04-24:
Another weird issue is that, since I work on a laptop but also have a desk with external monitors... sometimes the "Reload Files" popup ends up off-screen when I un-dock. This forces me to CTRL-ALT-DEL to get into task manager (gone are the days when you could open task manager from a right-click on the taskbar), find KLayout, and force-kill it there.

klayoutmatthias, 2023-04-24:
@nmz787-intel The popup is not something special, that is a standard feature of Qt / Windows. I cannot change this behavior. The solution is only to turn off update detection which you can do by disabling "check for updates":

[image]

Sometimes people ask for automatic reload (no popup), but that does not work (see https://github.com/KLayout/klayout/issues/1343).

klayoutmatthias, 2023-04-24:
Other programs don't show a popup but a status bar like this:

[image]

Maybe that is an option to prevent the modal popup.

nmz787-intel, 2023-04-24:
hmm, ok, looking at the QMessageBox (https://github.com/KLayout/klayout/blob/34d147f1cecc53daae85bdec56755a5632ebe883/src/lay/lay/layMainWindow.cc#L627) API here:
https://doc.qt.io/qt-6/qmessagebox.html#question
I see a warning:
```
Warning: Do not delete parent during the execution of the dialog. If you want to do this, you should create the dialog yourself using one of the [QMessageBox](https://doc.qt.io/qt-6/qmessagebox.html) constructors.
```

presumably since I cannot close the window due to the modal action, they are warning about programmatically closing the parent. It makes me wonder however, if one of the other constructors has some parameter for "level of modality" (i.e. none of the GUI widgets can be interacted with, except the X button or close event/action (whatever the right-click and close from the Windows task bar is)).
The generic `QMessageBox` constructor has this parameter `Qt::WindowFlags`: https://doc.qt.io/qt-6/qt.html#WindowType-enum
but I'm not immediately sure which (combination of) options might be interesting. 

I certainly click "yes" (or press enter) on reload enough that it does save me a button press (CTRL+R), that I probably would prefer it to remain enabled... but I guess I also press CTRL-R enough out of paranoia (as in, "really, that code fix still didn't fix my polygons?!?!") that maybe I should just disable the update check. Hmm.

Googling for solutions or similar scenarios isn't turning up much, so it seems there isn't a great/easy fix for this. So I guess "no big deal", thanks!

nmz787-intel, 2023-04-24:
> Some programs don't show a popup but a status bar like this:
> 
> [image]
> 
> Maybe that is an option to prevent the modal popup.

status bar idea seems reasonable.

closed 2023-04-24 by nmz787-intel