KLayout/klayout #1350: Inconsistent behavior of LayoutView.on_show

EugeneZelenko, 2023-04-25:
`LayoutView.on_show` behaves inconsistently for session with single (not called) or multiple layout views (called for first view and then for each clicked). I tried 0.28.7.

Script to reproduce (`klayout -rm <script> -u <session>`):

```
import pya

def on_show(current_view):
    with open('Log.txt', 'a') as file:
        file.write('current_view, cellviews: {0} {1}\n'
                   .format(current_view, current_view.cellviews()))

def on_current_view_changed():
    current_view = pya.Application.instance().main_window().current_view()
    if current_view is None:
        return
    current_view.on_show = lambda __current_view=current_view : on_show(__current_view)

with open('Log.txt', 'w') as file:
    file.write('')
main_window = pya.Application.instance().main_window()
main_window.on_current_view_changed = on_current_view_changed
```

klayoutmatthias, 2023-04-30:
The behavior I observed (Ubuntu-22) is slightly different: on_show is never called initially, only when a tab is clicked.

"on_show" is essentially a replication of Qt's "showEvent". This event is raised when the widget becomes visible. I assume it is not called because when "on_current_view_changed" is issued, the widget already is visible.

I can change this, but on the other hand, "on_current_view_changed" is essentially the same thing than "on_show". In the main application context, "on_current_view_changed" is called when the tab is changed and this comes with "on_show". So in that context, "on_show" does not have a special meaning and you should be able to use "on_current_view_changed". Only when using the view as a standalone widget, "on_show" will be useful as there is no "on_current_view_changed".

EugeneZelenko, 2023-05-02:
Hi, Matthias!

Since changes for #1349 serves my purpose in much better way (process everything without no need to track processed views/cellviews) than `on_current_view_changed` and `on_show` it does not make too much sense to fix anything about them.

klayoutmatthias, 2023-05-12:
Thanks ... so OK to close this ticket?

EugeneZelenko, 2023-05-12:
Hi, Matthias!

Sure!

closed 2023-05-12 by EugeneZelenko